OpenCPN/OpenCPN #1705: Win Start up message about failed last run

Hakansv, 2020-02-09:
On Windows:
This message is shown before a check if another instance is running. 
[image]

So, if one instance is running and I, by mistake, start another this message can be shown but is of course quietly closed since the start up process is closed down a few rows down. I know what's it's about but it would confuse a standard user.  Could we move it further down or if not suitably for other OS make a check also here for another running instance (and not portable)?

[image]

leamas, 2020-02-27:
hm... if we just move it, the check will  be disabled for Android. We don't want that.

Still, I see your point  and it's a good catch. One simple-minded approach:

```
diff --git a/src/chart1.cpp b/src/chart1.cpp
index 7745d51..d558efe 100644
--- a/src/chart1.cpp
+++ b/src/chart1.cpp
@@ -1707,11 +1707,6 @@ bool MyApp::OnInit()
     // Instantiate the global OCPNPlatform class
     g_Platform = new OCPNPlatform;
 
-    // Check if last run failed, set up safe_mode.
-    if (!safe_mode::get_mode()) {
-        safe_mode::check_last_start();
-    }
-
 
 #ifndef __OCPN__ANDROID__
     //  On Windows
@@ -1759,7 +1754,13 @@ bool MyApp::OnInit()
             return false;               // exit quietly
         }
     }
-#endif
+#endif  //__OCPN__ANDROID__
+
+    // Check if last run failed, set up safe_mode.
+    if (!safe_mode::get_mode()) {
+        safe_mode::check_last_start();
+    }
+
 
     //  Perform first stage initialization
     OCPNPlatform::Initialize_1( );

```

Going to bed now. WiIll try to look at it later. If you have time to try my patch it would of course be helpful

Hakansv, 2020-02-27:
Yes, your move is fulfilling the intention. My arrow was more a point of direction anywhere below the portable check than an exact position. (At least in my mind :) )
On Win a second start of O while already running will just not work, as before. And no safe mode message. Also tested OK now.
But Linux can allow a second instance - or? There you've to test. My Rpi is busy on a long run test.

Now following your example....
Håkan

Hakansv, 2020-02-27:
And BTW. My opinion:
Make the safe mode message default No. Otherwise we will get support issues about "No plugins after crash!!??" The most frequent crash may be accidentally power off on a RPi or similar. Let the user make an active decision to run in safe mode. Most may use the method to answer "default" on silly PC questions.

leamas, 2020-02-28:
> But Linux can allow a second instance - or?
No.

leamas, 2020-02-28:
> Make the safe mode message default No. 

@bdbcat : thoughts?

bdbcat, 2020-02-28:
Agree.
Default should be "No", for reasons stated

leamas, 2020-02-28:
OK. Will make a PR

transmitterdan, 2020-02-28:
The “safe message box” must have a timeout that witching say 30 seconds automatically selects NO and continues to run the program. Otherwise we will get massive complaining. 

Why, you might ask? Well, if you have a headless system it may be not so easy to answer that little box.  I can think of other projects where a stuck box at startup that prevents the system from running generated a lot of complaints.

Hakansv, 2020-02-28:
TDan.,
Many thanks, I do agree. This come to my mind today riding the car. What you say all make sense. We have not so long ago for the same reason dismissed the new log file and rename of the old to loglog message.

leamas, 2020-02-28:
Also I agree. I'm on it, time permitting.

leamas, 2020-03-01:
Default option is 'No', and we have a timeout. Time to close?

tai-fun, 2020-03-01:
Just a minute....
Testing on Ubuntu 19.10
I "xkill" an instance. Dialog comes up with "No" pre-selected. Times out
and starts.....
but in  minimal configuration without any plugins.
Some logic twist here!

leamas, 2020-03-01:
Ack. It is a bug in OCPN_TimedHTMLMessageDialog. I'm on  it.

leamas, 2020-03-01:
This patch should fix it:

    --- a/src/chart1.cpp
    +++ b/src/chart1.cpp
    @@ -12434,7 +12434,7 @@ void OCPN_TimedHTMLMessageDialog::OnYes(wxCommandEvent& WXUNUSED(event))
     {
         SetReturnCode(wxID_YES);
         if(IsModal())
    -        EndModal( wxID_YES );
    +        EndModal( m_style & wxNO_DEFAULT ? wxID_NO :  wxID_YES );
         else
             Hide();
     }

Arguably, the patch does the right  thing, returning the default value on timeout. However, this is tinkering with code used in a lot of places, most of which above my paygrade.

@bdbcat: thoughts?

leamas, 2020-03-01:
Here is a IMHO better patch (please disregard the previous...)

    diff --git a/src/chart1.cpp b/src/chart1.cpp
    index 7745d51..6e596d6 100644
    --- a/src/chart1.cpp
    +++ b/src/chart1.cpp
    @@ -12471,7 +12471,7 @@ void OCPN_TimedHTMLMessageDialog::OnClose( wxCloseEvent& event )
     void OCPN_TimedHTMLMessageDialog::OnTimer(wxTimerEvent &evt)
     {
         if(IsModal())
    -        EndModal( wxID_YES );
    +        EndModal( m_style & wxNO_DEFAULT ? wxID_NO :  wxID_YES );
         else
             Hide();
     }

The question is the same as above:  while seemingly OK, is it worth tinkering with the code and behaviour of this function?

leamas, 2020-03-01:
Filed  new PR with last patch above. It seems that safe_mode is the first example of using wx_NO_DEFAULT with the OCPN_TimedHTMLMessageDialog() method, so the patch seems safe.

rgleason, 2020-03-01:
In Ubuntu 18.04 I get a similar message along with one that I don't have the correct permissions, however Opencpn does run. (Also I am missing tide data according to another message)

bdbcat, 2020-03-02:
Alec...
Your patch OK.  Never seen a use with "NO" as default before. Of course, we could change the wording sense of the message, and use default as "Yes". But I prefer your current solution.

Hakansv, 2020-03-02:
Works fine.
A "small" issue grew to a complex but ended very nice.
Thanks

closed 2020-03-02 by Hakansv