OpenCPN/OpenCPN #1708: Plugin Manager: WeatherRouting  Ubuntu  url misdirect

rgleason, 2020-02-10:
This is where the xml is currently pointed:

https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names//versions/1.13.3.1+177.e93a58d/opencpn-plugin-weather_routing-1.13.3.1-ov50-1.16_ubuntu-16.04.tar.gz
  </tarball-url>

This is the correct URL
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-ubuntu-16.04-tarball/versions/1.13.3.1+177.e93a58d/opencpn-plugin-weather_routing-1.13.3.1-ov50-1.16_ubuntu-16.04.tar.gz

1. Using "...//versions/1.13.3.1+177.e93a58d/opencpn-plugin-weather_routing-1.13.3.1-ov50-1.16_ubuntu-16.04.tar.gz

2. Should be  "../weather_routing-ubuntu-16.04-tarball/versions/1.13.3.1+177.e93a58d/opencpn-plugin-weather_routing-1.13.3.1-ov50-1.16_ubuntu-16.04.tar.gz"

Temporarily I will fix it in the xml file, but it should be fixed in the plugin.

bdbcat, 2020-02-11:
Rick...
Serious question:
Who will be responsible for this plugin at Production release of OCPN with managed Plugins?
It is that person who should fix it.
Dave

rgleason, 2020-02-11:
So it is a problem with the plugin configuration, for sure?

I expect it will be me... so I have to fix it. Jon has been helping a lot however.
Also I have intended to use this format as a kind of template for Sean's other plugins. 
I guess I'll just keep plugging away, but won't be able to do anything more for about a week, as I won't have this laptop.  Do you want this closed now?

bdbcat, 2020-02-11:
Rick...
The problem is with one of the scripts in the ci directory, probably.

FYI, I took a sightly different approach for oeRNC and oeSENC plugins.  Somewhat simpler, for my mind.  Less chasing around variable names from one script to another.
But to each his own.

Also, I found that I needed to use travis/docker to build for raspberry Pi.  So we need all three ci build-bots.  You will want to build for these two Pi platforms (stretch and buster).  Feel free to look at github oeRNC branch "ci" to see how I did it.

And you may leave this issue open, as a reminder.

rgleason, 2020-02-11:
Thanks, I will look at the oesenc way too. 
I've been into ci and I think the problem is likely in circleci-upload.sh.in

bdbcat, 2020-02-11:
Rick...
Please dis-regard oeSENC at the moment.  I have not finished updating that one yet.  Concentrating on getting oeRNC right, then will transfer to oeSENC.

rgleason, 2020-02-11:
Ok. no problem. 
plugin.xml.in    has  
`<tarball-url>    ${PKG_URL}   </tarball-url>
`Maybe  PKG_URL is twisted instead?
cmakelists.txt has 
`set(PKG_URL "https://dl.cloudsmith.io/public/--pkg_repo--/raw/names/--name--/versions/--version--/--filename--")
`

rgleason, 2020-02-11:
Ok the problem is that /--name--/ is not appearing, IE   "weather_routing-ubuntu-16.04-tarball"
Some format issue I think.

rgleason, 2020-02-22:
A similar malformed "tarball-url" problem is occurring with Windows and the other 8 environments.

rgleason, 2020-02-22:
CMakeLists.txt
Line 100   set(PKG_URL "https://dl.cloudsmith.io/public/--pkg_repo--/raw/names/**--name--**/versions/--version--/--filename--")

XML Tarball-url Fail:
..public/rick-gleason/opencpn-plugins-pkg/raw/names/**(--name-- is missing)**/versions/v1.13.5.0/weather_routing_pi-1.13.5.0-ov50-1.16_mingw-10.tar.gz

XML Tarball-url Correct:
..public/rick-gleason/opencpn-plugins-prod/raw/names/**weather_routing-mingw-10-tarball**/versions/v1.13.5.0/weather_routing_pi-1.13.5.0-ov50-1.16_mingw-10.tar.gz

rgleason, 2020-02-23:
Search of  --name-- in weather_routiing_pi

C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\appveyor-upload.sh
Line 67: line=${line//--name--/$tarball_name}

C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\circleci-upload.sh
Line 74: sudo sed -i -e "s|--name--|$COMMON_NAME|" $xml

C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\weatherrouting-plugin-msvc-10.0.18363.xml 
Line 20:  https://dl.cloudsmith.io/public/--pkg_repo--/raw/names/--name--/versions/--version--/--filename--

C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\weather_routing-plugin-msvc-10.0.18363.xml 
Line 20:  https://dl.cloudsmith.io/public/--pkg_repo--/raw/names/--name--/versions/--version--/--filename--

C:\Users\Rick\Documents\GitHub\weather_routing_pi\cmake\in-files\appveyor-upload.sh.in
Line 67:  line=${line//--name--/$tarball_name}

C:\Users\Rick\Documents\GitHub\weather_routing_pi\cmake\in-files\circleci-upload.sh.in 
Line 74: sudo sed -i -e "s|--name--|$COMMON_NAME|" $xml

C:\Users\Rick\Documents\GitHub\weather_routing_pi\CMakeLists.txt (1 hit)
Line 100: set(PKG_URL "https://dl.cloudsmith.io/public/--pkg_repo--/raw/names/--name--/versions/--version--/--filename--")

rgleason, 2020-02-23:
When I build weather_routing locally the xml file is left in the build directory as
`weather_routing-plugin-msvc-10.0.18363.xml `

with an uncompleted line for tarball-url

```
<tarball-url>
    https://dl.cloudsmith.io/public/--pkg_repo--/raw/names/--name--/versions/--version--/--filename--
  </tarball-url>
```

What mechanizm completes the line and when does this occur?
Because the "--name--" does not work, it just ends up blank.

rgleason, 2020-02-24:
Dave, 
Thank you!! This is much better, just one flatpak problem. 

I've manually checked 8 environment's for  Tarball-Url's generated in the corresponding xml files, compared to the actual cloudsmith tarball urls and they are identical except for flatpak

Notice the difference in   "-tarball"  vs ".gz"  in Bold below

```
    <target>flatpak</target>
  <target-version>18.08</target-version>
  <tarball-url>
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-flatpak-18.08**_-tarball_**/versions/1.13.5.0+228.3790b25/weather_routing_pi-1.13.5.0-ov50-1.16_flatpak-18.08.tar.gz
  </tarball-url>

``` 
When the Actual Link: 
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-flatpak-18.08.**_gz_**/versions/1.13.5.0+228.3790b25/weather_routing_pi-1.13.5.0-ov50-1.16_flatpak-18.08.tar.gz
. 

I was missing MacOS after three retries in circleci it uploaded to cloudsmith successfully. 

I guess the next step would be to increment the version, collect the xmls and copy to my opencpn/plugins beta branch and make a pr.

bdbcat, 2020-02-24:
Rick...
1.  I see no apparent problem.  Could be some sync glitch.  Very similar code in my plugins works correctly.  Please push a change, and trigger a new circleci build.  See if it is different next time.
2.  Please close this issue.  This issue is not directly concerning opencpn.  It is with your plugin tooling, specifically.  Continue with direct email to me if you need help. 

Dave

rgleason, 2020-02-24:
Thank you. Closing

closed 2020-02-24 by rgleason