OpenCPN/OpenCPN #1709: Plugin Manager: Ubuntu - Alpha and Custom catalogs not working

rgleason, 2020-02-10:
Under Ubuntu the Alpha catalog does not appear,

[image]

 nor does the Custom URL
for example try
https://raw.githubusercontent.com/rgleason/plugins/master/ocpn-plugins.xml

[image]

rgleason, 2020-02-11:
I wonder if the alpha catalog is actually shown, but the meta-url to my repository rgleason/rg-alpha branch  (for sean's plugins) is not working in Ubuntu?

bdbcat, 2020-02-16:
Rick...
Alpha catalog seems to work for me.  Today i get an offer to install WeatherRouting 1,13,3, and it installs just fine.
More info?
Dave

rgleason, 2020-02-16:
That's good. Let me give it it another try when I get back to laptop & virtualbox linux

rgleason, 2020-02-22:
Alpha Catalog is working now.  Ubuntu 18.04 and Window10
Custom URL seems to be problematic still.
Closing however.

closed 2020-02-22 by rgleason