OpenCPN/OpenCPN #1712: Compiling with cmake flag -DOCPN_BUNDLE_GSHHS=FULL cause error

FredericGuilbault, 2020-02-11:
In a chrooted qemu environment on arch arm64 if im running the following build process: 

```
apt-get install -y -q build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev wx-common libwxgtk3.0-dev libglu1-mesa-dev libgtk2.0-dev wx3.0-headers libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev
	
git clone  --depth 1 git://github.com/OpenCPN/OpenCPN.git
mkdir -p ./OpenCPN/build
pushd ./OpenCPN/build
	cmake  -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=FULL ../
	make
	make install
popd
rm -r ./OpenCPN
```
Im getting the following error message. 

```
[....3 hours of logs....]
/usr/bin/cmake -E cmake_progress_start /lysmarine/OpenCPN/build/CMakeFiles 0
make -f CMakeFiles/Makefile2 preinstall
make[1]: Entering directory '/lysmarine/OpenCPN/build'
make[1]: Nothing to be done for 'preinstall'.
make[1]: Leaving directory '/lysmarine/OpenCPN/build'
Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/share/opencpn/license.txt
-- Installing: /usr/local/bin/opencpn
-- Installing: /usr/local/share/opencpn/LINUX_DEVICES.md
-- Installing: /usr/local/share/opencpn/ocpn-plugins.xml
-- Installing: /usr/local/share/opencpn/gshhs/poly-c-1.dat
-- Installing: /usr/local/share/opencpn/gshhs/wdb_borders_c.b
-- Installing: /usr/local/share/opencpn/gshhs/wdb_rivers_c.b
CMake Error at cmake_install.cmake:69 (file):
file INSTALL cannot find "/lysmarine/OpenCPN/data/gshhs/poly-l-1.dat".
```
---
From what I found with grep.
```
OpenCPN/CMakeLists.txt:81:  - BUNDLE_GSHHS (NONE,MIN,[CRUDE],LOW,INTERMEDIATE,HIGH,FULL) governs the
```
`FULL` looks to be a valid entry

leamas, 2020-02-11:
This looks pretty bad. Looking in top-level CMakeLists.txt we find

```
set(gshhs_MIN data/gshhs/poly-c-1.dat)

set(
  gshhs_CRUDE
  ${gshhs_MIN}
  data/gshhs/wdb_borders_c.b
  data/gshhs/wdb_rivers_c.b
)

set(
  gshhs_LOW
  ${gshhs_CRUDE}
  data/gshhs/poly-l-1.dat
  data/gshhs/wdb_borders_l.b
  data/gshhs/wdb_rivers_l.b
)

set(
  gshhs_INTERMEDIATE
  ${gshhs_LOW}
  data/gshhs/poly-i-1.dat
  data/gshhs/wdb_borders_i.b
  data/gshhs/wdb_rivers_i.b
)

set(
  gshhs_HIGH
  ${gshhs_INTERMEDIATE}
  data/gshhs/poly-h-1.dat
  data/gshhs/wdb_borders_h.b
  data/gshhs/wdb_rivers_h.b
)

set(
  gshhs_FULL
  ${gshhs_HIGH}
  data/gshhs/poly-f-1.dat
  data/gshhs/wdb_borders_f.b
  data/gshhs/wdb_rivers_f.b
)

```
However, the only files found in data/gshss is:

```
$ ls data/gshhs/
poly-c-1.dat  wdb_borders_c.b  wdb_rivers_c.b
```

Ergo: all sorts of files seems to be gone. They can be found in the history:

```
$ git checkout opencpn-3.3.719
$ ls data/gshhs/ | more
poly-c-1.dat
poly-i-1.dat
poly-l-1.dat
wdb_borders_c.b
wdb_borders_i.b
wdb_borders_l.b
wdb_rivers_c.b
wdb_rivers_i.b
wdb_rivers_l.b
```

Can they be used? 
@bdbcat: @nohal: Thoughts?

nohal, 2020-02-11:
You need to download the data if you want to bundle it. The latest links are available from the respective chart downloader catalog at https://github.com/chartcatalogs/catalogs/blob/master/GSHHG_Catalog.xml

bdbcat, 2020-02-12:
At some point we decided not to burden the github repo with the full GSHHG dataset, simply due to size and weight.  We should discuss this again.
Dave

FredericGuilbault, 2020-02-12:
That make sense, those file are effectively heavy for a git repo. 

Idk anything about cmake and im not gonna do any pull request so my 2 cents don't worth much here. But stackoverflow says that cmake can download files form the internet.
```
set(MY_URL "http://...")
set(MY_DOWNLOAD_PATH "path/to/download/to")
set(MY_EXTRACTED_FILE "path/to/extracted/file")

if (NOT EXISTS "${MY_DOWNLOAD_PATH}")
    file(DOWNLOAD "${MY_URL}" "${MY_DOWNLOAD_PATH}")
endif()

add_custom_command(
    OUTPUT "${MY_EXTRACTED_FILE}"
    COMMAND command to unpack
    DEPENDS "${MY_DOWNLOAD_PATH}")
```

leamas, 2020-02-12:
> At some point we decided not to burden the github repo with the full GSHHG dataset

OTOH, it's seemingly still part of the history.  Which means that if you clone the current repo you will still download these files. It's just a question of checking out  a tag to make them visible.

For distributed source tarballs it would IMHO make a lot of sense to drop these files. But in the git repo, I only see two consistent options:

  - Removing the dataset completely, also from the history. This is complicated since it will affect published tags. But it will make the git repo smaller.
  - Re-adding them again. From the git repo size perspective this should actually not change anything.

Hakansv, 2020-02-12:
The small GSHHG set was as Dave says adaption for small system. Today even a RPi3 often use at least 16 GB card so 0.05 GB more in the package make no difference. Most "common" users don't use the chartdownloader to get the full set so an already implemented fully detailed GSHHG will for sure satisfy newcomers. And one moment less on our training sessions. :)

leamas, 2020-02-12:
Some more datapoints, expanding on the git repo size:
```
$ git clone --bare OpenCPN opencpn.git
$ git -C OpenCPN archive --format tar HEAD > opencpn.tar
$ du -sh opencpn.git/ opencpn.tar
347M	opencpn.git/
173M	opencpn.tar

```

That is, the git database which already contains the GSHHS dataset is about 350Mb whereas the actual, visible sources are about 170Mb. One could argue that the added 50Mb to the actual sources is problem; I don't, though.

Note that this is about the sources. What's actually installed is basically another issue -- even if we re-add the gshhs data it's still possible to  make a minimum installation.

leamas, 2020-02-26:
Filed a PR (#1740, above)

Not really sure that this is the way to go. It adds 734 MB. One alternative would be to:

  - Bundle everything besides the "full" files, about 20MB.
  - Detect missing files when doing  -DOCPN_BUNDLE_GSHHS=FULL, and have
    feedback + tooling in place to download these as required.

Thoughts?

transmitterdan, 2020-02-27:
We should not do something "crazy" in the interest of consistency.  The documentation for builders can describe the error message and where to get the needed data files.  A clever cmake programmer could issue an appropriate message which is always nice.

I don't think we should have cmake launch a nearly 1GB download automagically.

Personally, I think the "best" option is to change cmake file to only allow the option of bundling the CRUDE data.  If a user wants the higher resolution data sets they are always able to install them either manually or via the downloader.

leamas, 2020-02-27:
>  don't think we should have cmake launch a nearly 1GB download automagically.

That was a typo, fixed above. We are talking about 73Mb, not 1GB. However, the conclusion could be the same.

> Personally, I think the "best" option is to change cmake file to only allow the option of bundling the CRUDE data. If a user wants the higher resolution data sets they are always able to install them either manually or via the downloader.

Right. Couldn't we then just drop option completely? Most users have charts covering the area they are sailing in, but needs the gshhs data to zoom in on it when the cursor for whatever reason is placed somewhere else. The crude data covers that purpose, and I see no reason why it shouldn't be bundled.
 
> If a user wants the higher resolution data sets they are always able to install them either manually or via the downloader.

Agreed.

leamas, 2020-03-01:
Filed a new PR (above) which handles the basic problem by simply removing the  OCPN_BUNDLE_GSHHS option instead.

Thoughts?

Hakansv, 2020-03-01:
As long Crude is safely installed I've no immediate doubts.

rgleason, 2020-03-01:
I think it is ok to simply load Crude to start and let the user improve it to high res lwith ChartDownloader which is more familiar and easy.

leamas, 2020-03-01:
> As long Crude is safely installed 

It is

closed 2020-03-01 by bdbcat

free-x, 2020-03-03:
this breaks compatibility when updating from previous packages

leamas, 2020-03-03:
This is  a problem in the debian packaging, probably best handled in a separate bug. THe generated package probably needs some Replace:/Breaks: stannzas, see https://www.debian.org/doc/debian-policy/ch-relationships.html#s-replaces

PR #1746 (2020-03-01): CMakeLists: Drop OCPN_BUNDLE_GSHHS option (#1712).
Always bundle the crude GSHHS data -- an opencpn installation
without this would have big problems zooming in to installed
chart sets.  Higher resolutions are available using the
downloader, and there is no need bundle these in build. 

Closes: #1712