OpenCPN/OpenCPN #1713: Direction to cursor in statusbar.

tai-fun, 2020-02-11:


tai-fun, 2020-02-11:
The direction to cursor in the status-bar is out 180 degrees.
[image]
[image]

Hakansv, 2020-02-11:
So, you mean we want bearing from boat to cursor not bearing from cursor to boat?
Of course easy fix if all agree.

tai-fun, 2020-02-11:
Yes, the same way it's now in 5.0. Compare the screen dumps.

Hakansv, 2020-02-12:
I see the difference between v 5.0 and present master. It's easy to turn the status bar print 180° but I wonder why it has changed. Would be some other change for any reason. Anyone remember?

tai-fun, 2020-02-12:
I noticed this first, when testing the AIS-Radar plugin. As i is now,
everything is really back to front.
Obviously this plugin is dependent on the "old" way.

Hakansv, 2020-02-12:
It may be this commit introduced the change:
0d0c19ec90d037820be78ae5a1899d9eb850bd47
> Prevent nan return from /src/georef.cpp ll_gc_ll_reverse(..) and correct
>     DistanceBearingMercator (was giving a bearing 180 degr off)

This commit was for sure introduced for a good reason but may have had side effects?
@RooieDirk  may know more and can judge my thoughts. 
It's of course easy to turn only the status bar but I haven't checked where AIS radar plugin read.

RooieDirk, 2020-02-12:
Yes I do. There is already a correction puss request posted, [I](https://github.com/OpenCPN/OpenCPN/pull/1668) altough I see now it needs a rebase.

RooieDirk, 2020-02-12:
New pull-request posted.

Hakansv, 2020-02-12:
Very good! This is a fantastic community.
Thanks.

closed 2020-03-13 by tai-fun