OpenCPN/OpenCPN #1715: Managed Plugins for next OpenCPN Beta Test program

bdbcat, 2020-02-16:
It is time to start distilling the list of managed plugins planned for the initial Beta Release of the next version of OCPN.

Here is my logic:
1.  Some plugins are no longer under active development (e.g.Logbook Konni), yet are strangely popular.  I suggest that existing Legacy plugins are OK for this class.  Call these "Class-1"
2.  Other plugins are in varying stages of active development.  It is this class that we should attempt to migrate to managed status first.  Call these "Class-2"

I propose the following plugins be designated as Class-2, and should be targeted for initial Release:

1.  oeSENC_PI (responsible dev:  bdbcat)
2.  oeRNC_PI (responsible dev:  bdbcat)
3.  Radar_pi ( Responsible dev:  Hakan )
4.  WeatherRouting_pi  (Responsible dev:  Rick)
5.  OCPN_DRAW  (responsible dev:  ???)
6.  VDR_PI (responsible dev:  bdbcat)

I base my list on the perceived interest in these plugins on CF, as well as their recent development activity.  There seems no reason to expend the dev energy to upgrade an old or less popular plugin to Class-2, if it works fine as-is.

All others are in Class-1, and Legacy plugins will suffice.

I note that we have not changed the wxWidgets binary API.  Thus, all Legacy plugins will install as normal on the OCPN core Beta, regardless of their stated API requirements. 

I invite comments.
Specifically, if anyone has strong opinions about any other plugins that should immediately be promoted from Class-1 to Class-2, bring them on.

Dave

rgleason, 2020-02-16:
Jon Gough is the dev for Ocpn_draw. I think he is waiting until Plugin Manager code settles.

I would like to eventually add some other Sean plugins like weatherfax, watchdog, celestial_nav, climatology and maybe rotationctrl, projections, findit and plots.

However I need to confirm that everything works with weather_routing first, for all OS, installing properly, etc. I think I can test for Win and Linux Ubuntu, but not MacOS or RPI, or flatpak.

bdbcat, 2020-02-16:
@jongough ...
The current builds of OCPN support managed plugins adequately, though there is more work to do.
If you would like a quick migration path, take a look at oeRNC.  Create a new branch called "ci" Add the circleci, appveyor, and travis scripts.  Adapt slightly to your cloudsmith repo.  Push the commits to "ci", and the builds start automatically.
When all is well, submit a PR to my "plugins" project, on branch "Beta".  This PR will contain the metadata XML files for the various platforms( 10, the the moment).  I will merge immediately, and all will be ready to go.

bdbcat, 2020-02-16:
@rgleason 
I am waiting for PRs against plugins Beta branch, especially for WeatherRouting.  I want to move away from Alpha and custom as soon as most of the platforms are ready, in order to encourage managable testing.

rgleason, 2020-02-16:
Ok, I looked at  https://github.com/bdbcat/oernc_pi  and it does not seem to be building, is that because the ci folder is missing?

jongough, 2020-02-16:
I have created a pull request for 'testplugin_pi' build xml files on the beta branch. This is the plugin I have been using to help build the generic versions of the cmake file set to act as a base for all other plugins to use. So I would rather try this one out and get it working before changing ODraw to use the new process as this protects the users from any mistakes.

jongough, 2020-02-16:
How are the Raspberry PI builds done as I cannot find the process in the oesenc_pi plugin tree?

bdbcat, 2020-02-17:
@rgleason 
Your list of "eventually" plugins sounds OK, but I see no reason for them to be available to start OCPN Beta.  I would expect them to magically appear in the catalogs over the next few months. Am I missing something?

bdbcat, 2020-02-17:
Rick...
on oeRNC, all the CI build stuff is in branch "ci".  I retain the "master" branch for building unmanaged legacy plugins.

bdbcat, 2020-02-17:
Jon...
Raspberry Pi builds are done by Travis-ci.  Scripts are in branch "ci", same organization as for oeRNC.  See .travis.yml

bdbcat, 2020-02-17:
Jon...
I accepted your PR.
But it is incomplete.  You also need to make the new "ocpn-plugins.xml" file, which is the merged results of everything in metadata directory.

I run like this:

`python tools/ocpn-metadata generate --userdir metadata --destfile ocpn-plugins.xml --force
`
Then commit the resulting ocpn-plugins.xml, along with the 8 individual platform metadata files.

jongough, 2020-02-17:
Just submitted a patch with the updated xml file.

bdbcat, 2020-02-17:
Jon...
Got it.  You should now be able to update the Beta catalog in OCPN, and see your work in action.
Dave

rgleason, 2020-02-17:
Dave, can do this thurs.

jongough, 2020-02-17:
I have tried with the latest version of OpenCPN, master branch commit ID:3842f2c, but I the plugin manager does not show anyway to install new plugins even though I have updated the 'beta' stream. 

The plugin manager shows:
[image]

In fact this version seems to terminate some of the time with little information about why.
```
Unhandled unknown exception; terminating the application. I get in the debug output:
../src/common/wincmn.cpp(477): assert "GetEventHandler() == this" failed in ~wxWindowBase(): any pushed event handlers must have been removed
*** Exited with return code: 0377 ***
```
Should I be using the master branch for this testing or is there a different branch that has the requisite code in it?

bdbcat, 2020-02-17:
Jon...
Master branch of OCPN is current.
You need to make a PR to the plugins project, including the Draw Plugin, just as you did with test plugin.

jongough, 2020-02-17:
I am trying to use the testplugin_pi as I have not done anything with ODraw (and am waiting for things to settle down before doing that piece of work). As you can see there is no test plugin showing up. Should I have modified anything else to make this work? And I am using the current Master branch.

jongough, 2020-02-17:
One thing that has affected this in the past is that I use Linux Mint, a ubuntu derivative, and this has not shown available plugins. I thought this issue was solved, but......

jongough, 2020-02-17:
It would appear that the testing for compatible versions of ubuntu does not work. I am using Linux Mint 19.3, I have created the environment variable, OPENCPN_COMPAT_TARGET, and set it to "ubuntu', but now the process is failing on the version number. Within Linux Mint there is a file, /etc/os-release, which could be used to work all this out. The contents of this file is:
```
NAME="Linux Mint"
VERSION="19.3 (Tricia)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.3"
VERSION_ID="19.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tricia
UBUNTU_CODENAME=bionic
```

leamas, 2020-02-17:
Besides the environment variable there are also  configuration variables CompatOS and CompatOsVersion. They have as yet no UI, but can be set manually in the opencpn.conf/.ini. 

Admittedly not user-friendly, but could become with a GUI.

/etc/os-release is probably not the right tool. It's not generally available, and lacks the crucial info on what ubuntu version it matches.

leamas, 2020-02-17:
> , I have created the environment variable, OPENCPN_COMPAT_TARGET, and set it to "ubuntu'

The correct, undocumented way to use it is like _ubuntu:16.04_. This should resolve the version issues.

bdbcat, 2020-02-17:
Alec...
re:
"/etc/os-release is probably not the right tool. It's not generally available, and lacks the crucial info on what ubuntu version it matches."

Did you mean "lsb-release"?
It seems that parsing /etc/os-release will give us what we need, at least for Ubuntu and Mint.
What am I missing?

Dave

bdbcat, 2020-02-17:
jon...
Just to be clear:
Are you finding that your Mint installation will run plugins built for Ubuntu 18.04?

leamas, 2020-02-17:
> t seems that parsing /etc/os-release will give us what we need, at least for Ubuntu and Mint.

Not really even for Mint. The os version is not available. What we have is the codename, but it's not really what we need. And maintaining tables for codename -> version is cumbersome.

I think we should first of all design a generic mechanism which could be used on any os to load plugins for another os:version combination. With this in place we could look into specific cases like Mint. It could be in a GUI with pre-loaded values based on what we could get from /etc/os-release, lsb_release or so. 

But the underpinnings should IMHO be generic rather then for example Mint-specific fixes.

bdbcat, 2020-02-17:
Alec...
IMHO, this system detection should be done at run-time, not at build-time.
Easy enough to add some methods to OCPNPlatform to identify the system more exactly.  Then the pluginhandler can handle all the special cases as required.

In /etc/os-release, I see the string "VERSION_ID=blahblah".  Seems clear to me.

leamas, 2020-02-17:
> IMHO, this system detection should be done at run-time, not at build-time.

We are basically doing a great job for the build-time defaults. There are corner-cases such as mint and what not (testing comes to mind) -- no less and no more.

> in /etc/os-release, I see the string "VERSION_ID=blahblah". Seems clear to me

However, this is the mint version. What we need (in this case) is the Ubuntu version which corresponds to this...

bdbcat, 2020-02-17:
```
dsr@dsr-HPE-580t:~/Projects/opencpn$ cat /etc/os-release                                                                                                                        
NAME="Ubuntu"                                                                                                                                                                   
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

```
I'll build something for run-time.  We need this to distinguish between X86_64, and ARMHF, for example.  This could also be done at build time, but I think better at run-time.

leamas, 2020-02-17:
As long as these values are overridden by OPENCPN_COMPAT_TARGET and CompatOS/CompatOSVersion if set I'm fine. But I still think we need these as last resort (testing, failed/missing runtime checks)

bdbcat, 2020-02-17:
Alec...
Run-time overrides are fine.
btw, what says /etc/os-release for Fedora?

leamas, 2020-02-17:
```
NAME=Fedora
VERSION="31 (Workstation Edition)"
ID=fedora
VERSION_ID=31
VERSION_CODENAME=""
PLATFORM_ID="platform:f31"

```

This is just the first few lines. That said, I think we should focus on getting flatpak in shape. This will work on both Fedora and other non-debian distros. 

BTW: could you possibly check https://github.com/OpenCPN/plugins/issues/25 ? I'm sort of blocked on this...

jongough, 2020-02-18:
Dave,
   On my Linux Mint 19.3 development machine I can get the plugins installed, and with the change to `OPENCPN_COMPAT_TARGET` or opencpn.conf `CompatOS` and `CompatOsVersion` and can see it to activate. But as soon as I activate it OCPN crashes. If I try to install the 'deb' version rather than the 'tar' version I get an error during the install process which complains about the version level of certain wxWidgets libraries. I am currently using version 3.1 of widgets but the circleci build is using 3.0.x (as required by the Plugin Manger).

   So, if I go to a non-development VM version of Linux Mint 19.2, without OCPN installed and do a apt install of the latest (1.0.43) testplugin_pi it will install OCPN from the normal repositories. This works fine. I don't have a version of OCPN with the latest changes that has been built by circleci (is there one) that I can then test the tarball install.
> jon...
> Just to be clear:
> Are you finding that your Mint installation will run plugins built for Ubuntu 18.04?

jongough, 2020-02-18:
Be careful about calling Linux Mint a 'corner case'. There are many other distributions based on Ubuntu and Debian which a user may be using and which will have similar issues to the one I am having. The user should not be required to know any of this stuff. For an 'alpha' or possibly a limited 'beta' test we can work around the issues. But they do need to be fixed before normal, non-developer, users try to use this process. Debugging of Plugin Manager is not for the feint hearted!

jongough, 2020-02-18:
Can you point me at circleci built, cloudsmith downloadable version of the current OCPN? I can then use this in a vanilla VM to try out installations and runtimes. Otherwise I will have to build another development environment.

bdbcat, 2020-02-18:
Jon...
We do not have continuous deployment setup for OCPN base builds.  Just have not needed it yet, since most devs are able to build with wxWidgets 3.0.

I would invest the effort to get your Mint 19.3 to build OCPN from github source, using wx3.0.  The OCPN master branch is, right now, a fast moving target, and you may like to keep up with local builds.
The way I would do this is to download the wx3.0 sources, build wx3.0 as documented, and configure OCPN build to use the wx3.0 libraries and includes just built.  I have done this many times without trouble.

bdbcat, 2020-02-18:
Regarding Mint:
I am now working on some code that will allow "ubuntu-like" distros to find the correct managed plugin.  Should hit github master tonight, probably.  With this patch, the config file overrides of CompatOS and CompatOsVersion will not be necessary.

jongough, 2020-02-18:
I am creating a new VM for building OCPN for this case so I should be able to get a runnable version using wx 3.0. When is the plan to allow wx 3.1 as I thought that was happening with OCPN 5.x.x?

bdbcat, 2020-02-18:
At the moment, very few linux distros are including wx3.1.
We use privately built wx3.1 on Windows and Mac, since we have to package the wx libraries with the download, anyway.  We really prefer not to include "custom" wx3.1 libraries in our linux packages.

leamas, 2020-02-24:
> At the moment, very few linux distros are including wx3.1.

I doubt any distribution will include 3.1, which by definition havn't a stable ABI. They are waiting for the stable 3.2.

leamas, 2020-03-02:
One problem: The current Bionic plugins on cloudsmith are gtk2. This means that they are incompatible with the Debian package as well as some other plugins built for Bionic (citation needed).  

We basically need to add a gtk3 build chain, and define how to handle this when displaying available plugins and download/installation. A new target ubuntu-gtk3?

See also: #1536.

Thoughts?

bdbcat, 2020-03-02:
Would not the new target need to be:
ubuntu-18.04-gtk3?
And then next month we will need ubuntu-20.04-gtk3, etc..

leamas, 2020-03-02:
There is target and target-version. In this case target-version is the same, 18.04, so we need different targets like ubuntu-gtk3

leamas, 2020-03-02:
In any case we need to define target and target-version in a way which allows us to create a usable tarball name, so using a hyphen like in  ubuntu-gtk3 might be a bad idea. 

But before diving into details: we need a new, separate build line for bionic/gtk3. OK?

bdbcat, 2020-03-02:
Maybe....  Too early here....
But back to the bigger question:
What plugin in cloudsmith today depends on a specific gtk version?

leamas, 2020-03-02:
Being compatible with the official Debian package, which is gtk3.  There is already a downstream bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=948702.

leamas, 2020-03-02:
> What plugin in cloudsmith today depends on a specific gtk version?

Sorry, too quick again. All plugins depends on gtk2 today, they have  runtime deps like described in https://github.com/OpenCPN/OpenCPN/issues/1536#issuecomment-570183401. Yes, they are source compatible, but not ABI-compatible

bdbcat, 2020-03-02:
Alec...
Let's be precise.
The plugins all depend on wx3.x, which on some platforms depends on some version of gtk.  The plugin itself has no direct dependency on gtk.
So really, we need a build path for various libWX versions, including implicit gtk version dependency.

This will mean that the compatibility test need consider the WX version only, I think.

leamas, 2020-03-02:
> So really, we need a build path for various libWX versions, including implicit gtk version dependenc

Right, my bad.

leamas, 2020-03-08:
Filed a new PR #1759.  This adds a wxgtk3-based  bionic build. This is the transition branch. From upcoming debian bullseye/ubuntu focal only plugins built against wxgtk3 will work since wxgtk2 is dropped..

Focal is due April, 23.

Next step in this is adding wxgtk3 -based builds to the plugins.

leamas, 2020-03-08:
Filed a new PR https://github.com/bdbcat/oesenc_pi/pull/70 against oesenc_pi, first bionic-gtk3 builds.  I have focused to get it up & running, being the last missing part.  Will try to do some basic testing when the first gtk3 builds shows up in the catalog. 

Until then, I could need some rest...

leamas, 2021-04-15:
time to close?

closed 2021-04-29 by bdbcat