OpenCPN/OpenCPN #1718: Plugin Manager - Not displaying plugins when using a Ubuntu derivative

jongough, 2020-02-17:
I use Linux Mint 19.3 to build and test OCPN. When trying to use the new plugin manager no plugins show up that are included in the new process. This is due to Linux Mint not being recognised correctly. There is a file, `/etc/os-release`, which could be used to ascertain which version of Ubuntu the Linux Mint environment is based on. The contents of this file are:
```
NAME="Linux Mint"
VERSION="19.3 (Tricia)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.3"
VERSION_ID="19.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tricia
UBUNTU_CODENAME=bionic
```

The current code also relies on a new environment variable, `OPENCPN_COMPAT_TARGET` being set to `ubuntu`. This is not user friendly as most users would know nothing about this or how to set it.

bdbcat, 2020-02-18:
Jon...
When able, you might like to try building OCPN from github master now.  Hopefully the improved system detect logic should allow your plugins from Beta catalog to load now.

jongough, 2020-02-18:
Dave,
   Its getting close. It currently fails on line 232 in PluginHandler as `return plugin_os_version == os_detail->osd_version;` returns false, `plugin_os_version: 18.04` and `os_detail->osd_version: 19.3`.

In the case of Linux Mint there is another file that can be looked at, `/etc/upstream-release/lsb-release` and this contains:
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04 LTS"
```
So knowing that if you find the installed software is linux mint you could then look in this file to get the 'Ubuntu' version that you need for the comparison.

jongough, 2020-02-18:
I have just created a pull request to handle Linux Mint upstream release number. I have tried it and it seems to work OK.

bdbcat, 2020-02-18:
Jon...
We would rather NOT add distro specific code like this, if at all possible.  It gets pretty fragile over time.
Let us think about this some more.  Ideas solicited...

jongough, 2020-02-18:
The only other way that I have found is to maintain a list of distribution names and versions.

bdbcat, 2020-03-13:
jon...
After some thought...
I accept your PR.  We can continue to load up OCPNPlatform with whatever is needed to accurately identify the details.  That is the whole point of OCPNPlatform, after all;  To isolate platform differences...

bdbcat, 2020-03-17:
Fixed now?
Dave

jongough, 2020-03-18:
For Linux Mint it is working OK. I have not tested other distributions yet, but will close this as for me it is fixed.

closed 2020-03-18 by jongough