OpenCPN/OpenCPN #1728: On Windows. The dialog frame label is not dimmed on Color Scheme toggle.

Hakansv, 2020-02-20:
Does anyone know a way to force MS-Windows to change color on the upper label of a dialog window. We would like to get away from that spot when on Dusk or Night mode. 
The OCPN window itself can be worked around and hid by the full screen mode.
See shot from a Dashboard dialog.

[image]

transmitterdan, 2020-02-21:
Create new registry entry here:
```
Computer\HKEY_CURRENT_USER\SOFTWARE\Microsoft\Windows\DWM
```

Create a new item called AccentColorInactive and assign it the value 0xff0f0f0f.

Hakansv, 2020-02-21:
Titter......
A funny individual tool to make the light in another color. Thanks Dan.
It may not be the most convenient script to use in the day, dusk, night toggle function though? :)

[image]

transmitterdan, 2020-02-21:
I think in Windows OS the individual programs cannot control the color of the "grab" bar at the top of the window.  This area is controlled exclusively by Windows.  Maybe I am wrong but that is what I learned long time ago.

Hakansv, 2020-02-21:
Sure. So wxWidgets just borrows a container and can't control more than the size and modality.

Hakansv, 2020-03-01:
Tdan's advice would be the best solution for us Win users. Your note about Windows "keep this out of any API" is likely through.
So I set it to this:
[image]
Getting a not disturbing bar for not activated windows:
[image]

closed 2020-03-01 by Hakansv

rgleason, 2020-03-02:
Made an [FAQ link to here ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#windowsdialog_frame_label_not_dimmed_on_color_scheme_toggle)so it is not lost.