OpenCPN/OpenCPN #1729: Flatpak User Experience and Training.

bdbcat, 2020-02-20:
Considering OCPN core, we agree that flatpak will be our distribution strategy for a variety of linux distros.  So it is time to get comfortable with the flatpak UX, mainly so we can respond to user Tech Support questions as they arise.

I'll  start.

On ubuntu Xenial:
Since flatpak in not in Ubuntu 16.04 repos, I must add a launchpad PPA to install the flatpak core:

```
$ sudo add-apt-repository ppa:alexlarsson/flatpak
$ sudo apt update
$ sudo apt install flatpak

```
Then I do:
```
$ flatpak remote-add --user --no-gpg-verify ocpn-beta http://opencpn.duckdns.org/opencpn-beta/website/repo/

$ flatpak install ocpn-beta org.opencpn.OpenCPN
Looking for matches…
error: The application org.opencpn.OpenCPN/x86_64/master requires the runtime org.freedesktop.Platform/x86_64/18.08 which was not found

```
What now?

tai-fun, 2020-02-20:
Dave

I'm not used to flatpak at all, but have I have used it a few times.

I have no problem with flatpak install O.

"
$ flatpak install ocpn-beta org.opencpn.OpenCPN
Looking for matches…

org.opencpn.OpenCPN permissions:
    network      pulseaudio      x11      devices      file access [1]

    [1] home

        ID                                 Arch           Branch
  Remote            Download
 1.     org.opencpn.OpenCPN                x86_64         master
  ocpn-beta         < 51.9 MB
 2.     org.opencpn.OpenCPN.Locale         x86_64         master
  ocpn-beta          < 2.3 MB (partial)

Proceed with these changes to the user installation? [Y/n]: n

"

My take is that before you try to install O you also have to run.

$ flatpak remote-add --if-not-exists flathub
https://flathub.org/repo/flathub.flatpakrepo
$ flatpak install flathub org.freedesktop.Platform//19.08
org.freedesktop.Sdk//19.08

More here https://docs.flatpak.org/en/latest/available-runtimes.html

bdbcat, 2020-02-21:
Thanks, Thomas.
So I did:
```
$ flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
$ flatpak install flathub org.freedesktop.Platform//18.08
$ flatpak install ocpn-beta org.opencpn.OpenCPN

```
and then

`$ flatpak run org.opencpn.OpenCPN
`
And there it is...

tai-fun, 2020-02-21:
Dave

Two things. Flatpak is in Bionics repro so no need for your first two steps.
The flatpaks I have installed has automatically shown up in the graphical
menus.
You should find O in Education->Science.
So flatpak behaves like a normal dep package from the  users viewpoint.

sailoog, 2020-02-21:
**Summarizing**

Installing in Ubuntu Bionics and Debian Buster:

```
$ sudo apt update
$ sudo apt install flatpak
$ sudo flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
$ sudo flatpak remote-add --user --no-gpg-verify ocpn-beta http://opencpn.duckdns.org/opencpn-beta/website/repo/
$ sudo flatpak install flathub org.freedesktop.Platform//18.08
$ sudo flatpak install ocpn-beta org.opencpn.OpenCPN
```

Running:

`flatpak run org.opencpn.OpenCPN`

**Questions**

- What is the plan? running independent instances or sharing /home/.opencpn/?
- Coexistence with old PPA installations or forced transition?

leamas, 2020-02-22:
The overall plan I have envisioned is to keep the .deb packages, adding the flatpak packages for other linux distros we don't have a working support for right now. 

The main reason to keep the .deb packages is, as I see it, raspberry'ish  systems which are somewhat constrained. Here, the smaller .deb packages makes sense. Also, since opencpn now is part of official debian it will become much easier to support opencpn  here once the debian package trickles down to actual releases.

The most important user experience part for flatpak and the official debian package is the missing plugins. We need to make plugins available in the installer to finalize these distributions,

leamas, 2020-02-22:
Installation: The basic flatpak design makes it possible to make package installation a single-click operation. We could achieve this by publishing a link to a .flatpakref file, or by just making opencpn available at http://flathub.org. 

In any case, this is not that interesting until we the most basic plugins available, see #1715.

leamas, 2020-02-22:
> What is the plan? running independent instances or sharing /home/.opencpn/?

Truth is, there is no plan. As of now, we are using a shared dir.

The rabbit hole is deeper. It's really a question of how much of the home directory which should be visible to the flatpak sandbox. As of now, the complete home dir is visible. If we dropped this, using a home dir private to the sandbox opencpn should more stable.

Still, we need at least a shared dir for the charts, I guess.

Thoughts?

bdbcat, 2020-02-22:
I think we should position OCPN Flatpak as an alternative use pattern for those distros that don't look enough like Ubuntu/debian to install from dpkg, or so.  For example, we can deprecate native Fedora (.rpm) support immediately, even though there will be power users willing to build for their own use.  These edge case distros pop up every so often, frustrating users and consuming dev resources.

I'm not too worried about exposing the entire $HOME directory.  We are not promising super-secure sandbox operation.  We are promising basic functionality.

leamas, 2020-02-22:
> I think we should position OCPN Flatpak as an alternative use pattern for those distros that don't look enough like Ubuntu/debian

Agreed

sailoog, 2020-02-24:
OpenCPN in official Debian repositories for all platforms (arm for Raspberry included), flatpak for the rest of distros, plugins directly managed by OpenCPN stored in cloudsmith and accessible to all distros sounds like a perfect ecosystem.

I do not see problem exposing the entire home directory either. It would be the most logic place for any charts folder.

If we are using an "embedded" system like flatpak we could try to keep the .opencpn folder (conf and log files) inside the flatpak sandbox and not shared. This would add the extra feature of allowing different opencpn versions in the same machine even in Debian systems. If this does not add complexity to the ecosystem we should consider it.

P.S. I am quite sure we will be able to easily adapt the new OpenPlotter 2 to any of these scenarios the same day it is released.

rgleason, 2020-02-24:
Created a page in the wiki Getting Started - Installation  [Under Development]
Feel free to edit or delete.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak

leamas, 2020-03-06:
Have added some ci configuration and a cron job which updates the ocpn-beta repo from master branch daily. Will make a PR later, after stabilizing.

leamas, 2020-03-06:
@bdbcat : in reply to https://github.com/OpenCPN/OpenCPN/issues/1729#issuecomment-589460305

>Not exactly rocket science, but maybe not exactly intuitive to a novice, either.
Needs Wiki documentation.

This is not what end users will meet. Before we release both installation and starting opencpn are simple, single-click operations.

Actually, after installation opencpn can be started from the gui just as any other app already now.

bdbcat, 2020-03-07:
Alec...
How will you implement this?
```
On ubuntu Xenial:
Since flatpak in not in Ubuntu 16.04 repos, I must add a launchpad PPA to install the flatpak core:

$ sudo add-apt-repository ppa:alexlarsson/flatpak
$ sudo apt update
$ sudo apt install flatpak

```

bdbcat, 2020-03-07:
Also,
`Before we release .. installation .. are simple, single-click operations.`
How?

leamas, 2020-03-07:
One of two ways:

  - Publishing opencpn on flathub.org
  - Iron out the details and publish  opencpn.flatpakref and (perhaps) opencpn.flatpakrepo,,
    currently generated in  flatpak/website by flatpak/Makefile These are designed to provide
    single-click installs  We might need two clicks, one extra for the flathub repo, 

See: https://flatpak-test.readthedocs.io/en/latest/hosting-a-repository.html

leamas, 2020-03-07:
> sudo add-apt-repository ppa:alexlarsson/flatpak

This cannot be automated, but is only necessary on older distros (definitely trusty, perhaps xenial, definitely not current LTS bionic)

bdbcat, 2020-03-07:
Alec...
I have not answered your private email about flathub,  Details to consider.
But I think we can say definitely that we WILL host OCPN on flathub.  So feel free to move in that direction.

And flatpak is definitely NOT in xenial distro repos..  I suppose we will find out by field experimentation where else it is not present, and handle by exception.

leamas, 2020-03-07:
> But I think we can say definitely that we WILL host OCPN on flathub.

OK. Alll things considered I agree this might be the best solution. 

I think it's a bad idea to publish it until we have the initial set of plugins available. Most users need a plugin, and not having anything available might create badwill.

leamas, 2020-03-07:
> And flatpak is definitely NOT in xenial distro repos.

We get somewhat off-topic here. For Debian/Ubuntu we have .deb packages.  The interesting stuff is other distros, notably Fedora/CentOS, OpenSUSE and Arch. For those (and many others)  there is excellent documentation at https://flatpak.org/setup/

leamas, 2020-03-07:
>  >But I think we can say definitely that we WILL host OCPN on flathub.

> OK. Alll things considered I agree this might be the best solution.

Going this way our current beta repo will stay that way, a pre-release playgound. This makes it much more viable to just use it as-is. It needs some documentation, though. Things like dns name and ownership could also be postponed.

I have a PR under way which basically reduces the maintainer role to a single encryption key. I would still like to share it, but this could be discussed in private after handling this PR.

rgleason, 2020-03-07:
Flatpak is a repository. I hope this is not another repository that Plugin Devs need to know about!
I've been working on the wiki PI Manager Dev Setup and have been simplifying things, but this is pretty complex really just for a Plugin Dev to publish.

(I do not have access to that page now to provide a link, because every 10 or so edits, the wiki page disappears, and now I have to wait just 2 hours or so before it reappears, rather than 2-3 days.) Pretty annoying though.

bdbcat, 2020-03-07:
Rick...
Not to worry (much).
The current plugin build process as you have implemented for WeatherRouting builds a flatpak-compatible plugin directly, hosted on cloudsmith.  This plugin may be installed into a flatpak instance of OCPN just like any other "managed" plugin.
As Alec points out, we must move forward on adapting older plugins to the "managed" scheme as expeditiously as possible.  To that end, I appreciate your Wiki work to document the complex process. Thanks.

Dave

leamas, 2020-03-07:
I actually explored the path to build "real" flatpak plugins, generating flatpak artifacts. However, I dropped this, it was just too complicated.

So the current setup is, as Dave says,  to just build a simple tarball with some specific layout. I don't foresee that plugin devs have to care about the flatpak repo housing the opencpn main build. 

It's perfectly fine as long as the tarball layout matches the spec (and the spec is right...)

rgleason, 2020-03-07:
Dave, working on vdr_pi.  Almost have it I think. Compiles on Windows local now. Need to get creds working for the build part.  Now trying to build in Ubuntu. ...learning.

rgleason, 2020-03-08:
Good call Alec.  

More precise wiki now  [PI Manager Dev Setup](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installler_build_deploy). Comments welcome.

https://github.com/rgleason/vdr_pi/commits/pkg-pavel-ci
2 Travis complete.
5 Circleci complete.

2 Errors
Does Flatpak only build the Master branch?
Why would MacOS not see  	vdr_pi/buildosx/InstallOSX/plugin_pkgproj.in
when it does exist!  Is it permissions?

leamas, 2020-03-08:
> Does Flatpak only build the Master branch?

It's a setting in flatpak/*.yaml;  the `branch`: line should be easy to spot. Here is also the decision for the code used  (in `url:` ) -- you might want to compare to the setup in the main OpenCPN/flatpak  directory.

> Sorry for thread contamination

It might make sense to open an issue for plugin developer info. I see no way to add comments to that wiki, it is closed.

BTW: Why is it closed? It sort of defeats the idea of a wiki. If wikipedia can be open, why not this? Perhaps best discussed in a new thread.

rgleason, 2020-03-08:
> It might make sense to open an issue for plugin developer info. I see no way to add comments to that wiki, it is closed.

I thought you had access to the wiki, but you  aren't in the user list. Have sent you an email with access pw

> BTW: Why is it closed? It sort of defeats the idea of a wiki. If wikipedia can be open, why not this? Perhaps best discussed in a new thread.

Hackers hacked, many other wikis have found they have to restrict too.  We limit edit access to known community.  Had this discussion before.

rgleason, 2020-03-08:
What is this url?

> Here is also the decision for the code used (in url: ) -- you might want to compare to the setup in the main OpenCPN/flatpak directory.

leamas, 2020-03-08:
The "what" in your question implies that you found it, right? In the last part of flatpak/*yaml.

This url is the url of the git repo used -- the flatpak build always starts by cloning a repo. The default `..`  means that flatpak will clone the directory above the flatpak dir i. e., the OpenCPN sources.  However, you can put any url to a git repo here, and a corresponding branch:

rgleason, 2020-03-08:
Now I get it
I would add this to my yaml somewhere. 
Jon has a different schema, that does not have a flatpak dir, but I think he has this file.

```
    sources:
          - type: git
            url: ..    <---Right here.  So I could put  https://github.com/rgleason/vdr_pi/tree/pkg-pavel-ci
            branch: master
          - type: patch
            path: src/0008-flatpak-Add-a-shell-wrapper.patch
```

Also found an answer to my zlib1.lib issue "not found"

============
zlib1.lib
============

 - name:  zlib
      sources:
          - type: archive
            url: http://zlib.net/zlib-1.2.11.tar.gz
            sha256: c3e5e9fdd5004dcb542feda5ee4f0ff0744628baf8ed2dd5d66f8ca1197cb1a1

Thanks!  Lets hope thiis works.

rgleason, 2020-03-08:
Some idiot put    set(GIT_REPOSITORY_BRANCH "pkging-pavel-ci")
  in the CMakeLists.txt   instead of pkg-pavel-ci

leamas, 2021-04-15:
time to close?

rgleason, 2021-04-23:
Move to "Discussions"  Show & Tell or Q&A?

closed 2022-04-07 by rgleason