OpenCPN/OpenCPN #1734: Omxplayer hangs on startup

bdbcat, 2020-02-23:
Dan...
re: commit: fac2404a72cf3a5534dd3e01f59b31c7fbe08d1f

On rPI4, raspbian buster, HEADLESS (no monitor):
Hangs at startup.  Starts fine if I revert this commit.

Dave

transmitterdan, 2020-02-23:
Ok, will investigate.  For me it worked on my RPi 4.

NB, this flag is only used by self builders. It will not be in the .deb package unless you decide so.

transmitterdan, 2020-02-23:
@bdcat,

What happens if you edit the audio player command in opencpn.conf to something other than omxplayer? That’s what the PR does. It doesn’t change any C++ code.

bdbcat, 2020-02-23:
Dan...
1.  I built with default options.  IMHO, this should build and run out-of-box.

transmitterdan, 2020-02-23:
Dave,

Agreed.

transmitterdan, 2020-02-23:
Dave,

I have tracked this down.  It seems to be a bug in the Sound library. In a few places we see this code:
```
    m_isPlaying = true;
    do_play(m_cmd.c_str(), m_path.c_str());
    m_onFinished(m_callbackData);
    m_onFinished = 0; <----CRASH!!!
    m_isPlaying = false;
```
It seems the std template library function pointer object destroys ```m_onFinished``` pointer before the function it pointed to returns.  Thus the attempt to null it out after it is already destroyed fails with a seg fault.  At least it does in Windows.  I assume it is the same result in Linux.

I suspect this is some relatively new behavior of the std template library function pointer system.  But it can't be that new so maybe we just haven't used this method of making sound for a long time.  This is the declaration of ```m_onFinished```:
```
typedef std::function<void(void* userPtr)>  AudioDoneCallback;
:
:
AudioDoneCallback m_onFinished;
```
If you agree, I will make a PR that deletes these redundant assignments of 0 to ```m_onFinished``` callback function pointer.  It is found in ```MswSound.cpp```, ```OcpnWxSound.cpp```, ```SystemCmdSound.cpp```

Now, as to why this crashes in a normal no-option build on RPi I cannot fathom.  This command processor audio playout code is not supposed to be compiled unless in cmake we turn on one of the special sound library options. Specifically for Linux one would have to run cmake with option ```-D OCPN_ENABLE_SYSTEM_CMD_SOUND=ON``` to even get this code into the executable.  Perhaps you have this left over in the cmake cache?

I tried multiple times to get my RPi 4 to crash using this:
```
rm CMakeCache.txt
cmake ..
make clean
sudo make -sj4 install
```
But it would never crash.  Also, can't make any sounds either.

leamas, 2020-02-23:
Hm... I'm probably to blame, this is my code.

And you are right. Assigning 0 to the function pointer will cause all sorts of troubles for the thread executing in parallel, the same thread which  sets the pointer to 0. I didn't see this coming.

Let me sleep on it, it's late here. Back later

bdbcat, 2020-02-24:
Dan...
A red herring.
I am running the rPI4 headless, using NoMachine as VNC client/viewer.  And NoMachine sometimes has trouble with audio.
I found that I needed to disable OCPN_ENABLE_PORTAUDIO to allow it to startup in this environment.
So nothing to do with your patch.
Sorry for the alarm, OK to close now....
Dave

leamas, 2020-02-24:
Still, Dan's remark on the handling of m_onFinished is valid. I'll be back, this bug closed or not.

leamas, 2020-02-24:
Preliminary thoughts: If we just remove the `m_onFinished = 0 `this might create other problems: the next `Play() `might use the previous callback, defined for the previous `Play()`.

No more time now, but  I'm on it.

transmitterdan, 2020-02-28:
Ok, I’ll redo the regression testing and see if I can make it crash.

leamas, 2020-03-31:
@transmitterdan: ping?

transmitterdan, 2020-03-31:
Cannot make it crash using command line sound configuration.  We can close this issue.

closed 2020-04-05 by bdbcat