OpenCPN/OpenCPN #1751: Plugins: MacOS: data is not found.

leamas, 2020-03-03:
Preliminary testing of plugin loader in MacOS looks OK: plugins can be enumerated and installed into locations which look sane.

However, the plugins seems to not be able to locate their data directory. One problem is that support for MacOS is missing in OCPNPlatform::GetPluginDataPath(). There might be more.

rgleason, 2020-03-03:
Thank you Alec, for testing!

leamas, 2020-03-03:
This looks weird. After enabling  all available plugins in the beta catalog, no plugin calls GetPluginDataDir().

Now, on flatpak oesenc finds it's EULA file (which lives in the plugin data directory). But not using GetPluginDataDir(), there is nothing in the log indicating that it does.

@bdbcat: Have you replaced  GetPluginDataDir() somehow?

bdbcat, 2020-03-04:
Alec...
IIRC, In oeSENC_PI, the EULAs are packaged with the chart files.  This allows different EULAs for different chartset copyright holders.  Actually, it does not need to be packaged with plugin data.

Certainly, WeatherRouting calls GetPluginDataDir().
icons.cpp:

```
    wxFileName fn;
    fn.SetPath(GetPluginDataDir("weather_routing_pi"));
    fn.AppendDir(_T("data"));
    fn.SetFullName(_T("weather_routing_pi.svg"));
    _svg_weather_routing = fn.GetFullPath();
    wxLogMessage(_T("Loading toolbar icon: ") + _svg_weather_routing); 

```

leamas, 2020-03-04:
Here is a more basic error. While installing WeatherRouting looks OK, the dylib file ends up in `~/Library/Application Support/Contents/unknown-paths`.

This seems to be about the macos installation paths. Looking at the macos tarball, the layout looks like (after removing lot's of locales and SymbolPatch files):

    $ tree .
    .
    └── tmp
        ├── bin
        │   └── OpenCPN.app
        │       └── Contents
        │           ├── PlugIns
        │           │   ├── liboesenc_pi.dylib
        │           │   └── oeserverd
        │           ├── Resources
        │           │   ├── ar_SA.lproj
        │           │   │   └── opencpn-oesenc_pi.mo
        │           │   └── zh_TW.lproj
        │           │       └── opencpn-oesenc_pi.mo
        │           └── SharedSupport
        │               └── plugins
        │                   └── oesenc_pi
        │                       ├── data
        │                       │   ├── license.txt
        │                       │   ├── LUPPatch1.xml
        │                       │   └── SymbolPatch5.xml
        │                       └── rrc_eula_ChartSetsForOpenCPN.txt
        └── oesenc_pi
            └── oesenc_pi
                └── liboesenc_pi.dylib

I did some patches in "my" oesenc variant to clean up this mess, which is ubiquitous. The paths need to be like described at https://github.com/leamas/opencpn/wiki/Tarballs. It's basically about dropping the tmp/bin prefix and to drop the duplicate liboesenc_pi.dylib (last one)

@bdcat: can you have a look at this? At least oernc_pi, oesenc_pi and weather_routing_pi seems to be affected.

bdbcat, 2020-03-04:
Alec...
Sorry, just copied from your original build recipe, and did not check carefully .
Is your oesenc_pi github repo merged?  I'll take a look.
Or, I can take a PR...

leamas, 2020-03-04:
I can try to make a PR for oesenc, as a template.... stay tuned

leamas, 2020-03-04:
Done: https://github.com/bdbcat/oesenc_pi/pull/67

It would be extremely helpful if you could get this into the beta catalog to make further macos testing simpler.

Can you take the other ones?  It should be more or less same patches everywhere, I suppose.

bdbcat, 2020-03-05:
Alec...
I'll do oeRNC, and send a PR to Rick for WeatherRouting

leamas, 2020-03-05:
I can now see that oesenc ends up where it should be. After applying #1754 I can also see that the radar plugin successfully locates it's data directory using GetPluginDataDir().

There are multiple glitches in oesenc, notably in the UI's dark mode. However, I neither have the skills nor time required to iron this out. I'll try to handle the basic stuff (installing, data dir, etc.;  things directly related to the installer). 

If mac should stay in a reasonable state, we need someone who knows macos and is able to give it some love.

As I see it, this bug can be closed after merging #1754. Anyone against? Please speak up!

leamas, 2020-03-05:
#1754 merged, closing.

closed 2020-03-05 by leamas