OpenCPN/OpenCPN #1752: Operation out of the box: TCDATA missing fix

rgleason, 2020-03-04:
Users would appreciate, after installation, that things generally worked "out of the box".
For example, TCDATA location is under Data, and a recent fresh Ubuntu installation showed "Tide Data Missing" due to the path in the conf file.

free-x, 2020-03-04:
Did you use the corresponding option -DBUNDLE_TCDATA=ON when building?

rgleason, 2020-03-04:
No not required. however the data exists under /home/rgleason/OpenCPN/data/tcdata/Harmonic.idx
and I simply had to go to Options>Charts>Tide & Currents>Add Dataset

and remove the incorrect one that was there.  I think tcdata has perhaps been moved under "data" and the path has not caught up in the config file?

free-x, 2020-03-04:
it does not matter what for Linux distribution and which version 

or you copy the "tcdata" into the correct directory manually
```
sudo cp -a /home/rgleason/OpenCPN/data/tcdata /usr/local/share/opencpn/
```

rgleason, 2020-03-04:
Ok, but this is a NEW git clone of OpenCPN v5.0.522 (dev)  and build installation it should be correct "Out of the box" IMHO.  Did I execute git clone from the wrong directory? What is the standard location?

I am new to ubuntu, so exactly why is /usr/local/share/opencpn/   "...the correct directory?"
Is this  Computer > usr/local/....?

rgleason, 2020-03-04:
https://askubuntu.com/questions/150611/how-to-tell-ubuntu-where-to-install-a-program-and-how-to-tell-where-an-existing
Approximate equivalents of Windows install directories in Linux
    \Windows = /bin
    \Windows\System32 = /lib and /sbin
    \Program Files = /usr/bin and /usr/lib

Now I read I should have created a deb file so I could uninstall Opencpn easily. How to undo this?
What directory should git clone https://github.com/opencpn/opencpn.git be executed from?
https://git-scm.com/book/en/v2/Git-Basics-Getting-a-Git-Repository
First create  for Linux:  $ cd /home/user/my_project

Closing. I've got other problems to fix.

closed 2020-03-04 by rgleason