OpenCPN/OpenCPN #1753: Ubuntu 18.04  -Landing Paths for Plugins needs fixing

rgleason, 2020-03-04:
When looking at the file structure for OpenCPN I saw something funny.
Weather_routing and TestPlugin should be up one level! 
This is the same problem we had with msvc install that bdbcat fixed. 
I believe it may be a universal problem for all environments.
[image]
[image]

rgleason, 2020-03-04:
Alec's PI Manager Wiki has file structure for various environments.
https://github.com/leamas/OpenCPN/wiki/Tarballs#example-ubuntu-layout

He shows OpenCPN installed into  Home > Shares > OpenCPN.
I installed OpenCPN into Home > OpenCPN
Should I uninstall and reinstall properly to that location?

Also I see no "bin" or "lib" under "Home" even with show hidden toggled on.
I do see  Computer>bin  and Computer>lib,  are those correct?
I do not find Computer>bin> oeserverd  or Computer>lib>opencpn.

Where I find these is
Home > .local >bin > oeserverd
Home > .local > bin > opencpn

rgleason, 2020-03-04:
Back to the point of this Issue!
Under Home>Share I do not find "opencpn" data
I find it under  Home > .opencpn
  and oernc_pi and oesenc_pi are NOT under "plugins' as they should be according to Alec's diagram.

Alec's diagram under "Share"
        └── opencpn
            └── plugins
                └── oesenc_pi
                    └── data
                        ├── license.txt
                        └── SymbolPatch5.xml

leamas, 2020-03-05:
It's hard to read your quotation of https://github.com/leamas/OpenCPN/wiki/Installation-paths#linux -- try use the  Insert Code  <> above, or use four spaces indent to get a monospaced font and working indent.

There is nothing like Share mentioned in the docs. On linux, plugins are installed under .local/share, .local/bin and .local/lib. See reference above

leamas, 2020-03-05:
Installing weather_routing on bionic shows that  it is installed as expected in `~/.local/lib/opencpn` and `~/.local/share/opencpn/plugins/weather_routing_pi`. At a glance, it also seems to work. 

I have problems to understand what this bug is all about?

rgleason, 2020-03-05:
Alec, Thanks for confirming that weather_routing is installing properly on Ubuntu 18.04,  that is much appreciated, but I think you are missing the point completely.  Weather_routing install seems to be different than oesenc_pi and oernc_pi  and one of those is not right they are not under plugins. See the actual screenshots above.  

I find that the plugin data files are here
~/.local/share/opencpn/plugins/oesenc_pi/data
~/.local/share/opencpn/plugins/oernc_pi/data
~/.local/share/opencpn/plugins/weather_routing_pi/data
~/.local/share/opencpn/plugins/testplugin_pi/data
 This is
**~/.local/share/opencpn/plugins/**testplugin_pi/data
which is not as you have shown below.

From  https://github.com/leamas/OpenCPN/wiki/Installation-paths
```
~/.local/
     ├── lib
     │    └── opencpn
     │          └── LIBRARIES
     ├── share
     │    ├── opencpn
     │    │     └── <plugin>
     │    │              └── DATA FILES
     │    └── locale
```
It should be shown IMHO
...
     │    ├── opencpn
     │    │     └── plugin
     |      |               └── <plugin_pi>
     │    │                            └── DATA FILES
     │    └── locale
```

Yes,  <plugin_pi.so> files are all found at
     ~/.local/bin/opencpn/
                   libvdr_pi.so
                   libweather_routing_pi.so
                   libtestplugin_pi.iso
                   liboesenc_pi.so
                   liboernc_pi.so
Just as you said.

In the first screenshots above , I guess I was in the equivalent of Windows
C:\ProgramData\opencpn  directory

```
    Home/.opencpn/
                      Configs
                       iconCache
                       _oernc_pi_   (empty)
                       _oesenc_pi_ (empty)                           
                       plugins
                                 install_data
                                 testplugin_pi
                                 weather_routing_pi  
                       SENC
                       navobj.xml
                       navobj.xml1  etc

```

Why aren't oesenc_pi and oernc_pi down under "plugins" like the other plugins?
BTW it is the same structure as above in Windows.
This threw me off because I am unfamiliar with Ubuntu.

rgleason, 2020-03-05:
I can close this. Perhaps you could change the  

`~/.local/share/opencpn/plugin/<plugin_pi>/data
`
Path?
From https://github.com/leamas/OpenCPN/wiki/Installation-paths

bdbcat, 2020-03-05:
Rick....
The two directories "oernc_pi", and "oesenc_pi" are created at run-time by the plugins.  They are not part of the tarball, nor installed by the installer.  They are used for dynamic run-time data.  It could be argued that they might better be in ".local/", but I think not, for historical reasons.  If you install s63_pi, you will also see that plugin create a directory in .opencpn, also used for run-time storage.
Indeed, any plugin at run-time has full access to .opencpn/*, and should use that location for configuration storage and other live data.

So, I think all is well.
Dave

rgleason, 2020-03-05:
Thanks for the explanation. Will close.  Hope Leamas will change the path diagram.

closed 2020-03-05 by rgleason

leamas, 2020-03-05:
I have updated the diagram, it was indeed wrong, missing the the _plugins_ level in both linux and flatpak. Thanks for heads-up on this.