OpenCPN/OpenCPN #1756: Exception starting OpenCPN in VS 2017.

trudK45, 2020-03-07:
Starting Debug I get "Exception Thrown".
opencpn.exe has triggered a breakpoint.

OS = win 7 SP1, Version = 6.1.7601, Swedish
Keyboart layout = Swedish

OpenCPN = master branch
wxWidgets bersion = 3.1.3

> C:\Users\hasse\obeta\OpenCPN>git status
> On branch master
> Your branch is up to date with 'origin/master'.
>
> nothing to commit, working tree clean

File = **appbase,xpp** line =  1163, marked below.

> // default assert handler
> static void
> wxDefaultAssertHandler(const wxString& file,
>                        int line,
>                        const wxString& func,
>                        const wxString& cond,
>                        const wxString& msg)
> {
>     // If this option is set, we should abort immediately when assert happens.
>     if ( wxSystemOptions::GetOptionInt("exit-on-assert") )
>         wxAbort();
> 
>     // FIXME MT-unsafe
>     static int s_bInAssert = 0;
> 
>     wxRecursionGuard guard(s_bInAssert);
>     if ( guard.IsInside() )
>     {
>         // can't use assert here to avoid infinite loops, so just trap
>         wxTrap();     <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< line 1163
> 
>         return;
>     }
> 
>     if ( !wxTheApp )
>     {
>         // by default, show the assert dialog box -- we can't customize this
>         // behaviour
>         ShowAssertDialog(file, line, func, cond, msg);
>     }
>     else
>     {
>         // let the app process it as it wants
>         // FIXME-UTF8: use wc_str(), not c_str(), when ANSI build is removed
>         wxTheApp->OnAssertFailure(file.c_str(), line, func.c_str(),
>                                   cond.c_str(), msg.c_str());
>     }
> }

Call stack below:

> >	wxbase313ud_vc_custom.dll!wxDefaultAssertHandler(const wxString & file, int line, const wxString & func, const wxString & cond, const wxString & msg) Line 1163	C++
>  	wxbase313ud_vc_custom.dll!wxOnAssert(const char * file, int line, const char * func, const char * cond, const wchar_t * msg) Line 1264	C++
>  	opencpn.exe!wxRound(double x) Line 139	C++
>  	opencpn.exe!ChartCanvas::GetCanvasPointPixVP(ViewPort & vp, double rlat, double rlon, wxPoint * r) Line 4300	C++
>  	opencpn.exe!ChartCanvas::GetCanvasPointPix(double rlat, double rlon, wxPoint * r) Line 4286	C++
>  	opencpn.exe!ChartCanvas::ScaleBarDraw(ocpnDC & dc) Line 6218	C++
>  	opencpn.exe!glChartCanvas::DrawFloatingOverlayObjects(ocpnDC & dc) Line 2831	C++
>  	opencpn.exe!glChartCanvas::Render() Line 4734	C++
>  	opencpn.exe!glChartCanvas::OnPaint(wxPaintEvent & event) Line 1661	C++
>  	wxbase313ud_vc_custom.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 658	C++
>  	wxbase313ud_vc_custom.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 670	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::ProcessEventIfMatchesId(const wxEventTableEntryBase & entry, wxEvtHandler * handler, wxEvent & event) Line 1417	C++
>  	wxbase313ud_vc_custom.dll!wxEventHashTable::HandleEvent(wxEvent & event, wxEvtHandler * self) Line 1023	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::TryHereOnly(wxEvent & event) Line 1612	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::TryBeforeAndHere(wxEvent & event) Line 3912	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::ProcessEventLocally(wxEvent & event) Line 1545	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1518	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::SafelyProcessEvent(wxEvent & event) Line 1636	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindowBase::HandleWindowEvent(wxEvent & event) Line 1556	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::HandlePaint() Line 5255	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::MSWHandleMessage(long * result, unsigned int message, unsigned int wParam, long lParam) Line 3083	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::MSWWindowProc(unsigned int message, unsigned int wParam, long lParam) Line 3873	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWndProc(HWND__ * hWnd, unsigned int message, unsigned int wParam, long lParam) Line 2940	C++
>  	[External Code]	
>  	[Frames below may be incorrect and/or missing, no symbols loaded for user32.dll]	Unknown
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::Update() Line 1732	C++
>  	opencpn.exe!ChartCanvas::OnPaint(wxPaintEvent & event) Line 9846	C++
>  	wxbase313ud_vc_custom.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 658	C++
>  	wxbase313ud_vc_custom.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 670	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::ProcessEventIfMatchesId(const wxEventTableEntryBase & entry, wxEvtHandler * handler, wxEvent & event) Line 1417	C++
>  	wxbase313ud_vc_custom.dll!wxEventHashTable::HandleEvent(wxEvent & event, wxEvtHandler * self) Line 1023	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::TryHereOnly(wxEvent & event) Line 1612	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::TryBeforeAndHere(wxEvent & event) Line 3912	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::ProcessEventLocally(wxEvent & event) Line 1545	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1518	C++
>  	wxbase313ud_vc_custom.dll!wxEvtHandler::SafelyProcessEvent(wxEvent & event) Line 1636	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindowBase::HandleWindowEvent(wxEvent & event) Line 1556	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::HandlePaint() Line 5255	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::MSWHandleMessage(long * result, unsigned int message, unsigned int wParam, long lParam) Line 3083	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWindow::MSWWindowProc(unsigned int message, unsigned int wParam, long lParam) Line 3873	C++
>  	wxmsw313ud_core_vc_custom.dll!wxWndProc(HWND__ * hWnd, unsigned int message, unsigned int wParam, long lParam) Line 2940	C++
>  	[External Code]	
>  	wxmsw313ud_core_vc_custom.dll!wxRichMessageDialog::ShowModal() Line 89	C++
>  	wxmsw313ud_core_vc_custom.dll!wxGUIAppTraitsBase::ShowAssertDialog(const wxString & msg) Line 535	C++
>  	wxbase313ud_vc_custom.dll!ShowAssertDialog(const wxString & file, int line, const wxString & func, const wxString & cond, const wxString & msgUser, wxAppTraits * traits) Line 1386	C++
>  	wxbase313ud_vc_custom.dll!wxAppConsoleBase::OnAssertFailure(const wchar_t * file, int line, const wchar_t * func, const wchar_t * cond, const wchar_t * msg) Line 869	C++
>  	wxbase313ud_vc_custom.dll!wxDefaultAssertHandler(const wxString & file, int line, const wxString & func, const wxString & cond, const wxString & msg) Line 1178	C++
>  	wxbase313ud_vc_custom.dll!wxOnAssert(const char * file, int line, const char * func, const char * cond, const char * msg) Line 1255	C++
>  	wxbase313ud_vc_custom.dll!`anonymous namespace'::GetInfoFromLCID(unsigned long lcid, wxLocaleInfo index, wxLocaleCategory cat) Line 1633	C++
>  	wxbase313ud_vc_custom.dll!wxLocale::GetInfo(wxLocaleInfo index, wxLocaleCategory cat) Line 1732	C++
>  	wxbase313ud_vc_custom.dll!wxString::FromCDouble(double val, int precision) Line 1858	C++
>  	wxmsw313ud_core_vc_custom.dll!wxNativeFontInfo::ToString() Line 789	C++
>  	wxmsw313ud_core_vc_custom.dll!wxFontBase::GetNativeFontInfoDesc() Line 376	C++
>  	opencpn.exe!FontMgr::GetSimpleNativeFont(int size, wxString face) Line 228	C++
>  	opencpn.exe!FontMgr::GetFont(const wxString & TextElement, int user_default_size) Line 210	C++
>  	opencpn.exe!ChartCanvas::CreateDepthUnitEmbossMaps(ColorScheme cs) Line 10917	C++
>  	opencpn.exe!ChartCanvas::ChartCanvas(wxFrame * frame, int canvasIndex) Line 740	C++
>  	opencpn.exe!MyFrame::CreateCanvasLayout(bool b_useStoredSize) Line 3311	C++
>  	opencpn.exe!MyApp::OnInit() Line 2322	C++
>  	wxbase313ud_vc_custom.dll!wxAppConsoleBase::CallOnInit() Line 93	C++
>  	wxbase313ud_vc_custom.dll!wxEntryReal(int & argc, wchar_t * * argv) Line 491	C++
>  	wxbase313ud_vc_custom.dll!wxEntry(int & argc, wchar_t * * argv) Line 184	C++
>  	wxmsw313ud_core_vc_custom.dll!wxEntry(HINSTANCE__ * hInstance, HINSTANCE__ * __formal, char * __formal, int nCmdShow) Line 305	C++
>  	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * __formal, int nCmdShow) Line 1094	C++
>  	[External Code]	

/Hans

leamas, 2020-03-08:
Hej Hans :)

First: I don't know much about windows, so take anything I say with tons of salt.

That said, I presume that you are building the master branch(?) You might want to compare with the build at https://cloudsmith.io/~alec-leamas/repos/opencpn/packages/detail/opencpn-msvc_505224131ddd954_setupexe just too see if it's about your build environment or some more basic error in the code

trudK45, 2020-03-08:
leamas,

> OpenCPN = master branch
> wxWidgets bersion = 3.1.3

Yes, I build from master branch as I wrote in my first post.
I have no idea what you want me to do with the file at the link you included on your post.

/Hans

leamas, 2020-03-08:
> I have no idea what you want me to do with the file at the link you included on your post.

My idea is that you download (green button) and run it. It's a standard NSIS installer. With the new version in place, can you reproduce the crash?

trudK45, 2020-03-08:
leamas,

But isn't that an install file for opencpn (built with "cmake --build . --config release --target package") ?
My problem is that I get an exception when trying to debug opencpn with VS 2017.

/Hans

transmitterdan, 2020-03-08:
I suggested Hans post his crash report. The issue appears to be a conflict between opencpn when it is starting fresh on a new machine. I don’t know why exactly but when first accessing wxWidgets libraries they find there is a conflict between the decimal separator character ‘comma vs. period’ and create an endless trap. I believe O tries to start in English by default and maybe this is a problem whenever the machine’s native language is one that uses comma as the decimal separator.

transmitterdan, 2020-03-08:
> leamas,
> 
> But isn't that an install file for opencpn (built with "cmake --build . --config release --target package") ?
> My problem is that I get an exception when trying to debug opencpn with VS 2017.
> 
> /Hans
Hans,

He wants to know if that version of the program crashes on your computer.

trudK45, 2020-03-08:
Leamas transmitterdan,

No it does not crash, neither do  the official release OpenCPN 5.9.9+9065270.

/Hans

trudK45, 2020-03-08:
tansmitterdan
> I suggested Hans post his crash report. 

How do you find the report in VS 2017 ?

/Hans

leamas, 2020-03-08:
> No it does not crash, neither do the official release OpenCPN 5.9.9+9065270.

OK, then I rest my case. You are in the safe hands of @transmitterdan  :)

trudK45, 2020-03-08:
leamas,
> OK, then I rest my case. You are in the safe hands of @transmitterdan :)
Yes, I know. He is fabulous ! :)

/Hans

trudK45, 2020-03-08:
transmitterdan,

I changed the decimal separator in the windows system from , to . and now OpenCPN start in the debugger without any exceptions raised so you are most surely correct in your assumption that this is the problem.

This is no good, is it ? Should be fixed in master, correct ?

/Hans

transmitterdan, 2020-03-08:
@trudK45 

Right now I don't know what is causing the problem.  Since you have O running now after changing your system decimal separator to period please try the following:

Start O in debugger.
Navigate to options->User Interface and select the language native to your machine.
Close the options dialog and close OpenCPN normally.

Then change your decimal separator back to comma,'
Then start O in the debugger

Does it run or not?

transmitterdan, 2020-03-09:
@trudK45 

I could duplicate the wxAssert by setting my default decimal separator to comma (,).  But it is necessary for OpenCPN to change the separator because many of the numbers it processes have the period as the decimal separator.  When the wxAssert dialog box pops up there is the option to check a box saying never to so this message again.  When I click that box and click Continue then OpenCPN runs normally.

trudK45, 2020-03-09:
@transmitterdan 

I have run 2 cases

**Case 1**
1) dot separator set.
2) Start VS 2019.
3) decimal separator set.
4) Start debugging.
= > Trap with the separator mismatch.

**Case 2**
1) dot separator set.
2) Start VS 2019.
3) Start debugging and let the chart appear.
4) decimal separator set.

Paning and zooming seem to work fine.
Opening tools-> Plugins                       -> Trap with the separator mismatch.

/Hans

trudK45, 2020-03-09:
> @trudK45
>
> I could duplicate the wxAssert by setting my default decimal separator to comma (,). But it is necessary for OpenCPN to change the separator because many of the numbers it processes have [...]

Yes, the same here.

/Hans

transmitterdan, 2020-03-10:
Then I think we should just ignore this wxAssert from wxWidgets for now.  I will do more research to see if it is something we need to address.  But it will not be a simple fix.  Many data sets that O handles come only with period decimal separators.  We can't change that.

Hakansv, 2020-03-10:
For info I've built on Win the last five six years or so. From the very beginning I'd change to period in my Swedish Windows. The possible downgrade was I've to train me to use period also in Excel, :)

trudK45, 2020-03-10:
But, the surprising thing is that this is a new behavior. I have previously been able to use VS 2015 and Swedish Windows without changing the decimal separator to a dot. From 5 to 2 years ago I was doing extensive debugging reverse engineering and writing code for the xHD radar plugin using VS 2015 and Swedish Windows out of the box. It's now that I after 1-2 years I again started debugging that this problem shows up.

transmitterdan, 2020-03-10:
It is a change in wxWidgets.  They have added hundreds of wxAsserts to the debug libraries.  These are designed to alert developers about possible issues.

bdbcat, 2021-04-28:
Closing, inactive

closed 2021-04-28 by bdbcat