OpenCPN/OpenCPN #1761: Plugin Developer Thread: Plugin side of CI process

rgleason, 2020-03-08:
This thread is an attempt to contain Development issues, comments and  questions regarding the use of the new "Managed" Plugins. 

This thread is focused on the implementation of any plugin's changes to  CMake + CI needed to use the new "managed" installation tarballs and metadata.

Plugin Dev's are encouraged to review and implement the system. Please review these pages  (work in progress) in the Dev Manual:

[Plugin Manager: Dev Setup](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installler_build_deploy)
[Plugin Manager: Dev Procedure](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installer_procedure#pi_manager_dev_procedure)
[PI Manager Catalog & Meta-Url](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installer_opencpn_plugins#pi_manager_catalog_meta-url)
[PI Manager for Users](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installler_for_users#pi_manager_for_users)

leamas, 2020-03-09:
The current docs are focused on writing a new plugin based on the testplugin framework. While a reasonable approach, we are missing a Porting Guide which covers the necessary steps to convert an existing plugin to new installer. This must be focused on a minimal step-by-step description.

rgleason, 2020-03-09:
Documents are not finished and not focused on a new plugin.
1. Updates needed to the actual testplugin framework (which has evolved, as you know).
2. The process must be explicit and  detailed enough for Plugin Devs to follow easily.

rgleason, 2020-05-05:
I will continue working on the wiki document.. Closing

closed 2020-05-05 by rgleason