OpenCPN/OpenCPN #1762: Cloudsmith License deprecation

rgleason, 2020-03-10:
GPL-2.0 is deprecated on Cloudsmith.  What licensing should we be using?
https://spdx.org/licenses/GPL-2.0.html

rgleason, 2020-03-10:
PS: Also got this notice at the bottom of appveyor after uploading:

> Discovering tests...OK
> Collecting artifacts...
> Found artifact 'build\vdr_pi-0.4.0.0-ov50-1.16-pi-msvc-win32.exe' matching 'build\*.exe' path
> Uploading artifacts...
> [1/1] build\vdr_pi-0.4.0.0-ov50-1.16-pi-msvc-win32.exe (235,872 bytes)...100%
> Deploying using Cloudsmith provider
> The requested service 'Cloudsmith (Appveyor.Deploy.IDeploymentProvider)' has not been registered. To avoid this exception, either register a component to provide the service, check for service registration using IsRegistered(), or use the ResolveOptional() method to resolve an optional dependency.
> 

I haven't registered Appveyor with Cloudsmith?  
I believe I gave appveyor my key from cloudsmith. Isnt that enough?

leamas, 2020-03-10:
> GPL-2.0 is deprecated on Cloudsmith. What licensing should we be using?

We should certainly not change license, this is an utterly complicated thing requiring us to contact all of those which have been part of this sw. It just won't happen.

I would be really surprised if cloudsmith took any actions because spdx depreceted GPLv2. Yes, its' probably a bad idea to use it for new code, but that's another story. There are tons of GPLv2 code out there (the linux kernel comes to mind), and this wont change just because of a deprecation.

In short: we should not do anything at all because of this.

leamas, 2020-03-10:
> PS: Also got this notice at the bottom of appveyor after uploading: [...]

Looks like a cloudsmith problem. You might want to contact their support.

rgleason, 2020-03-10:
Yes I did.  They said it was the organization in the url. The organization deprecates it.

>  ..would be really surprised if cloudsmith took any actions because spdx depreceted GPLv2. Y

Yes you are exactly right.

Cloudsmith asked me if I had made this  Cloudsmith (Appveyor.Deploy.IDeploymentProvider)' 
I had to laugh.

lskillen, 2020-03-11:
Lee from @cloudsmith-io here. No need to laugh! 😆 

It's because Appveyor is a third-party service. When you specified `provider: cloudsmith` in the appyveyor configuration, it told appveyor to try and load a deployment provider for Cloudsmith, which doesn't exist (yet). You were asked if you made it, since we weren't aware of it, and there was a small chance that is what you were doing. :-)

The recommended advice for appveyor is to script the deployment, using the Cloudsmith CLI, of which the commands are available in the contextual documentation on-site, or here:
https://help.cloudsmith.io/docs/upload-package

For the GPL 2.0 license, Cloudsmith doesn't take any prescriptive stance on actual licenses, deprecated or not. As long as the license for the project is SPDX approved, we'll be OK with the repository being licensed like that. In other words, no action required.

Hope that helps!

rgleason, 2020-03-11:
Lee thank you!  (Lee to the rescue a second time.)

leamas, 2020-03-11:
Seems that the issue here could be resolved as NOT_A_BUG. Time to close?

rgleason, 2020-03-11:
What happened was that I had enabled deploy at the bottom of appveyor.yml
and put provider: Cloudsmith   with the key.
Appveyor's API does not list cloudsmith as a provider, so this cannot be done.
The reason it worked and uploaded was that the appveyor-upload.sh script worked.
Therefore I commented out the deploy at the bottom of appveyor.yml
and things work better.

closed 2020-03-11 by rgleason