OpenCPN/OpenCPN #1763: SignalK => missing non-verbose option?

hreuver0183, 2020-03-10:
I was testing SignalK in combination with OpenCPN and it occured to me that I would like silence the messages a little more. If every message gets printed to stdout or stderr every system will get slow.

What command do I need to prevent OpenCPN from putting every SignalK update on the terminal?
I get messages like 

"{"updates":[{"source":{"label":"canable0","type":"NMEA2000","pgn":130306,"src":"0"},"timestamp":"2020-03-10T20:28:03.054","values":[{"path":"environment.wind.speedApparent","value":0},{"path":"environment.wind.angleApparent","value":2.5385}],"$source":"canable0.0"}],"context":"vessels.urn:mrn:signalk:uuid:xxxxxxxxxxxxxx"}
"

Did I perhaps invoke a (default) debug option when compiling?

leamas, 2020-03-11:
We are under way to merge #1232. This might be able to resolve this, offering logging tools which can be configured in runtime.  This would mean converting these printf() calls to logging calls.

So, if you cold wait a little...

hreuver0183, 2020-03-11:
No problem.

leamas, 2020-03-12:
The new logging is merged. You might want to make a PR, perhaps by replacing the teminan logging (printf?) with LOG_DEBUG or so. 

There is some provisionary docs at https://github.com/leamas/OpenCPN/wiki/New-logging-system

hreuver0183, 2020-03-13:
The first part of my question is now resolved and earlier then I expected:
I created a big black hole and every message disappeared in it :-)

Now only to find out where the messages are going to.
Most information I found aimed at getting messages into wxlog (and the information you provided was really helpfull).

I guess it's the difference between users and developpers.

leamas, 2020-03-13:
To the logfile. Which platform are you on (windows, mac, linux)?

hreuver0183, 2020-03-13:
Ubuntu / Linux
I must ray I'm not too comfortable with systemd, but both "journalctl -f" and /var/log/syslog show no SignalK data.

I was hoping to test before providing a PR.

leamas, 2020-03-13:
The data goes to `~/.opencpn/opencpn.log `on linux. Remember, LOG_DEBUG output is suppressed unless you start opencpn witn _opencpn --loglevel debug_.

hreuver0183, 2020-03-13:
Works :-)

hreuver0183, 2020-03-13:
In my fork of OpenCPN I just tried to replace some of the fprintf with LOG_DEBUG or LOG_ERROR just to get familiar with "git push".
If nobody needs those commits I plan to revoke them in my repository since it was mainly for testing.

The SignalKEventHandler.cpp change has been committed as PR (tested).

closed 2020-03-15 by hreuver0183