OpenCPN/OpenCPN #1764: Circleci-Build-Fedora.sh  in squiddio_pi  (from oesenc)

rgleason, 2020-03-12:
**/bin/bash: ci/circleci-build-fedora.sh: Permission denied**
Exited with code exit status 126
[Workflow Job](https://app.circleci.com/pipelines/github/rgleason/squiddio_pi/165/workflows/8567d0e8-56b0-4939-8128-3a48eb07c20c/jobs/390)

Which makes me wonder if we have been banished from use of the repository
```
Notes:
=====
The build requires updated packages, notably wx3GTK, available at
https://copr.fedorainfracloud.org/coprs/leamas/opencpn-mingw.
The copr enable command above sets up access to this repo.

```
If so, I would appreciate knowing this, as there has been no notice, and I have spent considerable time trying to figure this out!   Both Fedora and Mingw builds depend on this access.

[CircleCI Support ](https://support.circleci.com/hc/en-us/articles/360003649774-Permission-Denied-When-Creating-Directory-or-Writing-a-File)
> 
> If you receive an error telling you that you do not have permissions to create a directory or to write a file to a directory then this is likely an indication that your script is attempting to write to a directory that the user running the build does not own. 
> 
> This is a somewhat common pitfall that many users run into when the move into a CI environment. 
> 
> The key thing to remember is that the builds run as the distiller user on MacOS builds and typically ubunutu on Linux builds. These users only have write permissions in their $HOME folders and places like /tmp. This is not unique to CI, this is true by default in almost all Linux/Unix environments.

bdbcat, 2020-03-12:
Rick...
Occam's razor:
Are you sure the shell script has the +x permission bit set?

bdbcat, 2020-03-12:
I just pulled from github, branch ci.  The permission bit is NOT set.

rgleason, 2020-03-12:
Thanks! Jeeze, I guess 

```
git update-index --add --chmod=+x build.sh
git commit -m 'Make build.sh executable'
git push
```
doesn't always work in Windows. Done it twice.  Maybe I should git pull my repos from Ubuntu Virtualbox and do it?  ...more reliable.

closed 2020-03-12 by rgleason

rgleason, 2020-03-12:
The problem was that I was doing it from a parent directory using the path. You must be in the actual directory for the commands to work!