OpenCPN/OpenCPN #1766: O RPi fails to start after a crash

Hakansv, 2020-03-12:
This is for RPi3 - Stretch.
If O crashes down and the a safe restart message appear the next message is the one showed below about an active instance or dialog. After that it's not possible to start O. Every attempt to start results in the messages shown. There are no visible other dialog.
The only way to start O after that is to reboot. 
There's a for me unknown file in the config folder but deleting make no difference. ("startcheck" I know. It's the other)

I've tried to same scenario on Win10 but nothing strange there.
This failure is implemented the last week. Cant be more specific since I haven't run the RPi3 since.

Can I in some way search for any running process when the "busy" message occur? ( A Linux (not-)geek may need some assistance in that.)
Håkan

[image]

[image]

[image]

leamas, 2020-03-12:
startcheck.dat is about the first messagebox "run seems to have filed..."

The last messagebox "too busy" comes from the stClient->makeConnection test in chart1.cpp. This seems to be a socket thing in linux, looks like we are trying to connect to ourselves. Havn't digged into the details.

One possible explanation could be that the old instance listened to a socket which was not teared down as it should.  After the  crash, can you check the listening ports by using` sudo netstat -tulpn`? This should reveal if there is an old, not fully closed listening opencpn port despite that opencpn is not running.

Sorry, I',m on thin ice here, only theories... remember something similar long ago.

leamas, 2020-03-12:
Thinking more about this: the approach to use sockets to guarantee a single instance, as done by wxClient, is fundamentally wrong for reasons above.

bdbcat, 2020-03-13:
Hakan...
Try my latest commit.
The problem was that the single instance detector found the lockfile, but there was actuully no reachable other instance of OCPN.  So, we can delete the lockfile, and exit with a message.  Next startup will be OK, I think.
Try it and see.

Thanks
Dave

leamas, 2020-03-13:
and yes, wxClient/wxServer might actually be usable. Looking at the sources, wxServer uses SO_REUSEADDR, and should thus be able to listen() to the socket even if there is a zombie still lingering there. So this "problem" shouldn't be a problem after all.

Hakansv, 2020-03-13:
Dave..
Yes, now it handle all O restarts as expected. Both after a crash, forced by a wall power off, and after a normal closure.
Win is still as it should at both cases.
Thanks
Håkan

closed 2020-03-13 by Hakansv

Hakansv, 2020-03-13:
Just to complete:
My crash method to un-plugg power was not a reliable test method. It's like a reboot.
But a forced alt-F12 crash was also followed by a normal start up after the "safe restart" message.