OpenCPN/OpenCPN #1767: Ubuntu: Building Plugins in-line  - Errors

rgleason, 2020-03-12:
I am getting some errors related to the new Managed Plugin files, such that OpenCPN does not compile using Ubuntu 18.04 in the standard manner.  The errors are related to the html files created by the plugin.  CMake may have to exclude those files somehow, if users are going to have that capability.
https://www.cruisersforum.com/forums/f134/os-linux-specific-201365.html#post3093494

I do not know what users have done in the past, nor do I know what the intentions are, I am just reporting my experience.

bdbcat, 2020-03-13:
Rick...
It appears that you are trying to build vdr_pi as an inline, or default included plugin.

_"The plugin clone landed at opencpn/plugins/vdr_pi. "_
This will not work if the cloned branch is intended to build a managed plugin by ci process.

In general, we have never supported adding arbitrary plugins to the ocpn build tree.  I'm surprised if it ever worked before.  Of course, it could be made to work, but why?

Suggest you clone your vdr_pi repo into "Projects", or somewhere other than the OCPN build tree.

rgleason, 2020-03-13:
I guess tried to combine two methods, not so good. I built the plugin from _opencpn/plugins/vdr_pi_ and then thought that Opencpn build would work, so I will remove vdr_pi and clone it into Projects/plugin-standalone/ .

This is now in my manual. Thankyou!

closed 2020-03-13 by rgleason