OpenCPN/OpenCPN #1772: Ubuntu 18.04 floating toolbar does not redraw correctly

jongough, 2020-03-14:
Using Ubuntu 18.04 Desktop LTS clean install then OpenCPN and some plugins from the standard opencpn repository on launchpad there is a problem with the redraw of the floating toolbar. The problem can be recreated by starting OCPN, resize window to cause multiple columns on the toolbar then double clicking the application title bar or clicking the maximise button (top right) to maximise the application. The following screenshots show the issue:
Initial screen:
[image]
Resized screen:
[image]
Maximised screen:
[image]
Notice the toolbar in the last screenshot does not show all the tools. If the rollup/rolldown button on the toolbar is clicked or if the screen is resized by grabbing the edge with the mouse pointer the toolbar displays correctly.

It appears that there is an issue with calculating the size of the toolbar after the resize.

bdbcat, 2020-03-14:
Jon...
I concur.  Seems to be only a factor of gnome desktop.  Works fine in KDE, for example.
I will investigate further.

transmitterdan, 2020-03-14:
There is perhaps a related issue with the dashboard docking on the left.  Occasionally the tool bar will not move back to the left after undocking the dashboard from left side.

bdbcat, 2020-03-15:
Fixed by 403837ed8701175365a56fdf9f823cb5e5a6da52
Apparently, gnome desktop is caching some top-level-window parameters in a way not seen on other platforms.  So call the fix a "code-workaround".

jongough, 2020-03-15:
Just tried it on my test system and it now displays the toolbar correctly on single and multiple displays. Thanks.

jongough, 2020-03-15:
There is another issue that was raised in https://github.com/OpenCPN/OpenCPN/issues/1772#issuecomment-599068058 that the toolbar does not drag correctly. In my Linux Mint test system it does drag but seems to get 'stuck' and stops moving at times. On Ubuntu 18.04 desktop this happens much faster, i.e. I can only drag it about a toolbar width before it gets stuck.

Do you want me to open another issue on this?

bdbcat, 2020-03-19:
Please close now, related issue opened.

closed 2020-03-19 by jongough