OpenCPN/OpenCPN #1773: ENC oeSENC - soundings font.

Hakansv, 2020-03-14:
This is a diligently discussed issue and now I'm here as well.
I've tried to find where in the code to control the font of the soundings but failed. (I'd like some testing.) Attached shot is an example from Win but it's similar on Linux.
It's hard to see the depth. The close ORIENT figure is very clear though. Couldn't soundings use the same font? 
The example is a oeSENC but pure ENC use corresponding code in S57chart I suppose.
To get rid of the discussion we may use a UI font item for this although it's not obvious since also plugins would need it's own unless it's not able to share via pluginmanager?
[image]

rgleason, 2020-03-14:
Hakan, I think it is buried in a single raster file. (From past investigation, becoming a cobweb right now.)  We were trying to address this very issue and it came down to editing the raster file and being able to recreate the exact locations again. I decided this was not a job for me (inadequate skills).   Does this make any sense from the Wiki?
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:user_interface_styling

There are some discussion threads about this if you want me to find them. Pavel would know more.
It would be really really really nice to get this fixed somehow as there are more tired 70 yr old eyes these days.

rgleason, 2020-03-14:
I did a search on depth and found some threads, but not much technically.
There were many observations about the legibility, which were addressed by Dave with the slider adjustments which made a big difference, but what I think you are talking about is the legibility of the characters which are not "crisp" like a font, but dull.  It would be very nice to change that.

Without dredging up past discussions here are some of the threads.
https://www.cruisersforum.com/forums/f134/sounding-size-adjustment-on-vector-charts-138786.html
https://www.cruisersforum.com/forums/f134/vector-palette-151141.html
https://www.cruisersforum.com/forums/f134/how-to-show-depth-soundings-in-vector-charts-200443.html#post2622256
https://www.cruisersforum.com/forums/f134/sounding-size-adjustment-on-vector-charts-138786.html#post2886193

All of which you know.

tai-fun, 2020-03-14:
Håkan 
I think I can point you to where to start in the code, not much more.
In /OpenCPN/data/s57data/ there are three files "rastersymbolsXXX.png"
These are referred to in "chartsymbols.xml" in the same directory.

One example. On line 39873...
		<symbol RCID="187">
			<name>SOUNDG03</name>
			<description>for deep soundings, greater than safety depth, full-chart and simplified</description>
			<bitmap width="6" height="10">
				<distance min="0" max="0" />
				<pivot x="-2" y="4" />
				<origin x="0" y="0" />
				<graphics-location x="736" y="415" />
			</bitmap>
			<color-ref>ACHGRDBDEPMD</color-ref>
			<definition>R</definition>
		</symbol>

Here <graphics-location x="736" y="415" />  refers to the pixel location in the png pictures mentioned earlier, which I get to a "white" number "3".

These files are then used in the code in (at least)  /OpenCPN/src/s52plib.cpp. 
I hope this get you started.
Good Luck!
Thomas.

tai-fun, 2020-03-14:
OOps
This editor didn't like xml.....See attached picfile:
[image]

rgleason, 2020-03-14:
With an underlying structure dependent on a big png file, isn't it very hard to recreate those pixels with a sharper character, and then put them back in exactly the same place?
If there were some program that allowed extraction and reinsertion, that might help.
It appears OpenCPN already is using this type of code, so maybe the helper tool is readily available?

Hakansv, 2020-03-14:
Thomas..
Thanks for the hints. Never been into these graphical tools. A lot to digest if so desired but it's not necessary for the moment. But now I do understand the base for the earlier discussions and it's not only to connect a font like any text object. And I'm not investigation for my own joy or individual adaptions. 
There has been a lot comments about visibility of the soundings on different screens. So it may be there's a need for some basic tools adaptation. Likely there are methods to achieve another shape of the soundings but if it's to big to dig into for this turn of O development I can't assess although I understand it's not an easy task to adapt to all available display types. 

For me on my rather common screen a slightly bigger and more black rather then the present gray font would do a lot, for the day scheme. 
I think Dave has been tweaking here a lot but could it be another turn if we have the time before spring is over?
Håkan

bdbcat, 2020-03-15:
Hakan, et al...
Some points:
1.  The current sounding characters are exactly extracted from the S52 PLIB v3.2 specs.  More modern S52 specs have changed these symbols to be more readable, but we have not yet adopted them.
2.  The grey color is as specified by S52 PLIB standards.  Soundings greater than the "safety depth" are presumably of less interest to commercial (i.e. large size) users. So they are de-emphasized on-screen.  Soundings less than safety depth are very interesting to everyone, so they are in a darker color.  That is the spec. But we are not tied to the spec, so it is a valid point that, for us, our user-base is more relevant than the super-tankers of the world.  We can discuss that.  We will have this issue even if we improve the bit-wise font legibility.
3.  It has been on my list for this coming release to improve the soundings rendering to use a modern vector font, e.g. TruType.  I will take a fresh look at that now that the "noise" is subsiding.

Thanks
Dave

Hakansv, 2020-03-15:
Dave..
Very good discussion. Your background description also woke me up. The S52 standard is of course relevant and O's adaption to that is very fine. On the shot below is the safety depth set to four meters and the sounding fonts compliance are obvious. I'd say at my discussions above I wasn't in all details thinking of the depth limits and it's impact on this. And to crowd areas deeper than safety depth with a intrusive font is of course not a good idea. You and the standard are correct. 

What also made me more "noisy" here is the fact that the effect of a changed "Chart Object Scale factor" is not achieved until O is restarted. Neither "Apply" nor "OK" make a change. So yesterday I for a time felt the font is not controllable. If this behavior is new I can't remember?

The needs for supertankers and a sailor like me are the same here. Talking to those found of or dependence of fishing activities there will be another view I suppose. They want to find the deeper escarpments to set on and need clear soundings also here. So your ideas about a more readable font is relevant. But also to train users to play more with the safety depth. This discussion have made it much clearer for me and now and can use this knowledge and deeper insight on the course next week. 

Conclusions:
- "Chart Object Scale factor" ask for a O restart before having effect. Tweak able?
- Another font, e.g. True type would be clearer.
- Could we discuss to move the "Safety Depth" control one step down. My thought is to avoid a "natural" feeling that the three "Shallow", "Safety" and "Deep" settings "must" be in a outgrowth order. I could make a suggest to discuss.

Many thanks. For me a very perceptive discussion.
Håkan

[image]

Hakansv, 2020-03-15:
Is this more clear?

[image]

rgleason, 2020-03-15:
Thanks for the discussion. I had some thoughts last night, before I saw this discussion which might be relevant.
The S52 graphic standard is essentially that collection of graphic files that the depths are taken from.
As monitors get denser and denser pixels the graphic files and apparent size of the depth has gotten smaller and smaller. The CM93 Detail Level slider has been a good response to this problem as it readjusts the scale of the raster based 'depth' instance so it is larger or smaller. However what happens when you overlay a resized 'depth' instance on the chart at a considerably larger size (because the newer monitors have a high pixel count) is that there is naturally fading which also occurs, and the font becomes less legible.

My conclusion then, is that we need large pixel count S52 graphics and (if there is a performance hit) a standard pixel count S52 graphics set.  This will also change the relative adjustment of the CM93 slider bar.

Is this at all right?

Hakansv, 2020-03-16:
Discussion closed. I've made PR of the proposed UI change since no one complained.
The "Chart Object Scale factor" ask for a O restart before having effect. Tweak able? will be a new issue.

closed 2020-03-16 by Hakansv