OpenCPN/OpenCPN #1774: PI Manager UI suggestion:  Plugin Catalog functions

rgleason, 2020-03-14:
Options > Plugins 
Bottom Plugin Catalog functions

All on the left side to avoid resizing issues.
If the dropdown is an issue, it could be put at the bottom.

Active Plugin Catalog:  Alpha   <--- Shows current selection
Choose Remote Catalog: Beta   <---Drop down selection - Master, Beta, Alpha, Custom
Update Plugin Catalog: Alpha   <---Update Button

[image]
[image]

bdbcat, 2020-03-19:
Duplicate, closing

closed 2020-03-19 by bdbcat

rgleason, 2020-03-19:
Dave, I don't believe this is a duplicate. This is about simplifying the catalog functions at the bottom.
The left hand ones show, and the right hand one gets cut off.

All on the left side
Active Plugin Catalog: Alpha <--- Static, Shows current selection
Choose Remote Catalog: Beta <---Drop down selection - Master, Beta, Alpha, Custom
Update Plugin Catalog: Alpha <---Update Button

Also, another thought.  Could the "Apply"  on lower left do two functions?
    "Update Plugin Catalog" and  "Apply"?
Would that simplify the interface?

bdbcat, 2020-03-19:
Rick...
We also have vertical real estate problems on some platforms (7" rPI, for example).
So I am hesitant to use three lines, when two will usually do...

rgleason, 2020-03-20:
Ok, I know menus like this are very very difficult.  I hope we have not messed it up. Thanks for considering it.