OpenCPN/OpenCPN #1776: Linux - dragging toolbar to new location does not stick

jongough, 2020-03-15:
On both Ubuntu 18.04 desktop and Linux Mint 19 if I drag the toolbar to another location on the OCPN screen it moves to that location. If I then open and close the 'settings' the toolbar moves back to the default location. Clicking on the 'Change Color Scheme' button also exhibits the same issue. Using other options on the toolbar does not do this.

jongough, 2021-04-26:
This issue still exists with OCPN 5.5.0. Worse is that dragging is now difficult as the mouse pointer moves relative to the toolbar during the drag process and ends up off of the toolbar effectively leaving it where the mouse moved off. So to move the toolbar from one side of the screen to another takes multiple attempts.

bdbcat, 2021-04-28:
Toolbar position persistence corrected by 82f3ffb583ea0db25a1f843c45beb4ea3a61208c.
Toolbar dragging performance not improved much.  Requires much deep re-factoring.  Won't fix for present.
Please test/comment/close as appropriate.

jongough, 2021-04-28:
Toolbar persistence fixed.

closed 2021-04-28 by jongough