OpenCPN/OpenCPN #1779: Pluginmanager prevents debugging in VS

transmitterdan, 2020-03-15:
When debugging in Visual Studio the pluginmanager causes a fatal exception trap.  The following code snippet shows the problem.
```
/**
 * Clear m_PluginsItems and remove everything in this instance besides:
 *   - The empty m_pitemBoxSizer01.
 *   - The AddPluginPanel
 */
void PluginListPanel::Clear()
{
    for (auto it = GetChildren().GetFirst(); it; it = it->GetNext()) {
        if (dynamic_cast<PluginPanel*>(it->GetData())) {
            it->GetData()->Destroy();
        }
        else if (dynamic_cast<wxStaticLine*>(it->GetData())) {
            it->GetData()->Destroy();
        }
    }
```
Apparently sometimes it->GetNext() returns a non-null but nevertheless invalid pointer.  In the example I can reproduce GetNext() returns 0xdddddddd which causes an exception when dereferenced.

leamas, 2020-03-16:
Strange, indeed. Could it be that *it* is valid, but *it->GetData()* is not?

Just a shot in the dark: Does adding this line as the first line in the loop change anything:

    if (!it->GetData()) continue;

transmitterdan, 2020-03-18:
From what I can tell it->GetData()->Destroy() makes the iterator invalid.  What it points to becomes gibberish. So at that point the for loop is doomed.

[image]

leamas, 2020-03-18:
Yes, of course  "banging head to table".  We (or , really I) enumerate the children of current window, and in that loop removes a child. I'll come up with a patch. 

Good catch! Stay tuned.

transmitterdan, 2020-03-18:
I can probably devise a solution but will hold off for now.  If you have more pressing issues let me know.

leamas, 2020-03-18:
> I can probably devise a solution 

Indeed you can. Here is my attempt to clean up this mess. Could you possibly run it through your setup (if you feel that another patch would be better, just do it...)

    diff --git a/src/pluginmanager.cpp b/src/pluginmanager.cpp
    index f96f944f6..3a88ec858 100644
    --- a/src/pluginmanager.cpp
    +++ b/src/pluginmanager.cpp
    @@ -5199,14 +5199,22 @@ PluginListPanel::PluginListPanel(wxWindow *parent, wxWindowID id,
      */
     void PluginListPanel::Clear()
     {
    +    std::vector<wxWindow*> victims;
         for (auto it = GetChildren().GetFirst(); it; it = it->GetNext()) {
    -        if (dynamic_cast<PluginPanel*>(it->GetData())) {
    -            it->GetData()->Destroy();
    +        if (!it->GetData()) {
    +            continue;
    +        }
    +        else if (dynamic_cast<PluginPanel*>(it->GetData())) {
    +            victims.push_back(it->GetData());
             }
             else if (dynamic_cast<wxStaticLine*>(it->GetData())) {
    -            it->GetData()->Destroy();
    +            victims.push_back(it->GetData());
             }
         }
    +    for (auto victim: victims) {
    +        victim->Destroy();
    +    }
    +

transmitterdan, 2020-03-20:
Fixed by 4c3c06dbab93403209a2015bbd6b27f69f3eefdb

closed 2020-03-20 by transmitterdan