OpenCPN/OpenCPN #1782: Grib plugin does not work on Ubuntu 18.04

farotherside, 2020-03-16:
Using stock, up-to-date Ubuntu 18.04 and .deb packages from  http://ppa.launchpad.net/opencpn/opencpn/ubuntu, grib plugin fails to launch.

message in opencpn.log:
`22:36:20 CST: Error: /usr/lib/opencpn/libgrib_pi.so: undefined symbol: _ZThn696_N6wxGrid31GetSizeAvailableForScrollTargetERK6wxSize`

leamas, 2020-03-16:
I cannot reproduce this on my own, clean vm. Ubuntu 18.04, updated as of today.

My first guess: some runtime linkage to left overs in for example /usr/local/lib. What does *ldd usr/lib/opencpn/libgrib_pi.so* say?l Attaching my own  which looks just fine, as does the plugin.

[ldd.txt](https://github.com/OpenCPN/OpenCPN/files/4340563/ldd.txt)

farotherside, 2020-03-18:
here's the output of that command for me. fwiw I believe my Ubuntu 18.04 is pretty much stock, though I guess I could be wrong on that.

`[xwing@xwing] ~ $ ldd /usr/lib/opencpn/libgrib_pi.so
	linux-vdso.so.1 (0x00007ffe72d2c000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_adv-3.0.so.0 (0x00007f2dc72f7000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0 (0x00007f2dc6ac5000)
	libwx_baseu-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0 (0x00007f2dc6636000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2dc6419000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f2dc6209000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2dc5e80000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2dc5ae2000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2dc58ca000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2dc56ab000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2dc52ba000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f2dc4c78000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f2dc49c3000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f2dc479f000)
	libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f2dc4552000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f2dc42fe000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f2dc3fe7000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f2dc3caf000)
	libnotify.so.4 => /usr/lib/x86_64-linux-gnu/libnotify.so.4 (0x00007f2dc3aa7000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f2dc389a000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f2dc357d000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f2dc3377000)
	libSM.so.6 => /usr/lib/x86_64-linux-gnu/libSM.so.6 (0x00007f2dc316f000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f2dc2f3d000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f2dc2cd5000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f2dc2a5e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2dc285a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2dc7a15000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f2dc2656000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f2dc2450000)
	libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007f2dc222a000)
	libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f2dc1e8b000)
	libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f2dc1c75000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f2dc1a30000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f2dc1826000)
	libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f2dc1623000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f2dc1413000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f2dc1208000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f2dc0ffe000)
	libXcomposite.so.1 => /usr/lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007f2dc0dfb000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f2dc0bf8000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f2dc09e6000)
	libthai.so.0 => /usr/lib/x86_64-linux-gnu/libthai.so.0 (0x00007f2dc07dd000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f2dc05d5000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f2dc0363000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f2dc013b000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f2dbfe87000)
	libpixman-1.so.0 => /usr/lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007f2dbfbe2000)
	libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f2dbf9df000)
	libxcb-render.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f2dbf7d2000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f2dbf5ca000)
	libICE.so.6 => /usr/lib/x86_64-linux-gnu/libICE.so.6 (0x00007f2dbf3af000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f2dbf1a8000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f2dbef82000)
	libjbig.so.0 => /usr/lib/x86_64-linux-gnu/libjbig.so.0 (0x00007f2dbed74000)
	libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007f2dbeb4c000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f2dbe931000)
	libmount.so.1 => /lib/x86_64-linux-gnu/libmount.so.1 (0x00007f2dbe6dd000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f2dbe43f000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f2dbe20d000)
	libdatrie.so.1 => /usr/lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f2dbe006000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f2dbde02000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f2dbdbfc000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f2dbd9e7000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f2dbd79a000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f2dbd56d000)
[xwing@xwing] ~ $ 
`

farotherside, 2020-03-18:
sorry, that came out unreadable. here's a text version:
[ldd.txt](https://github.com/OpenCPN/OpenCPN/files/4347096/ldd.txt)

bdbcat, 2020-03-18:
May we see the same for:
$ldd usr/bin/opencpn

farotherside, 2020-03-18:
absolutely: 
[opencpn_ldd.txt](https://github.com/OpenCPN/OpenCPN/files/4349749/opencpn_ldd.txt)

transmitterdan, 2020-03-18:
@farotherside 

In a post on CF you reported this:

```
[root@xwing] ~ # apt-cache depends opencpn
opencpn
  Depends: libarchive13
  Depends: libbz2-1.0
  Depends: libc6
  Depends: libcairo2
  Depends: libcurl4
  Depends: libexpat1
  Depends: libgcc1
  Depends: libgl1
  Depends: libglib2.0-0
  Depends: liblzma5
  Depends: libpango-1.0-0
  Depends: libpangocairo-1.0-0
  Depends: libportaudio2
  Depends: libstdc++6
  Depends: libtinyxml2.6.2v5
  Depends: libwxbase3.0-0v5
  Depends: libwxgtk3.0-0v5
  Depends: libx11-6
  Depends: zlib1g
  Depends: wx3.0-i18n
  Depends: opencpn-tcdata
  Depends: opencpn-gshhs-low
  Conflicts: <opencpn-plugin-chartdldr>
  Conflicts: <opencpn-plugin-wmm>
  Recommends: xcalib
  Recommends: xdg-utils
  Recommends: opencpn-doc
  Recommends: opencpn-gshhs-crude
  Recommends: opencpn-gshhs-intermediate
  Recommends: opencpn-gshhs-high
  Recommends: opencpn-gshhs-full
  Replaces: <opencpn-plugin-chartdldr>
    opencpn
  Replaces: <opencpn-plugin-wmm>
    opencpn
[root@xwing] ~ #
```
It shows that dpkg thinks the installed version of opencpn depends on  libwxbase3.0-0v5 and libwxgtk3.0-0v5. Is that still the cases? Those don't show up in the ldd report.

When O loads the program the kernel tries to load the grib shared object file and can't find a wx library symbol. This is from the opencpn.log file:
```
19:22:57 CST: Checking plugin candidate: /usr/lib/opencpn/libgrib_pi.so
19:22:57 CST: PLugin is compatible: true
19:22:57 CST: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libgrib_pi.so
19:22:57 CST: Error: /usr/lib/opencpn/libgrib_pi.so: undefined symbol: _ZThn696_N6wxGrid31GetSizeAvailableForScrollTargetERK6wxSize
19:22:57 CST:    PlugInManager: Cannot load library: /usr/lib/opencpn/libgrib_pi.so
```
This seems like some kind of ABI issue.

rgleason, 2020-03-18:
I thought I saw that that was removed by Alec for some environment, either flatpak or fedora I think.
`n/libgrib_pi.so`

farotherside, 2020-03-18:
> It shows that dpkg thinks the installed version of opencpn depends on libwxbase3.0-0v5 and libwxgtk3.0-0v5. Is that still the cases? Those don't show up in the ldd report.

correct, that is still the case - they're still listed in the package manager as dependencies:
`  Depends: libwxbase3.0-0v5
  Depends: libwxgtk3.0-0v5
`

based on your feedback that it might be O compiled against wx for gtk2, I installed packages for libwxbase2.8 and libwxgtk2.8, but that didn't change the behaviour, so I have since uninstalled those packages.

leamas, 2020-03-18:
> This seems like some kind of ABI issue.

Yes

> dpkg thinks the installed version of opencpn depends on libwxbase3.0-0v5 and libwxgtk3.0-0v5. Is that still the cases? Those don't show up in the ldd report.

Perhaps they do. libwxgtk3.0-0v5 contains libwx_gtk2u_gl-3.0.so.0  and friends which indeed are listed.  See https://packages.debian.org/buster/amd64/libwxgtk3.0-0v5/filelist

leamas, 2020-03-18:
The ldd output for the plugin is identical with mine, sans the addresses.

What does `dpkg -l | grep wx | awk '{print $2}'`say? My bionic says

    libwxbase3.0-0v5:amd64
    libwxbase3.0-dev
    libwxgtk3.0-0v5:amd64
    libwxgtk3.0-dev 
    wx-common
    wx3.0-headers
    wx3.0-i18n

farotherside, 2020-03-18:
here's what I get:

```
[root@xwing]` ~ # dpkg -l | grep wx | awk '{print $1,$2}'
ii libgnuradio-wxgui3.7.11:amd64
rc libwxbase2.8-0:amd64
ii libwxbase3.0-0v5:amd64
ii libwxbase3.0-dev
ii libwxgtk-media3.0-gtk3-0v5:amd64
ii libwxgtk-webview3.0-gtk3-0v5:amd64
rc libwxgtk2.8-0:amd64
ii libwxgtk3.0-0v5:amd64
ii libwxgtk3.0-dev
ii libwxgtk3.0-gtk3-0v5:amd64
ii owx
ii python-wxgtk-media4.0
ii python-wxgtk3.0
ii python-wxgtk4.0
ii python-wxversion
ii wx-common
ii wx3.0-headers
ii wx3.0-i18n
[root@xwing] ~ #
```

leamas, 2020-03-18:
hm... seems that you have all sorts of libwx\* packages besides my list, which is what I got after a clean install. What happens if you trim that list to match mine? FWIW, my gut feeling is that it is about this list... 

In particular: I'm suspicious about the libwx\*2.8-0 ones.

rgleason, 2020-03-18:
Ubuntu Grib_pi cannot read normal grib files.  They fail or are not recognized.

farotherside, 2020-03-18:
> hm... seems that you have all sorts of libwx* packages besides my list, which is what I got after a clean install. What happens if you trim that list to match mine? FWIW, my gut feeling is that it is about this list...
> 
> In particular: I'm suspicious about the libwx*2.8-0 ones.
> 
you'll note that I added an extra parameter to 'awk', so that you can see the first column of the dpkg output - 'ii' means the package is installed, but 'rc' means the package has been removed and only the config files remain. the libwx*2.8 packages have been removed.

rgleason, 2021-04-23:
@farotherside  Is this still an issue?  I haven't tried 18.04 recently.

bdbcat, 2021-04-28:
Closing, inactive

closed 2021-04-28 by bdbcat

rgleason, 2021-04-28:
To confirm Grib_pi is working: Last night I tested Ubuntu 18.04  Ov5.5.0  and the Grib plugin works fine with Weather_routing, running optimized routes.