OpenCPN/OpenCPN #1784: wxWidgets font point size compatibility/portability

bdbcat, 2020-03-17:
Hello all...
Working on high definition depth sounding fonts. Fun to have a real new feature to dig into.

Question:
On computer screens, font sizes in points are somewhat meaningless. It is a carryover from the print industry, and the conversion from points to pixels is not done well on most widgets libraries.
When rendering soundings, I want to start by following the OHO S52 standards. This says that a depth sounding character should be 2.50 mm in height.
So, here is my request:
I need everyone (with the time) to load some ENCs, turn on text, and adjust the ChartObject font size to yield characters that are 2.5 mm high. I use a caliper to measure on-screen sizes.
Please report:

    Font point size required to yield 2.5 mm.
    Screen size, max horizontal in mm.
    Screen resolution.
    Host OS (linx, win, etc..)

From this I hope to discover the wxWidgets algorithm for converting from points to pixels. It will vary depending on underlying libraries (gtk vs Win10). And, of course, if you run something like a headless rPi over a VNC connection, the information will be of not much direct use.

Thanks for your help
Dave

bdbcat, 2020-03-17:
And for inspiration, I show a functioning prototype of TrueType depth soundings on OpenGL. New on the left, old on the right.
[image]

Hakansv, 2020-03-17:
Dave..
Just so we talk about the same. You say "ChartObject font size" 
Do you mean to adjust Font: "ChartText"  or slider "Chart Object Scale factor" ? Since you refer to size I suppose the Font. I also suppose the measuring would be on a capital?

Hakansv, 2020-03-17:
Dave.. Using my assumptions from above:
Are these what you want? (All system measures are confirmed manually.)
Win10: 
Metrics:  m_display_size_mm: 309     g_Platform->getDisplaySize():  1366:768 
Font size to get 2.5 mm Capitals: 12
[image]
[image]

Hakansv, 2020-03-17:
Dave..
On the RPi4 buster O 5.0
Metrics:  m_display_size_mm: 527     wxDisplaySize:  1920:1080 
The screen width is 527 but due to a black border only 500 is used.
The get capitals to 2.5 mm:
Font: PibotoLt Regular size: 10

leamas, 2020-03-17:
> From this I hope to discover the wxWidgets algorithm for converting from points to pixels.

I might be totally lost here, but  doesn't GetTextExtent() return the size in pixels?  If so, doesn't https://developer.gnome.org/gdk3/stable/GdkDisplay.html#gdk-display-get-primary-monitor and https://developer.gnome.org/gdk3/stable/GdkMonitor.html#gdk-monitor-get-width-mm and friends gives  enough info to translate pixels to mm, assuming gtk3?

Just my 5 öre...

tai-fun, 2020-03-17:
Dave

Measured on seabed NATSUR (clay) using the "l".

Font point size required to yield 2.5 mm -> 9
Screen size, max horizontal in mm. -->Measured 265 mm
(My system reports : 1,024 x 768 Pixels (270 x 203 mm))
Screen resolution. --> 96 x 96 dpi
Host OS (linx, win, etc..) Ubuntu 19.10

Thomas

rgleason, 2020-03-17:
Dave, when changing "ChartText" font size, with ENC charts and "T" "Enc Text" toggled on the text sizes do change, however there are different sized lines of text based upon some hierarchy of importance.  So the real question I have is what size text do you want us use to measure?
The biggest size text?, the smallest?   Do you know how these different sizes are scaled up and down?

leamas, 2020-03-17:
Looking at https://github.com/OpenCPN/OpenCPN/issues/1784#issuecomment-600291614:

Pixels/mm = 203/768 &#8776; 0.264
pixels /2.5 mm = 2.5 / 0.264 &#8776; 9.5 &#8776; 9

Too simple?

bdbcat, 2020-03-18:
Alec...
Your math is good.
The problem is that we ask for fonts only in "Point size", not pixels.  And there is, as we have just demonstrated, only a vague relationship between Point size requested, and pixel size returned.  Thus the question...

bdbcat, 2020-03-18:
In theory, requesting a font by point size should give exactly the same physical size on all systems and displays.  Like it does on paper.
In fact, it does not, except on Android, the most modern of the supported display engines.  Some systems attempt to adjust to physical capabilities.  Others just pass, and hope for the best.

rgleason, 2020-03-18:
From my Post above  "Height for the Depth Soundings now = 1.5 mm"

bdbcat, 2020-03-18:
OK, it is clear that the code will need to iteratively "calibrate" the font logic to produce a fixed physical size on the screen.  We do this elsewhere, for other fixed graphics items.
Anyway, the new code for high-definition soundings fonts is building in github master now.
Comments appreciated.
Dave

Hakansv, 2020-03-18:
Dave..
Win10. First test.
The figures looks nice and clear at a good contrast. A significant difference to the better!
While "Chart Object scale factor" = "0" The font size is closer to to 2 mm than 2.5 mm so you may slightly increase the font calculation factor for this screen type? 
While "...Scale factor" = "1" the font size is close to 3 mm. (Same screen as my Win10 preferences from above.) 
A fail note: If "Graphics style": "Paper chart" is selected the soundings are not visible. At "Simplified" soundings are visible.
[image]
The difference is clear. Here oeSENC for the same area:
[image]

rgleason, 2020-03-18:
Very nice. I find that it is just a little under the 2.5mm, similar to Hakan.
Now I find that the   "d"  taller leg in "sand" is about 2.5 mm when set to Cambria Regular 13. 

[image]

tai-fun, 2020-03-18:
Dave
I agree with Håkan about the looks of your new fonts.
Testing with CM93, and comparing to oeSENC, the new fonts are slightly
smaller for the same setting.
Both Graphics Styles works on Ubuntu.
No emphasized safety contour in the new style, yet.....
I'm quite happy with the present scheme were the font gets gray if deeper
than the safety contour, but I think that a possible tweak were the gray
starts at the "deep" contour instead is OK.

Thomas

Hakansv, 2020-03-18:
Thomas wrote: 
_I'm quite happy with the present scheme were the font gets gray if deeper
than the safety contour_
And the same is very good also for me. (If that's not clear from before?) :-)

rgleason, 2020-03-18:
Oddly running Ubuntu 18.04 in VirtualBox,  with very recent git pull of Opencpn and build, 
The depth soundings measure   2 mm.   I have tried to get the Seabed turned on and it does not seem to work, so that I could adjust the font and get the size that results in 2.5mm. 

I do not know why it would be less on a Ubuntu system than when running windows (Windows seems to have depth that is a little less than 2.5.mm.)

[image]

Hakansv, 2020-03-18:
Dave..
You may have tested yourself on the RPi3-stretch 7 inch screen?
On pure ENC and object scale factor set to "0" the measured depth font hight is actually 2.5 mm.  That's perfectly correct although weird on that small screen area. :) So here we have to play with the scale factor.
Functionally it's as desired, also both Graphics style are working on this Linux as well.

tai-fun, 2020-03-18:
Dave
With the new fonts the subscript - decimal part - of a sounding is more or
less a full font height below the integer number. In the old system the
subscript was about 1/2 a font height below.
I have never before seen such a low subscript. In some way it probably
makes the sounding clearer, and easier to read, but it also looks a bit
disjointed.
Intentional?
Thomas

bdbcat, 2020-03-18:
Tai-fun...
Yes, the subscript looks a little low.  Compare Hakan's post above, showing original oeSENC display.

bdbcat, 2020-03-18:
I forgot to mention:
The sounding font size calibration only works if you physical screen size parameter is correct, either automatically sensed, or manually overridden.

bdbcat, 2020-03-19:
Some tweaks in github now:
1.  Correct color logic to work with dusk/night color schemes.
2.  Slightly increase default font size.
3.  Adjust position of subscript character, if present.
Thanks
Dave

Hakansv, 2020-03-19:
Dave..
Win10
1. The adjusted font size is now to big. At Object scale factor = 0 now 3 mm compared to previous 2 mm. Shot 1
2. Depth figures at smale scales deeper than safe depth are way to dominant. Shot 2.
3. Graphic style:  When using "Paper chart" soundings are not visible on ENC nor oeSENC, but on CM? See shot 3

[image]

[image]

[image]

rgleason, 2020-03-19:
Well, All I can say right now, is that this is going to be much better after fine tuning. Bravo and Thank you Dave and Hakan.

rgleason, 2020-03-19:
Pulled O and built.  I need a magnifier to really see the scale, but using Windows Opencpn the depths appear to be a little over 2.5mm so I think Hakan is right. They need to be scaled back a bit. Thanks.

Hakansv, 2020-03-19:
Rick..
At our age googles will do. I also use this: https://a-ruler-for-windows.en.softonic.com/

rgleason, 2020-03-19:
Thanks very much, have dwnlded.

Hakansv, 2020-03-20:
Dave..
More info about the Graphic style issue:
It seems it's the shift between the two modes that's the cause of the problem not the actual status.
Today soundings are visible when on "Paper chart". When I shift to "Simplified" the soundings disappear. And vice versa.

bdbcat, 2020-03-20:
Hakan....
I sometimes see that.  On restart, all is OK, for me.  But then, switch again, and the soundings are lost again.  I think this is not related to the font change, but I will try to debug it.
You may experiment.  See s52plib.cpp:6328

Hakansv, 2020-03-20:
I agree this is probably not caused by the font job. Only detected by a chance when the font change forced me in here. 
My slow debug method has so far reached to that s52plib::StringToRules() --> RUL_MUL_SG is not called in the bad, no sounding, case. I'll follow further back later on. But you may find it faster?

Hakansv, 2020-03-20:
Dave..
No I can't manage to find this.
A trace? See the two shots. One "Paper Chart" one "Simplified". The view of soundings alternates between cells when I switch Graphics style.
[image]
[image]

bdbcat, 2020-03-21:
Hakan...
Let us move this to Flyspray.
It is not a new problem, by any means.  I think I know what is going on, and will not apply to all charts.  Relates to symbol lookup procedures.  Also fixed by restart, as you know.
Strange that we have never heard this problem from the field....

Hakansv, 2020-03-21:
Dave..
OK, it could be like that. I think no one is playing with the Graphic style anymore? We may make "Paper chart" mode the one and only? 
The reason for me to play with it this time was I added a pure ENC chart set and then the "old one" returned. That the Graphics style and others was reset. 

Commit: "Revert tweak to default soundings font size."
You pushed back from 3.5 to 3.0. On my screen and personal view 3.2 is good:
if(font_size_mm >= (3.2 * scale_factor)){
3.5 yields just over 2.5 mm and although 3.3 yields 2.5 mm as standard says it looks rather big. Again; for me.

Thanks
Håkan

bdbcat, 2020-03-22:
OK, one more tweak, this time to 3.2.
We will see what the general user base thinks when we go to Beta...

closed 2020-04-05 by bdbcat

rgleason, 2020-04-05:
Looks good. Thnks.