OpenCPN/OpenCPN #1785: Plugin Manager - Update Plugin Catalog: Custom... - cannot close window

jongough, 2020-03-18:
I opened Options Plugins and changed 'Choose Remote Catalog' to 'Custom...' then clicked 'Update Plugin Catalog: Custom...' button but with no URL in the 'Custom url' field. This showed an 'OpenCPN Catalog update' dialog box, but there was no way to close it. The following picture shows this:
[image]

jongough, 2021-04-11:
This issue still exists

bdbcat, 2021-05-02:
Strange, works fine here, linux gtk2.  What system details, please?  Anyone else verify on another platform?
```
    if (status != CatalogHandler::ServerStatus::OK) {
        message = _("Cannot download data from url");
        OCPNMessageBox(this, message, _("OpenCPN Catalog update"), wxICON_ERROR);
        return;
    }

```
[image]

Hakansv, 2021-05-02:
I get the upper right "X" on my systems but this dialog is not like most message boxes used within OCPN. We normally not rely on the system's close "X". (I try to message my "students"  -Don't use that "X" in OCPN.)
Why not add a wxOK to pluginmanager.cpp: 5467
OCPNMessageBox(this, message, _("OpenCPN Catalog update"), wxICON_ERROR | wxOK);
And for consistency reasons the same also to the rows 5474 and 5480.
(Or is a PR preffered?)

[image]

bdbcat, 2021-05-03:
Sure, I'll do it.

commit bfdff2c1 (2021-05-03): Ensure ability to close message boxes with wxOK flag.  Closes #1785
closed 2021-05-03 by bdbcat