OpenCPN/OpenCPN #1786: Plugin Manager: Squiddio_pi  has no installation options under Beta or Alpha

rgleason, 2020-03-18:
Since the ampersand issue has settled out now, and those wrinkles are straightened out, and we are using updated catalogs, including the catalog on Alpha  ---> rg-alpha (meta-url)

and since I have added 12 env for Squiddio 1.2 to the rg-alpha (Alpha catalog), 

using Opencpn 5.0.522  built yesterday for windows, when I update Alpha, and go down to Squiddio there is no option to install version 1.2.   Nor is there an option to downgrade.  

Why?   I believe there are older "managed" versions installed and there are some instances of squiddio in Beta and master.

bdbcat, 2020-03-18:
Rick...
I have removed Squiddio from master.  Does not exist in Beta, either.
Try again, please.

rgleason, 2020-03-18:
Maybe there is some "CommonName" issue or the xml is not formatted correctly, but these are the results...  starting with Master ... then Beta ... then Alpha.   Squiddio is always last, is it alphabetical always?

Master Catalog
[image]

[image]

Beta Catalog
[image]

[image]

Alpha Catalog

[image]
[image]
[image]

bdbcat, 2020-03-18:
Rick...
Check the OCPN logfile.  It shows error parsing your XML.

Line 5 is wrong, should not be there.

rgleason, 2020-03-18:
```
11:42:57.422 WARNING catalog_parser.cpp:153 XML or text declaration not at start of entity at line 5

11:42:57.422 WARNING catalog_handler.cpp:168 Cannot parse xml starting with: <?xml version="1.0" ?>
<plugins>
  <version>0.0.0</version>

11:42:57.422 WARNING catalog_handler.cpp:168 Cannot parse xml starting with: <?xml version="1.0" ?>
<plugins>
  <version>0.0.0</version>

Also I have a bunch of these in ProgramData

```
11:43:47.903   ERROR file.cpp:610 can't remove file 'C:\ProgramData\opencpn\MMSINAME.CSV' (error 5: Access is denied.)
11:43:48.184 MESSAGE chart1.cpp:2615 opencpn::MyApp starting exit.

What are those from? Can I remove them?

``````

rgleason, 2020-03-18:
```
<?xml version="1.0" ?>
<plugins>
  <version>0.0.0</version>
  <date>2020-03-17 17:34</date>
  <?xml version="1.0" encoding="UTF-8"?>
  <plugin version="1">
		<name> VDR </name>
		<version> 1.2.0 </version>
		<release> 1 </release>
		<summary> Voyage Data Recorder and Player </summary>

		<api-version> 1.16 </api-version>
		<open-source> no </open-source>
		<author> Jean-Eudes Onfray </author>
		<source> https://github.com/SethDart/vdr_pi/ </source>
		<info-url>https://opencpn.org/OpenCPN/plugins/vdr.html </info-url>

```

Dave, there is nothing wrong with 

 ` <plugin version="1">`

As I said before, it matches the  ocpn-plugins.xsd in oesenc_pi and what is used in Beta ocpn-plugins.xml   _exactly_

rgleason, 2020-03-18:
I see now,  line 5 is

`  <?xml version="1.0" encoding="UTF-8"?>
`
And should be somewhere else above that.

rgleason, 2020-03-18:
```
<?xml version="1.0" ?>
<plugins>
<version>0.0.0</version>
<date>2020-03-17 20:08</date>
    <plugin version="1">
              <name> WeatherRouting </name>
```

The top of ocpn-plugins.xsd  is

```
<?xml version="1.0"?>
<xs:schema xmlns:xs="http://www.w3.org/2001/XMLSchema">

<xs:element name="plugins">
  <xs:complexType>
    <xs:sequence>
      <xs:element name="version" type="xs:string"/>
      <xs:element name="date" type="xs:string"/>
      <xs:element ref="plugin" maxOccurs="unbounded"/>
   </xs:sequence>
  </xs:complexType>
</xs:element>

<xs:element name="plugin">
  <xs:complexType>
    <xs:choice>
      <xs:sequence>
        <xs:element name="meta-url" type="xs:token" maxOccurs="unbounded"/>
      </xs:sequence>
      <xs:group ref="metadata"/>
    </xs:choice>
    <xs:attribute name="version" type="xs:string" use="required"/>
  </xs:complexType>
</xs:element>
```

So it looks like 

`<?xml version="1.0" encoding="UTF-8"?>`

Should just be deleted as you said.

bdbcat, 2020-03-18:
It is duplicate(almost) of line 1. Either will do, but not both.

rgleason, 2020-03-18:
Sorry, when doing this manually, you must be very very careful. It is not easy, and the longer the file gets the more careful you have to be.

I think we need better indents in the formatting, so that each plugin's environment metadata is clearly delineated. That would make it easier for manual formatting and visually when checking.

rgleason, 2020-03-18:
Here is how it works now...much better..

[image]

[image]

That is much better!  I've made every mistake possible at this point ...using "&" in the metadata summary, causing Jon Gough to hang up and crashing my O, having an extra line in the ocpn-plugin.xml header, so that "Managed" plugins don't show as managed.  What else can happen?
....What happens if there is no internet connection or it gets pulled out and we are using the Alpha, or Beta branch?   or even the Master Branch for that matter?  Is there a back copy of the ocpn-plugins.xml that is stored to make this menu work, then when the user clicks to download a message can be delivered,... no internet?   That way at least we will be able to use our plugins when off line.

rgleason, 2020-03-18:
`Making sure the output of the error checker makes sense and identifies the bad filename, and having a message to developers to fix this before continuing, and not to do it manually.`

This needs to be done by someone. IMHO

leamas, 2020-03-18:
> Sorry, when doing this manually, you must be very very careful. It is not easy, and the longer the file gets the more careful you have to be.

You should be able to use xmllint to verify the xml file, just as `make validate` does.

If we don't have an automated way to verify that the xml file is correct the process is broken and  should be fixed. xmllint can always be used to verify basic xml syntax even if xsd file does not apply; I didn't think about "external" includes when designing the xsd.

rgleason, 2020-03-18:
-  I didn't think about "external" includes when designing the xsd.  - What significance is this?
- xmllint is linux unfortunately, I will look for an alternative. After further reading I believe xmllint can be a windows command line tool.  Also you can use[ Notepad++ with ](https://softwarerecs.stackexchange.com/questions/7076/windows-light-xml-validator-works-with-non-xml-files)a plugin [XML Tools](https://sourceforge.net/projects/npp-plugins/files/XML%20Tools/)
 
> If we don't have an automated way to verify that the xml file is correct the process is broken and should be fixed. 
OK that would be great.
1. Most of us when seeing the output will not know which file the errors are coming from. Therefore I had to check 12 files.  It was easier to just review by eye and cut and paste the xml into ocpn-plugins because nothing was working and I did not know where to look for that error  (NO FILENAME in the output!) 
2.  Making sure the output of the error checker makes sense and identifies the bad filename,
3.  Having a message to developers to fix this before continuing, and not to do it manually.

leamas, 2020-03-18:
> xmllint is linux unfortunately

No. It's part of the xsltproc package, readily available in chocolatey

> Most of us when seeing the output will not know which file the errors are coming from. Therefore I had to check 12 files

At least I will not patch xmllint, for sure. 

The normal way to handle such things (given that you run the check on a complete ocpn-plugins.xml) is to use git. Normally, you should certainly not add 12 files in one commit. Adding them one by one, running the checks in between (it's fast!) should make the trick.
 
And you can always run xmllint on the separate files. Not verifying the xsd, but the basic syntax. Which would have been enough in this case.

rgleason, 2020-03-18:
The ocpn-plugins.xml file is removed and the process does not complete at all and make a new ocpn-plugins.xml  if there are errors!!!  You should work your way through the process as I have.

Alec it is not fast to separately add xml files, and I am not going to do it that way for sure.

leamas, 2020-03-18:
hm... could you expand on the steps you do when creating a new ocpn-plugins.xml?

rgleason, 2020-03-18:
I have to download all the xml files from cloudsmith, then go to my local github plugins/  branch rg-alpha or  branch: Beta

Then copy the files into metadata.  Then from plugins/  execute the python command with --force.
The gibberish + Line number come up if there is an error with no filename.  Now I have to check all 12 env metadata.xml files.  Faster to check by hand and paste into ocpn-xml, but if we knew what the filename with the error was, it would be better doing it that way.

rgleason, 2020-03-18:
Let us build the ocpn-plugins.xml without interference.  Then let use run the checker on the whole file.

leamas, 2020-03-18:
I wouldn't work like that. I would start as you, but run xmliint on all downloaded files before trying to construct the complete ocpn-plugins.xml. You could script this to simplify the process.

ocpn-metadata parses the xml tree, and just cannot work  with broken xml as input. The only feasible way is to verify the xml before trying to merge (i. e., to use ocpn-metadata).

leamas, 2020-03-18:
Thinking more on this:

  - Yes, ocpn-metadata fails if input isn't well-formed. This is probably as it should, and an indication to run xmllint on the input. xmllint has proper error handling, pointing out offending line and actual error. No point making half-baked solutions in ocpn-metadata for this.
  - It might be argued that the plugin CMakeLists should have an new target like *verify-xml* so  this check could be run earlier. To discover XML syntax errors after uploading/downloading seems a bit late.

rgleason, 2020-03-18:
This is turning into another pain. Plugin Developers arent going to like

rgleason, 2020-03-18:
I given you my opinion and you've chosen to not listen. Closing.

rgleason, 2020-03-18:
Ok, I am going to disable your checker, make a single file and then run it through the checker as I wanted it.

bdbcat, 2020-03-18:
Rick....
Ummm.....
Any of the twelve files will do.  If it is an XML syntax error, they all will have the same error, since they all are derived from the same xxx-plugin.xml.in during the CI build process.

So the error, if error there be, is to be fixed once in xxx-plugin.xml.in.
Dave

bdbcat, 2020-03-19:
Please close this issue.  Original bug report is resolved.

rgleason, 2020-03-19:
It is closed

closed 2020-03-19 by bdbcat