OpenCPN/OpenCPN #1787: Moving Options Menu, shifts to and changes Opencpn Window-Min- Max  in Ubuntu 18.04

rgleason, 2020-03-18:
Grab the head of the Options window and try to drag or move it. Somehow the Opencpn Window always gets involved and Opencpn goes from full screen mode to something else and is being dragged around.

[image]

[image]

bdbcat, 2020-03-18:
Rick...
I concur.
There have been, and still are, numerous problems with gnome desktop interface to wxWidgets.  Probably nothing can be done about this.
@tai-fun...
How is 19.10?
Dave

rgleason, 2020-03-19:
Should I close it?

bdbcat, 2020-03-19:
Nope, leave it open as a reminder.

tai-fun, 2020-03-19:
Dave
I'm no fan of Gnome desktop, I use KDE.
Never seen this problem.
Thomas

leamas, 2020-04-01:
This might be related to gnome settings. Using gnome-tweaks, there are _Windows | Attach Modal Dialogs_ and _Windows | Edge  Tiling_. 

I suspect this behaviour is a combination of these two settings.

rgleason, 2020-05-05:
Should I close this now?

bdbcat, 2020-05-05:
Leave it open for future reference

rgleason, 2020-10-17:
I think this is fixed and can be closed?

rgleason, 2021-03-05:
I think this was fixed. Can it be closed?

rgleason, 2021-04-28:
April 28 build of Ov5.5  Ubuntu 18.04   
This is still an issue.  Grab the top of Options and drag it.  OpenCPN resizes from Max to a smaller window, and OpenCPn and Options get moved around, rather than just options.

I think this should be fixed if possible.

bdbcat, 2021-05-03:
Rick...
A Desktop Manager workaround is required.  Not possible to change behavior in OCPN.

For Ubuntu 18.04, Gnome desktop:
One should install the package "Gnome Tweaks", and change the setting "Windows->AttachModalDialogs" to OFF.

Closing issue.

closed 2021-05-03 by bdbcat

rgleason, 2021-05-03:
Would this be something for the manual?  or not.

bdbcat, 2021-05-03:
Yes, though I do not know where best.  "Tips and tricks", if we have such?

rgleason, 2021-05-03:
FAQ?  Maybe?

rgleason, 2021-05-03:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#how_to_fix_ubuntu_1804_options_menu_display_anomyly