OpenCPN/OpenCPN #1791: Question about a Segfault error in OpenCPN/oesENC

hreuver0183, 2020-03-21:
I have been looking into a problem I ran into today.

Under load my system Segfaulted on zooming in on an ENC-chart. It occured before but now I managed to recreate the problem more or less.

Details:
- I had some wrong settings from tinkering around, like Vectorchart Zoom / Scale factor set to 5 (yeah, I know it is insane).
- On the background Firefox was running with KIP
- In firefox quite a few links were open since I was lax in cleaning up (and some xterms)
- also on the background the i2c, pypilot and canboat were running
- Using "free" I found with OpenCPN the free memory was about 700-800MB, after crashing about 1200-1400MB (1400MG form 4GB).
- No cache (makes no sence with emmc or sd).

The system: Odroid N2 (ARM64 / 4 GB / 32 GB / Ubuntu 18.04 / OpenCPN 5.22 / oesENC 4.2.0 / o-charts via FPR).

Just to see if it has something to  with the platform (arm64) I ran a test with an RPI3 (armhf / 1GB / 64GB / Openplotter 2.0pre / OpenCPN 5.0.0 / oesENC-4.0.0 / o-charts via USB dongle, same chart).
The fault was somehow reproduceable (using chrome instead of firefox).
No I2C, no IMU, no can available during tests.
I also set the monitor in the RPI3 to a manual size bigger than fits in my boat (since some ENC charts work with the fysical screen size).
However it was much harder and the only output I got was the first s52plib line.
Also the segfault roughly corresponds with the memory being exhausted, with "free" yielding 500MB free memory without OpenCPN.

I think it was harder because the 1GB limit of the RPI3 does have some hard limit of loading. Overload and the RPI3 simply halts. 

Things I saw, somehow the problem occurs under heavy load. 
On RPI3 the problem seems less (perhaps in reference to other limits).
On the arm64, opencpn segfaults long before the memory is exhausted.

For now I am carefull not to run OpenCPN when multiple Firefox session are open and I raised the ulimit settings (soft limit for open files 1024 => 2048) for OpenCPN using a small opencpn-start script.

My questions:
What could cause this segfault?
Is it a kernel problem or an OpenCPN problem?
Shouldn't there be some checks after increasing the claimed system memory before a Segfault occurs?
Basically I raised the open files ulimits, but I don't think the problem is there.

Perhaps someone can point me to some gdb commands I can use to produce more helpfull information.

PS during the test I first updated my oesenc plugin on ARM64 which resulted in a one-time message "the system changed, key is invalid". Very disturbing.
PPS during the RPI3 test I got the same message and after a bit of tinkering and a few restarts of OpenCPN the USB dongle was finally recognized. 

Is there an easy validation "my FPR/USB dongle" is recognized correctly?
In other words, is there a validation where the result could be "you screwed up your system, FPR lost", presented in a more non-committal way.

With regards,
Huub

====== gdb output info stack ======

`Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x0000007f9c4791dc in s52plib::RenderLCLegacy(_ObjRazRules*, _Rules*, ViewPort*) () from /usr/lib/opencpn/liboesenc_pi.so
(gdb) info stack
#0  0x0000007f9c4791dc in s52plib::RenderLCLegacy(_ObjRazRules*, _Rules*, ViewPort*) () at usr/lib/opencpn/liboesenc_pi.so
#1  0x0000007f9c47993c in s52plib::RenderLC(_ObjRazRules*, _Rules*, ViewPort*) () at /usr/lib/opencpn/liboesenc_pi.so
#2  0x0000007f9c48047c in s52plib::DoRenderObject(wxDC*, _ObjRazRules*, ViewPort*) () at /usr/lib/opencpn/liboesenc_pi.so
#3  0x0000007f9c449988 in eSENCChart::DCRenderLPB(wxMemoryDC&, PlugIn_ViewPort const&, wxRect*) () at /usr/lib/opencpn/liboesenc_pi.so
#4  0x0000007f9c44e60c in eSENCChart::DCRenderRect(wxMemoryDC&, PlugIn_ViewPort const&, wxRect*) () at /usr/lib/opencpn/liboesenc_pi.so
#5  0x0000007f9c45913c in eSENCChart::DoRenderViewOnDC(wxMemoryDC&, PlugIn_ViewPort const&, bool) () at /usr/lib/opencpn/liboesenc_pi.so
#6  0x0000007f9c459afc in eSENCChart::RenderRegionView(PlugIn_ViewPort const&, wxRegion const&) () at /usr/lib/opencpn/liboesenc_pi.so
#7  0x0000007f9c445dcc in eSENCChart::RenderRegionViewOnDCNoText(PlugIn_ViewPort const&, wxRegion const&) () at /usr/lib/opencpn/liboesenc_pi.so
#8  0x000000555598cfd0 in ChartPlugInWrapper::RenderRegionViewOnDCNoText(wxMemoryDC&, ViewPort const&, OCPNRegion const&) ()
#9  0x00000055559e49ac in Quilt::DoRenderQuiltRegionViewOnDC(wxMemoryDC&, ViewPort&, OCPNRegion&) ()
#10 0x0000005555851dcc in ChartCanvas::OnPaint(wxPaintEvent&) ()
#11 0x0000007fb722c480 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /usr/lib/aarch64-linux-gnu/libwx_baseu-3.0.so.0
#12 0x0000007fb722c5d8 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /usr/lib/aarch64-linux-gnu/libwx_baseu-3.0.so.0
#13 0x0000007fb722ca9c in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/aarch64-linux-gnu/libwx_baseu-3.0.so.0
#14 0x0000007fb722cb6c in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/aarch64-linux-gnu/libwx_baseu-3.0.so.0
#15 0x0000007fb722cc88 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/aarch64-linux-gnu/libwx_baseu-3.0.so.0
#16 0x0000007fb722c884 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/aarch64-linux-gnu/libwx_baseu-3.0.so.0
#17 0x0000007fb7573500 in wxWindow::GTKSendPaintEvents(_GdkRegion const*) () at /usr/lib/aarch64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#18 0x0000007fb7573c2c in  () at /usr/lib/aarch64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#19 0x0000007fb5e393a4 in  () at /usr/lib/aarch64-linux-gnu/libgtk-x11-2.0.so.0
#20 0x0000007fffffe4a0 in  ()
(gdb) exit`

hreuver0183, 2020-03-25:
Doubles with https://github.com/bdbcat/oesenc_pi/issues/74

closed 2020-03-25 by hreuver0183

bdbcat, 2020-03-26:
Just to by crystal clear:
I assume from the gdb trace that you are NOT using OpenGL.
Is that correct?