OpenCPN/OpenCPN #1799: O loads same unmanged plugin twice

transmitterdan, 2020-03-30:
On Raspberry Pi 4 (Bionic) we see two set of icons for unmanaged plugin VDRplayer. Both sets of icons are functional.

[image]

First time O loads the plugin:
```
04:16:11.446 MESSAGE pluginmanager.cpp:2078 PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libvdr_pi.so
```
Second time O loads same plugin:
```
04:18:26.995 MESSAGE pluginmanager.cpp:2078 PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libvdr_pi.so
```
Relevant section of opencpn.log:

```
04:10:59.485 MESSAGE chart1.cpp:2675 opencpn::MyApp exiting cleanly...

04:16:00.990 MESSAGE chart1.cpp:1851 

________

04:16:00.990    INFO logger.cpp:169 ------- OpenCPN version 5.0.522 restarted at 2020-03-30 -------

04:16:00.990    INFO logger.cpp:169 Using loglevel INFO
04:16:00.990 MESSAGE chart1.cpp:1877 wxWidgets version: wxWidgets 3.0.4 Linux 32 bit wxGTK
04:16:01.019 MESSAGE chart1.cpp:1882 OS Version reports as:  4.19
04:16:01.020 MESSAGE chart1.cpp:1884 MemoryStatus:  mem_total: 3917 mb,  mem_initial: 99 mb
04:16:01.027 MESSAGE chart1.cpp:1893 SData_Locn is /home/pi/.vs/opencpn/d2d28a1f-d981-4089-8a6f-d2917fec4f66/out/install/Linux-Debug/share/opencpn/
04:16:01.027 MESSAGE chart1.cpp:1909 PrivateDataDir is /home/pi/.opencpn
04:16:01.027 MESSAGE chart1.cpp:1966 Using existing Config_File: /home/pi/.opencpn/opencpn.conf
04:16:01.046 MESSAGE navutil.cpp:1193 winPluginDir, read from ini file: 
04:16:01.046 MESSAGE navutil.cpp:1286 Setting Ownship Lat/Lon 37.8991, -122.807
04:16:01.048 MESSAGE navutil.cpp:3143 Switch InlandEcdis mode Off
04:16:01.058 MESSAGE styles.cpp:854 Styles loading from /home/pi/.vs/opencpn/d2d28a1f-d981-4089-8a6f-d2917fec4f66/out/install/Linux-Debug/share/opencpn/uidata/styles.xml
04:16:01.081 MESSAGE styles.cpp:831 No styles found at: /home/pi/
04:16:01.084 MESSAGE styles.cpp:831 No styles found at: /home/pi/.opencpn/
04:16:01.122 MESSAGE styles.cpp:786 Using SVG Icons
04:16:01.122 MESSAGE chart1.cpp:2024 Detected display size (horizontal): 507 mm
04:16:01.122 MESSAGE OCPNPlatform.cpp:890 Getting DefaultSystemLocale...
04:16:01.122 MESSAGE chart1.cpp:2061 System default Language:  en_US
04:16:01.122 MESSAGE chart1.cpp:2064 Config file language:  en_US
04:16:01.122 MESSAGE chart1.cpp:2069 Adjusted App language:  en_US
04:16:01.122 MESSAGE OCPNPlatform.cpp:975 ChangeLocale: Language load for:  en_US
04:16:01.125 MESSAGE OCPNPlatform.cpp:1027 ChangeLocale: Locale Init OK for:  en_US
04:16:01.125 MESSAGE OCPNPlatform.cpp:1046 Loading catalog for opencpn core.
04:16:01.126 MESSAGE OCPNPlatform.cpp:890 Getting DefaultSystemLocale...
04:16:01.126 MESSAGE chart1.cpp:2077 Opencpn language set to:  en_US
04:16:01.127 MESSAGE ConfigMgr.cpp:685 Loading Configs catalog: /home/pi/.opencpn/Configs/configs.xml
04:16:01.135 MESSAGE chart1.cpp:2311 Creating MyFrame...size(1920, 1028)  position(0, 88)
04:16:01.148 MESSAGE chart1.cpp:2878  **** Connect stuff
04:16:03.880 MESSAGE chcanv.cpp:970 Creating glChartCanvas
04:16:03.984 MESSAGE chcanv.cpp:2335 Metrics:  m_display_size_mm: 507     g_Platform->getDisplaySize():  1920:1080   
04:16:03.984 MESSAGE chcanv.cpp:2340 wxDisplaySize(): 1920 1080
04:16:03.985 MESSAGE TCDataSource.cpp:52 Loading Tide/Current data source: /usr/local/share/opencpn/tcdata/HARMONIC.IDX
04:16:04.472 MESSAGE glChartCanvas.cpp:1212 OpenGL-> Renderer String: V3D 4.2
04:16:04.472 MESSAGE glChartCanvas.cpp:1221 OpenGL-> Version reported:  2.1 Mesa 19.3.2
04:16:04.477 MESSAGE glChartCanvas.cpp:1315 OpenGL-> Texture rectangle format: de1
04:16:04.477 MESSAGE glChartCanvas.cpp:1360 OpenGL-> Using Vertexbuffer Objects
04:16:04.525 MESSAGE glChartCanvas.cpp:1178 OpenGL Framebuffer OK, size = 1918
04:16:04.604 MESSAGE glChartCanvas.cpp:1441 OpenGL-> Using Framebuffer Objects
04:16:04.604 MESSAGE glChartCanvas.cpp:1444 OpenGL-> Using FBO Stencil buffer
04:16:04.604 MESSAGE glChartCanvas.cpp:1450 OpenGL-> Using Stencil buffer clipping
04:16:04.604 MESSAGE glChartCanvas.cpp:1455 OpenGL-> Using Scissor Clipping
04:16:04.604 MESSAGE glChartCanvas.cpp:1610 OpenGL-> Not Using compression
04:16:04.604 MESSAGE glChartCanvas.cpp:1465 OpenGL-> Minimum cartographic line width:  1.0
04:16:04.604 MESSAGE glChartCanvas.cpp:1467 OpenGL-> Minimum symbol line width:  1.0
04:16:04.609 MESSAGE gshhs.cpp:1377 Loading World Chart Q=0 in 0 ms.
04:16:04.609 MESSAGE gshhs.cpp:179 Background world map loaded from GSHHS datafiles found in: /usr/local/share/opencpn/gshhs/
04:16:04.837 MESSAGE chartdb.cpp:237 ChartDB Cache policy:  Application target is 1024 MBytes
04:16:04.839 MESSAGE chartdbs.cpp:145 Loading chart db version: V018
[... 344 lines cut ...]
04:18:26.994 MESSAGE pluginmanager.cpp:1019 PlugInManager searching for PlugIns in location /usr/local/lib/opencpn
04:18:26.995 MESSAGE pluginmanager.cpp:1056 Found 6 candidates
04:18:26.995 MESSAGE pluginmanager.cpp:1060 Checking plugin candidate: /usr/local/lib/opencpn/libwmm_pi.so
04:18:26.995 MESSAGE pluginmanager.cpp:1060 Checking plugin candidate: /usr/local/lib/opencpn/libdashboard_pi.so
04:18:26.995 MESSAGE pluginmanager.cpp:1060 Checking plugin candidate: /usr/local/lib/opencpn/libchartdldr_pi.so
04:18:26.995 MESSAGE pluginmanager.cpp:1060 Checking plugin candidate: /usr/local/lib/opencpn/libstatusbar_pi.so
04:18:26.995 MESSAGE pluginmanager.cpp:1060 Checking plugin candidate: /usr/local/lib/opencpn/libvdr_pi.so
04:18:26.995 MESSAGE pluginmanager.cpp:2015 Plugin is compatible: true
04:18:26.995 MESSAGE pluginmanager.cpp:2078 PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libvdr_pi.so
04:18:26.996    INFO pluginmanager.cpp:2249 PlugInManager:  /usr/local/lib/opencpn/libvdr_pi.so
04:18:26.996    INFO pluginmanager.cpp:2250         Plugin common name: VDR
04:18:26.996    INFO pluginmanager.cpp:2252         API Version detected: 106
04:18:26.996    INFO pluginmanager.cpp:2253         PlugIn Version detected: 0.3
04:18:27.000 MESSAGE pluginmanager.cpp:1060 Checking plugin candidate: /usr/local/lib/opencpn/libgrib_pi.so
04:18:27.002 MESSAGE OCPNPlatform.cpp:975 ChangeLocale: Language load for:  en_US
04:18:27.004 MESSAGE OCPNPlatform.cpp:1027 ChangeLocale: Locale Init OK for:  en_US
04:18:27.004 MESSAGE OCPNPlatform.cpp:1040 Loading catalog for:  opencpn-wmm_pi
04:18:27.004 MESSAGE OCPNPlatform.cpp:1040 Loading catalog for:  opencpn-dashboard_pi
04:18:27.005 MESSAGE OCPNPlatform.cpp:1040 Loading catalog for:  opencpn-chartdldr_pi
04:18:27.006 MESSAGE OCPNPlatform.cpp:1040 Loading catalog for:  opencpn-vdr_pi
04:18:27.006 MESSAGE OCPNPlatform.cpp:1040 Loading catalog for:  opencpn-grib_pi
04:18:27.007 MESSAGE OCPNPlatform.cpp:1040 Loading catalog for:  opencpn-vdr_pi
04:18:27.008 MESSAGE OCPNPlatform.cpp:1046 Loading catalog for opencpn core.
04:18:27.008 MESSAGE OCPNPlatform.cpp:890 Getting DefaultSystemLocale...
```

rgleason, 2020-03-30:
Current Source for VDR_pi  for PI Manager version
https://github.com/rgleason/vdr_pi/tree/ci   Branch: ci

Search for "GetCommonName" in the vdr_pi.cpp file

```
wxString vdr_pi::GetCommonName()
{
      return _("VDR");
}

```
So this is used for the <name> field in the  plugins xml file
See [LN 245 in CMakeLists.txt](https://github.com/rgleason/vdr_pi/blob/ci/CMakeLists.txt) 
So in [VDR_plugin.xml.in ](https://github.com/rgleason/vdr_pi/blob/ci/VDR-plugin.xml.in) 
The name field has  "VDR"

```
<?xml version="1.0" encoding="UTF-8"?>
<opencpn-plugin version="1">
  <name> VDR </name>
  <version> ${PACKAGE_VERSION} </version>
  <release> ${PKG_RELEASE} </release>
  <summary> Voyage Data Recorder & Player </summary> 

```

Now I note that the  opencpn-plugin version="1"  delimiter should be 

`<plugin version= "1">`

Which will change when we update Frontend 1 from Dave's oeSENC
However I have changed this manually in my meta-url   ocpn-plugins.xml file located here:

https://github.com/rgleason/plugins/blob/rg-alpha/ocpn-plugins.xml

Therefore I wouldn't think that you should be having this double instance problem, so I'm very "puzzled"

bdbcat, 2020-03-30:
Dan...
Does this happen always, from initial startup?
Or does it happen only after you access settings->Plugins ??

rgleason, 2020-03-30:
Windows VDR 1.2 works as expected.

rgleason, 2020-03-30:
RPI (bionic)?  I don't think we have that.   Just stretch and buster, 10 and 9.4

```
  <plugin version="1">
    <name> VDR </name>
    <version> 1.2.0 </version>
    <release> 1 </release>
    <summary> Voyage Data Recorder and Player </summary>
    <api-version> 1.16 </api-version>
    <open-source> no </open-source>
    <author> Jean-Eudes Onfray </author>
    <source> https://github.com/SethDart/vdr_pi/ </source>
    <info-url>https://opencpn.org/OpenCPN/plugins/vdr.html </info-url>
    <description>
VDR Voyage Data Recorder Use Voyage Data Recorder (VDR) to record and play NMEA files.
Save NMEA stream to a file. Replay NMEA stream previously saved. Plugin by Seth Dart
</description>
    <target>raspbian</target>
    <target-version>10</target-version>
    <tarball-url> https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-prod/raw/names/VDR-raspbian-10-tarball/versions/v0.5.0/vdr_pi-0.5.0-1_raspbian-10.tar.gz </tarball-url>
  </plugin>
  <plugin version="1">
    <name> VDR </name>
    <version> 1.2.0 </version>
    <release> 1 </release>
    <summary> Voyage Data Recorder and Player </summary>
    <api-version> 1.16 </api-version>
    <open-source> no </open-source>
    <author> Jean-Eudes Onfray </author>
    <source> https://github.com/SethDart/vdr_pi/ </source>
    <info-url>https://opencpn.org/OpenCPN/plugins/vdr.html </info-url>
    <description>
VDR Voyage Data Recorder Use Voyage Data Recorder (VDR) to record and play NMEA files.
Save NMEA stream to a file. Replay NMEA stream previously saved. Plugin by Seth Dart
</description>
    <target>raspbian</target>
    <target-version>9.4</target-version>
    <tarball-url> https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-prod/raw/names/VDR-raspbian-9.4-tarball/versions/v0.5.0/vdr_pi-0.5.0-1_raspbian-9.4.tar.gz </tarball-url>
  </plugin>

```

Hakansv, 2020-03-30:
I tried agian on Win10:
From Catalog: Beta  VDR 1.2 was availble to install over un-managed v- 0.2.
I clicked Install -- and then OK >>> O crashed.
Reopened O and Activated the VDR plugin. Now only one set of VDR icons. and no fake radar puzzle.
So far so good.
Now tried to update master catalog:
After that neither VDR nor radar_pi in the plugin list!?
Went back to Beta catalog and both VDR and radar_pi returned. So it seams that installed beta managed versions is not listed when I show available master versions. Hmm..

The crash log:
`17:26:37.007 MESSAGE pluginmanager.cpp:2197 PlugInManager: Loading PlugIn: C:\Users\admin\AppData\Local\opencpn\plugins\vdr_pi.dll
17:26:37.024    INFO pluginmanager.cpp:2368 PlugInManager:  C:\Users\admin\AppData\Local\opencpn\plugins\vdr_pi.dll
17:26:37.024    INFO pluginmanager.cpp:2369         Plugin common name: VDR
17:26:37.024    INFO pluginmanager.cpp:2371         API Version detected: 106
17:26:37.024    INFO pluginmanager.cpp:2372         PlugIn Version detected: 1.2
.....
...
17:26:37.309 MESSAGE pluginmanager.cpp:1179 Checking plugin candidate: C:\Program Files (x86)\OpenCPN\plugins\ofc_pi.dll
17:26:37.309 MESSAGE pluginmanager.cpp:1179 Checking plugin candidate: C:\Prog`
There O crashed!

transmitterdan, 2020-03-30:
@bdbcat 

I noticed two sets of icons in the toolbar when O opened.  But only one instance in the plugin manager screen.  I cannot reproduce this issue with latest git master commit.

transmitterdan, 2020-03-30:
> RPI (bionic)? I don't think we have that. Just stretch and buster, 10 and 9.4

My apologies, it is buster not bionic. Anyway, I cannot reproduce using latest git master.

closed 2020-03-30 by transmitterdan