OpenCPN/OpenCPN #1802: Plugin Manager - incorrect installation location when running OCPN in portable mode

jongough, 2020-03-31:
Starting OCPN with the "-p" flag puts it into portable mode and all the programs and data it needs are supposed to be contained in the directory from which OCPN was launched. Using Plugin Manger it installs files in the default, non-portable, locations which then at runtime cannot be found. 

Running ODraw by installing through Plugin Manger it installs OK but the icons are in the wrong location. The are installed into $USER/.local/share/opencpn/plugins/ocpn_draw_pi , however, when looking for these they are not found in $USER/opencpn/OpenCPN-5.0/build-kde181/plugins/ocpn_draw_pi/data , which is where they should have been installed.

Here is the log file showing the issue
[log of running OCPN in portable mode and icons being put in wrong location.txt](https://github.com/OpenCPN/OpenCPN/files/4411874/log.of.running.OCPN.in.portable.mode.and.icons.being.put.in.wrong.location.txt)

bdbcat, 2020-04-01:
There are several problems with portable mode.
Working on it.....

jongough, 2020-04-01:
I have now run the latest version of OD on the latest master build on Ubuntu 18.04 and the same issue occurs. Here is the log of this:
[log of running OCPN no Ubuntu 18.04 and icons being put in wrong location.txt](https://github.com/OpenCPN/OpenCPN/files/4413242/log.of.running.OCPN.no.Ubuntu.18.04.and.icons.being.put.in.wrong.location.txt)

bdbcat, 2020-04-01:
Jon...
For managed plugins, you should use the newer API -16 function to find your local data.
In ODIcons.cpp, you now have:

```
   fn.SetPath(*GetpSharedDataLocation());
    fn.AppendDir( wxT("plugins") );
    fn.AppendDir(wxT("ocpn_draw_pi"));
    fn.AppendDir(wxT("data"));
    g_SData_Locn = new wxString(fn.GetFullPath().c_str());

This should be changed to:

   fn.SetPath(GetPluginDataDir( "ocpn_draw_pi"));
    fn.AppendDir(wxT("data"));
    g_SData_Locn = new wxString(fn.GetFullPath().c_str());

```

jongough, 2020-04-01:
I have put in the new code but it returns an incorrect location:
```
09:48:31.472 MESSAGE pluginmanager.cpp:486 PlugInManager: Using data dirs from: /home/jon/opencpn/OpenCPN-5.0/build-kde181plugins
```
There is a '/' missing betweek build-kde181 & plugins

I have submitted a pull request for this.

bdbcat, 2020-04-01:
Jon...
I accpted the PR, but I suspect it will still be broken in portable mode, since I did my testing before you patched your code.

So, please try to avoid portable mode for now, and confirm that the install works properly in standard mode.  PR that code to plugins master, and then I will touch-up portable mode.

bdbcat, 2020-04-06:
Close?

jongough, 2020-04-07:
I am checking it out.

transmitterdan, 2021-02-05:
I don't think very many people use portable mode on the pi.  Reason being that the microSD card is in effect "portable".  So all you need to have a portable OpenCPN is just carry around the microSD card with everything on it.

So I guess what I am asking is, what is the use case for portable mode on pi?  I will try to figure it out based on the use case.

Bergum, 2021-02-05:
I'm going to utilize bouth hdmi screens to display the same map in two difrent resolution.
I have one map/screen to show details around me, and another for ais/other boats/more ahead.

I have 2x 7" screens mounted over the cockpit.

transmitterdan, 2021-02-05:
I don't know if portable mode on Linux supports that use case.  Have to investigate.  Please be patient.

transmitterdan, 2021-02-05:
I have not tested this much but it seems to work for me:

1. Decide location for portable OpenCPN folder:  My test folder is ~/transmitterdan/opencpn/portable but you can use any folder I think.
2. Have a current version of OpenCPN installed on the pi.  You may have to check the actual install folders because mine may be different than yours.  My opencpn is installed in /usr/local but yours will probably be somewhere else.  So you will substitute /usr/local for wherever your install folder is located.  It may be simply /usr. A little use of ls /usr/bin or /usr/share will give you a hint.  You need to copy 2 folders and the executable itself.
3. cp -R /usr/local/share/opencpn/. ~/transmitterdan/opencpn/portable/ 
4. cp -R /usr/local/lib/opencpn/. ~/transmitterdan/opencpn/portable/plugins/lib
5. cp /usr/local/bin/opencpn ~/transmitterdan/opencpn/portable/.
6. cd ~/transmitterdan/opencpn/portable
7. ./opencpn -p

If you ever update opencpn you will probably need to recopy these folders to your portable folder to be consistent.

Hope this helps.  If not send detailed questions.  I may not answer for a day or so.

transmitterdan, 2021-02-05:
Close?

closed 2021-04-26 by jongough