OpenCPN/OpenCPN #1803: Plugin Manager - Uninstall of plugin under Ubuntu 18.04 causes segmentation fault

jongough, 2020-04-01:
I have just installed OD on the latest master of OCPN (a0ea779e). There were issues with the icons being loaded into the wrong location. Attempting uninstall the plugin from the manger resulted in a segmentation fault, these are what is shown when running under gdb:
```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
PluginListPanel::ReloadPluginPanels (this=0x0, plugins=0x5555564eaba0) at /data/public/working area/opencpn/OpenCPN-5.0/src/pluginmanager.cpp:5245
5245	    Hide();
```
Here is the backtrace:
```
(gdb) bt
#0  0x00005555559df53d in PluginListPanel::ReloadPluginPanels(ArrayOfPlugIns*) (this=0x0, plugins=0x5555564eaba0)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/pluginmanager.cpp:5245
#1  0x00005555559e7519 in pluginUtilHandler::OnPluginUtilAction(wxCommandEvent&) (this=<optimised out>, event=...)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/pluginmanager.cpp:703
#2  0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#3  0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#4  0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#5  0x00007ffff58cda13 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#10 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#11 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#15 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#16 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#17 0x00007ffff5e42cf3 in wxDialog::ShowModal() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#18 0x00005555557f6480 in MyFrame::DoOptionsDialog() (this=0x55555644ada0) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:6131
#19 0x00005555557f6db9 in MyFrame::DoSettings() (this=0x55555644ada0) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5032
#20 0x00005555557f78b1 in MyFrame::DoSettings() (this=0x55555644ada0) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5029
#21 0x00005555557f78b1 in MyFrame::OnToolLeftClick(wxCommandEvent&) (this=0x55555644ada0, event=...) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:4659
#22 0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#23 0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#24 0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#25 0x00007ffff58cd783 in wxEvtHandler::DoTryChain(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#27 0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#28 0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#29 0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#30 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#31 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#35 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#36 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#37 0x00007ffff57510a6 in wxAppConsoleBase::MainLoop() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#38 0x00007ffff57dbae9 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#39 0x0000555555718bf8 in main(int, char**) (argc=<optimised out>, argv=<optimised out>) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:1095
```

leamas, 2020-04-01:
Indeed. 

Bisecting, bad commit c43e03d0bf2 "Correct Plugin panel GUI ancestor relationships"

Looking at that commit  (besides ḿore important stuff I don't really grasp ATM) here is a C cast which probably would be better done as a dynamic_cast with a runtime check:

    m_PluginListPanel = (PluginListPanel *)parent->GetParent();

vs

    m_PluginListPanel = dynamic_cast<PluginListPanel*>(parent->GetParent());
    wxAssert(m_PluginListPanel != 0);

bdbcat, 2020-04-01:
Sorry, cannot reproduce.
@Alec:  Thanks for the catch.  I think not related to this crash, but should be done regardless.
Do you reproduce the crash?

leamas, 2020-04-01:
> Do you reproduce the crash?

Yes. On the good commits, uninstalled and installed VDR without problems. On the bad, uninstall fails.

Building on updated bionic host, no arguments to cmake.

leamas, 2020-04-01:
But... I don't get the crash, I get an assertion error:

    /home/al/src/OpenCPN/src/pluginmanager.cpp(629): assert "plugin_list_panel != 0" failed in OnPluginUtilAction().

Perhaps @jongough  is building without enabled assertions?

bdbcat, 2020-04-01:
Building on virtual 18.04 now, will test.

bdbcat, 2020-04-01:
Strange, 18.04 works fine for me.  No assert, expected behavior observed.

leamas, 2020-04-01:
On next build I get a real crach, top
```
#0  0x00005555559df75d in PluginListPanel::ReloadPluginPanels(ArrayOfPlugIns*) (this=0x0, plugins=0x555556357dc0)
    at /home/al/src/OpenCPN/src/pluginmanager.cpp:5245
#1  0x00005555559e7739 in pluginUtilHandler::OnPluginUtilAction(wxCommandEvent&) (this=<optimized out>, event=...)
    at /home/al/src/OpenCPN/src/pluginmanager.cpp:703
#2  0x00007ffff56b14be in wxEvtHandler::ProcessEventIfMa

this == 0 should mean som kind of stack corruption?
```

leamas, 2020-04-01:
Using updated master catalog...

bdbcat, 2020-04-01:
Your crash happened on a build without asserts enabled?

leamas, 2020-04-01:
In hindsight, I might have been wrong the first time, but I re-ran the bisect  using a fresh build dir and a cmake without options.  In this case, a crash.

Which is about the default target. I'll be back...

leamas, 2020-04-01:
All sorted out. Configuring without arguments (i. e.,` CMAKE_BUILD_TYPE=RelWithDebInfo`) crashes. Rebuilding with `CMAKE_BUILD_TYPE=Debug` gives the assertion error

bdbcat, 2020-04-01:
Alec...
This is plain strange.

The parentage should be:
PluginListPanel->wxPanel(private)->PluginPanel

This added wxPanel is required for Mac

If you have the time, will you try this:

```
void pluginUtilHandler::OnPluginUtilAction( wxCommandEvent& event )
{
    auto panel = static_cast<PluginPanel*>(event.GetClientData());
    auto plugin_list_panel = dynamic_cast<PluginListPanel*>(panel->GetParent());
    if(!plugin_list_panel)
        plugin_list_panel = dynamic_cast<PluginListPanel*>(panel->GetGrandParent());
        
    wxASSERT(plugin_list_panel != 0);

```

leamas, 2020-04-01:
hm... could you paste the diff instead?

bdbcat, 2020-04-01:
Not easily.  I am in the middle of another rathole...
This is pluginmanager.cpp:625

leamas, 2020-04-01:
ok... I got it. stay tuned.

leamas, 2020-04-01:
progress... that actually seems to work. No crash, no assertion.

bdbcat, 2020-04-01:
Are you able to debug?
Which cast is being used?  The parent, or the grandparent?

leamas, 2020-04-01:
I could, but loggin is faster. Stay tuned

leamas, 2020-04-01:
It uses GetGrandParent()

bdbcat, 2020-04-01:
As it should.
I think something funny here.
Once more:

```
void pluginUtilHandler::OnPluginUtilAction( wxCommandEvent& event )
{
    auto panel = static_cast<PluginPanel*>(event.GetClientData());
    PluginListPanel *plugin_list_panel = dynamic_cast<PluginListPanel*>(panel->GetGrandParent());
        
    wxASSERT(plugin_list_panel != 0);

Thanks
Dave

```

leamas, 2020-04-01:
My attempt to a finalized patch: 
```

diff --git a/src/pluginmanager.cpp b/src/pluginmanager.cpp
index 262fc0bf6..a1cd86fe5 100644
--- a/src/pluginmanager.cpp
+++ b/src/pluginmanager.cpp
@@ -625,7 +625,8 @@ pluginUtilHandler::pluginUtilHandler()
 void pluginUtilHandler::OnPluginUtilAction( wxCommandEvent& event )
 {
     auto panel = static_cast<PluginPanel*>(event.GetClientData());
-    auto plugin_list_panel = dynamic_cast<PluginListPanel*>(panel->GetParent());
+    auto plugin_list_panel =
+        dynamic_cast<PluginListPanel*>(panel->GetGrandParent());
     wxASSERT(plugin_list_panel != 0);
     auto actionPIC = panel->GetPlugin();
     wxString name = actionPIC->m_common_name;
@@ -5401,7 +5402,9 @@ static bool canUninstall(std::string name)
 PluginPanel::PluginPanel(wxPanel *parent, wxWindowID id, const wxPoint &pos, const wxSize &size, PlugInContainer *p_plugin)
     :wxPanel(parent, id, pos, size, wxBORDER_NONE)
 {
-    m_PluginListPanel = (PluginListPanel *)parent->GetParent();
+    m_PluginListPanel =
+        dynamic_cast<PluginListPanel*>(parent->GetParent());
+    wxASSERT(m_PluginListPanel != 0);
     m_pPlugin = p_plugin;
     m_bSelected = false;

```

bdbcat, 2020-04-01:
And does that work?

bdbcat, 2020-04-01:
Aha...
The parenthesis is missing...

-    m_PluginListPanel = (PluginListPanel *)parent->GetParent();
+    m_PluginListPanel =  dynamic_cast<PluginListPanel*>(parent->GetParent());

bdbcat, 2020-04-01:
Alec...
I pushed (substantially) your patch.
Let me know how this works.

leamas, 2020-04-01:
Built a288b2ff9.  ALl looks good w r t this bug which from my perspective could be closed.

I suspect that some other of the last bunch of bugs filed by @jongough  might be related to this fix, but leaving this for now. I have remote education to plan...

Hakansv, 2020-04-01:
Dave and I have had a similar discussion/debug for Win10 that crashed when install/uninstall.
Now I've tested master source built half an hour ago on Win10
I can install/uninstall wo crash.
The plugins are directly present on the menu bar with correct icons.
So, it seams almost everything is solved.
Almost... Connected or not: When installing oeRNC/oeSENC it's not present in the options->Charts tab until O is restarted.
Håkan

bdbcat, 2020-04-05:
Close?

closed 2020-04-06 by jongough