OpenCPN/OpenCPN #1806: Plugin Manager - Assertion error when trying to display Plugins tab

jongough, 2020-04-01:
On OCPN master  (a0ea779)  when displaying the Plugins tab an assertion failed message shows:
[image]

This is the information in the assertion message
```
ASSERT INFO:
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxCommandEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?

BACKTRACE:
[1] wxClassInfo::Register()
[2] _dl_catch_exception
[3] _dl_catch_exception
[4] _dl_catch_error
[5] dlopen
[6] wxDynamicLibrary::RawLoad(wxString const&, int)
[7] wxDynamicLibrary::Load(wxString const&, int)
[8] PlugInManager::LoadPlugIn(wxString)
[9] PlugInManager::LoadPlugInDirectory(wxString const&, bool, bool)
[10] PlugInManager::LoadAllPlugIns(bool, bool)
[11] options::DoOnPageChange(unsigned long)
[12] options::OnPageChange(wxBookCtrlEvent&)
[13] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[14] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[15] wxEvtHandler::TryHereOnly(wxEvent&)
[16] wxEvtHandler::ProcessEventLocally(wxEvent&)
[17] wxEvtHandler::ProcessEvent(wxEvent&)
[18] wxBookCtrlBase::DoSetSelection(unsigned long, int)
[19] wxListbook::OnListSelected(wxListEvent&)
[20] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[21] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[22] wxEvtHandler::TryHereOnly(wxEvent&)
[23] wxEvtHandler::ProcessEventLocally(wxEvent&)
[24] wxEvtHandler::ProcessEvent(wxEvent&)
[25] wxWindowBase::TryAfter(wxEvent&)
[26] wxScrollHelperEvtHandler::ProcessEvent(wxEvent&)
[27] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[28] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[29] wxEvtHandler::TryHereOnly(wxEvent&)
[30] wxEvtHandler::ProcessEventLocally(wxEvent&)
[31] wxEvtHandler::ProcessEvent(wxEvent&)
[32] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[33] g_closure_invoke
[34] g_signal_emit_valist
[35] g_signal_emit
[36] gtk_propagate_event
[37] gtk_main_do_event
[38] g_main_context_dispatch
[39] g_main_loop_run
[40] gtk_main
[41] wxGUIEventLoop::DoRun()
[42] wxEventLoopBase::Run()
[43] wxDialog::ShowModal()
[44] MyFrame::DoOptionsDialog()
[45] MyFrame::DoSettings()
[46] MyFrame::OnToolLeftClick(wxCommandEvent&)
[47] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[48] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[49] wxEvtHandler::TryHereOnly(wxEvent&)
[50] wxEvtHandler::DoTryChain(wxEvent&)
[51] wxEvtHandler::ProcessEvent(wxEvent&)
[52] wxEvtHandler::ProcessPendingEvents()
[53] wxAppConsoleBase::ProcessPendingEvents()
[54] wxApp::DoIdle()
[55] g_main_context_dispatch
[56] g_main_loop_run
[57] gtk_main
[58] wxGUIEventLoop::DoRun()
[59] wxEventLoopBase::Run()
[60] wxAppConsoleBase::MainLoop()
[61] wxEntry(int&, wchar_t**)
[62] main
[63] __libc_start_main
[64] _start
```

bdbcat, 2020-04-01:
Probably casued by loading two copies of OCPN Draw, this being due to the Common Name problem in the metadata.  Should be fixed by latest metadata commit.

bdbcat, 2020-04-05:
Close?

jongough, 2020-04-05:
No occurring currently

closed 2020-04-05 by jongough