OpenCPN/OpenCPN #1808: PI Manager: Plugin List: VERSION (MAJOR, MINOR, PATCH) missing

rgleason, 2020-04-01:
Currently it shows major and minor.

bdbcat, 2020-04-01:
Screen shot would be good.
Unmanaged plugins have only major/minor available.

leamas, 2020-04-01:
Here is the major-minor Rick talks about + 

[image]

And here the version displayed  when updating which seems OK to me. 

[image]

The relevant code is in update_mgr.cpp around line 210; it should be possible to use it also in the pluginmanager.

bdbcat, 2020-04-01:
Chart Downloader, like all the system default plugins, is "legacy" plugin built to an older API.  It does not support major/minor/patch.

leamas, 2020-04-02:
Of course, by bad. 

"blushes"

Looking at the radar plugin in plugin list:
[image]

It looks just fine, and Rick needs to explain what this bug is all about. Sorry for the noise.

@rgleason: Where is the limited (just major.minor) version displayed?

rgleason, 2020-04-02:
Yes, It is in Windows.  Thanks
[image]
[image]

bdbcat, 2020-04-03:
Rick...
For me, this all works correctly.  In the attached pix, note that legacy unmanaged plugins show Major/Minor.  Managed plugins (after installation) show Major/Minor/Patch.
[image]

rgleason, 2020-04-03:
Sorry, I should update 5.0.522 again.

closed 2020-04-03 by rgleason

rgleason, 2020-04-04:
Thank you. This is working.  Needed to recompile.