OpenCPN/OpenCPN #1810: Plugin Manager - Installation of larger plugins gives no indication anything is happening

jongough, 2020-04-02:
Installing OD 1.6.39 takes quite a time as there are a large number of icons distributed with it. These have to be copied to various locations. However, all that shows for the user is the standard mouse pointer and no indication of the progress, or if any progress is being made. The progress can be seen on linux by looking at the OCPN log and following it, but this would not be normal user practice.

A progress dialog  box, or a message telling the user what stage the install is at would be useful.

jongough, 2021-04-26:
With OCPN 5.5.0 the situation still has not changed.

bdbcat, 2021-05-07:
I have added a "busy spinner" covering the entire plugin load process, to provide some feedback.

A plugin library normally loads and links in less than ~200 msec.  Activated plugins that consume long time during their Init() method may like to display their own progress dialog.  The OCPN core has no idea of how long this will take, or what the steps are.  So the plugin must be responsible.

jongough, 2021-05-07:
I think for most plugins the delay is in the unpacking and installing of items from the zip file which is before the plugin Init() method. The oesenc plugin is probably the only one that would have elongated startup due to the encryption process for the chartlets.