OpenCPN/OpenCPN #1811: Plugin Manager - Installing incorect version of plugin into Ubuntu 18.04

jongough, 2020-04-02:
With OCPN master (68f7ed24) installed on Ubuntu 18.04 installed OD 1.6.40 from PI Manager selection list. This is the message from the log:
```
11:37:56.087 MESSAGE Downloader.cpp:132 filesize https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.40-ubuntu-18.04-bionic-gtk3-tarball/versions/1.6.40.0+67.5a7a57b/ocpn_draw_pi-1.6.40.0_ubuntu-18.04-bionic-gtk3.tar.gz: 8603084 bytes
```
However, when attempting to use the plugin the following error messages are given and the plugin does not show up:
```
11:38:24.309 MESSAGE pluginmanager.cpp:2079 PlugInManager: Loading PlugIn: /home/jon/.local/lib/opencpn/libocpn_draw_pi.so
11:38:24.320   ERROR dlunix.cpp:162 libwx_gtk3u_core-3.0.so.0: cannot open shared object file: No such file or directory
11:38:24.320 MESSAGE pluginmanager.cpp:2138    PlugInManager: Cannot load library: /home/jon/.local/lib/opencpn/libocpn_draw_pi.so 
```

bdbcat, 2020-04-02:
Is your OCPN core built for gtk2 or gtk3?

$ldd /usr/bin/opencpn

jongough, 2020-04-02:
gtk2
```
jon@mint-19-2-cinnamon:/data/public/working area/opencpn/OpenCPN-5.0/build-mint-19.2$ ldd opencpn
	linux-vdso.so.1 (0x00007ffc7bde5000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f15c2929000)
	libcurl.so.4 => /usr/lib/x86_64-linux-gnu/libcurl.so.4 (0x00007f15c26aa000)
	libarchive.so.13 => /usr/lib/x86_64-linux-gnu/libarchive.so.13 (0x00007f15c23fa000)
	libelf.so.1 => /usr/lib/x86_64-linux-gnu/libelf.so.1 (0x00007f15c21e0000)
	libwx_gtk2u_gl-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0 (0x00007f15c1fc9000)
	libwx_baseu_net-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_baseu_net-3.0.so.0 (0x00007f15c1d85000)
	libwx_baseu_xml-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_baseu_xml-3.0.so.0 (0x00007f15c1b75000)
	libwx_gtk2u_html-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_html-3.0.so.0 (0x00007f15c18a2000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_adv-3.0.so.0 (0x00007f15c14ca000)
	libwx_gtk2u_aui-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_aui-3.0.so.0 (0x00007f15c1237000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0 (0x00007f15c0a05000)
	libwx_baseu-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0 (0x00007f15c0576000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f15c02ea000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f15c00c4000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f15bfebc000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f15bfc9d000)
	libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f15bfa50000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f15bf7fc000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f15bf5ef000)
	libexif.so.12 => /usr/lib/x86_64-linux-gnu/libexif.so.12 (0x00007f15bf3ab000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f15bf08e000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f15bee5c000)
	libportaudio.so.2 => /usr/lib/x86_64-linux-gnu/libportaudio.so.2 (0x00007f15bec2d000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f15bea18000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f15be70f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f15be386000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f15bdfe8000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f15bddd0000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f15bd9df000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f15bd7b7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f15bd5b3000)
	libnghttp2.so.14 => /usr/lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007f15bd38e000)
	libidn2.so.0 => /usr/lib/x86_64-linux-gnu/libidn2.so.0 (0x00007f15bd171000)
	librtmp.so.1 => /usr/lib/x86_64-linux-gnu/librtmp.so.1 (0x00007f15bcf55000)
	libpsl.so.5 => /usr/lib/x86_64-linux-gnu/libpsl.so.5 (0x00007f15bcd47000)
	libssl.so.1.1 => /usr/lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007f15bcaba000)
	libcrypto.so.1.1 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f15bc5ef000)
	libgssapi_krb5.so.2 => /usr/lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007f15bc3a4000)
	libldap_r-2.4.so.2 => /usr/lib/x86_64-linux-gnu/libldap_r-2.4.so.2 (0x00007f15bc152000)
	liblber-2.4.so.2 => /usr/lib/x86_64-linux-gnu/liblber-2.4.so.2 (0x00007f15bbf44000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f15bbd27000)
	libnettle.so.6 => /usr/lib/x86_64-linux-gnu/libnettle.so.6 (0x00007f15bbaf1000)
	libacl.so.1 => /lib/x86_64-linux-gnu/libacl.so.1 (0x00007f15bb8e9000)
	liblzo2.so.2 => /lib/x86_64-linux-gnu/liblzo2.so.2 (0x00007f15bb6c7000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f15bb4ab000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f15bb29b000)
	libxml2.so.2 => /usr/lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f15baeda000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f15c3821000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f15ba898000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f15ba5e3000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f15ba3bf000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f15ba0a8000)
	libnotify.so.4 => /usr/lib/x86_64-linux-gnu/libnotify.so.4 (0x00007f15b9ea0000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f15b9c9a000)
	libSM.so.6 => /usr/lib/x86_64-linux-gnu/libSM.so.6 (0x00007f15b9a92000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f15b9860000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f15b95f8000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f15b9381000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f15b9150000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f15b8e9a000)
	libthai.so.0 => /usr/lib/x86_64-linux-gnu/libthai.so.0 (0x00007f15b8c91000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f15b8a89000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f15b8844000)
	libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f15b862e000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f15b837a000)
[... cut ...]
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f15aec3e000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f15ae9f4000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f15ae7bc000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f15ae56f000)
```
The previous install of OD 1.6.39 picked the gtk2 installation and worked OK. Uninstalling 1.6.40 and reinstalling, from cache, 1.6.39 worked. Don't know why the gtk3 version was installed.

bdbcat, 2020-04-02:
I don't either.  Does the same for me.

After installing 1.6.40, I see:
$ ldd ~./local/lib/opencpn/libocpn_draw_pi.so | grep wx
	libwx_baseu-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0 (0x00007fcad7526000)
	libwx_gtk3u_core-3.0.so.0 => not found
	libwx_gtk3u_adv-3.0.so.0 => not found

So, 1.6.40 was built against gtk3.  Something in the CI workflow changed?

bdbcat, 2020-04-02:
Jon....
This is a funny corner case, which we have not encountered before.  We are not directly checking for gtk3, and choosing the correct plugin.  We seem to be picking the first that we see.
You can bypass now by editing ~/.opencpn/ocpn-plugins.xml and simply removing the OCPN Draw metadata section for bionic-gtk3.  This will leave only the gtk2 flavor, and should install correctly.

jongough, 2020-04-02:
The XML has both the gtk3 & gtk version in it and the build was done for both environments. Extracts:
```
   https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.40-ubuntu-18.04-bionic-tarball/versions/1.6.40.0+68.5a7a57b/ocpn_draw_pi-1.6.40.0_ubuntu-18.04-bionic.tar.gz
    https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.40-ubuntu-18.04-bionic-gtk3-tarball/versions/1.6.40.0+67.5a7a57b/ocpn_draw_pi-1.6.40.0_ubuntu-18.04-bionic-gtk3.tar.gz
```

bdbcat, 2020-04-02:
So manually remove the gtk3 entry, for now, to test the gtk2 flavor.

jongough, 2020-04-02:
Yep, that worked.

bdbcat, 2020-04-02:
Jon...
The metadata for a plugin build targeting gtk3 must say so, in this way:

<target>ubuntu-gtk3</target>
<target-version>18.04</target-version>

Your currently shows :
<target>ubuntu</target>

in both gtk2 and gtk3 targets.
Please check your CI workflow.
Dave

jongough, 2020-04-02:
Will this not cause issues when looking at the platform? Should there be another token to describe, say, gtk3, with this being absent/empty or containing a requirement?

bdbcat, 2020-04-02:
Maybe we can conjure up a problem if we think about it enough.
But lets solve symptoms now.
The patch I show above has been tested locally and works for me, so lets start there.

jongough, 2020-04-02:
Changed generated xml to include gtk3 for the <TARGET> property. This seems to have fixed the issue.

closed 2020-04-02 by jongough