OpenCPN/OpenCPN #1812: Plugin Manager - Ubuntu 18.04 plugins list not drawn correctly

jongough, 2020-04-02:
Using Ubuntu 18.04 master (68f7ed24) opening up the Options/Plugins tab shows:
[image]
Notice the OCPN Draw line is wrapped and not shown correctly. If the dialog is resized this line shows. There is adequate space to show the line if the 'enabled' check box and the icon to its right are right aligned. This is what it looks like if the dialog box is resized:
[image]

jongough, 2021-04-26:
There are still issues with the display of information for plugin:
[image]
Notice the testplugin description line is not complete as it ends in ' for'
[image]
Click on the plugin and the description field changes but is still incomplete

rgleason, 2021-04-28:
PR from Upstream Last night April 28 and compiled v5.5.   This is working better. The last case for Weatherfax does not seem to be wrapping. Is there something I should change in Weatherfax or can you wrap a single line at some reasonable width?    AND are there some rules or guidelines  for devs to follow for this text?

[image]
[image]
[image]
[image]

bdbcat, 2021-04-28:
Rick...
You should manually wrap the text with "\n" (new line) in the source text string.  It seems reasonable to expect that the top-level "settings" dialog will normally accommodate descriptive text ines of about 40 characters, depending upon the font selected.
And please, try to keep the "short description" really short, to accommodate very small screens, like Android phones.  "Logbook for OpenCPN" is good.

Dave

rgleason, 2021-04-28:
Ok, will do that.  Thanks.  Probably should be noted in Dev Manual too.

closed 2021-04-28 by rgleason

leamas, 2021-08-13:
BTW: To keep description short, especially the summary, one should IMHO avoid using the words "plugin" and OpenCPN". We already know the all plugins are plugins for OpenCPN.