OpenCPN/OpenCPN #1815: Plugin Manager - Private icons not found

jongough, 2020-04-02:
Using OCPN (a288b2ff) and OD 1.6.40 installed via Plugin Manager the standard icons are loaded into the ~/.local/share/opencpn/plugins/ocpn_draw_pi/UserIcons directory. However when OD tries to find them via the GetpPrivateApplicationDataLocation() call it effectively told that they are in ~/.opencpn/plugins/ocpn_draw_pi/UserIcons.

bdbcat, 2020-04-02:
Portable mode, or standard?

jongough, 2020-04-02:
Standard mode. Portable seems to be working.

bdbcat, 2020-04-02:
Jon...
You should probably not use:
GetpPrivateApplicationDataLocation()

Use
 wxString GetPluginDataDir(const char* plugin_name);

Same as for the icons.

jongough, 2020-04-02:
There used to be GetpSharedDataLocation and GetpPrivateApplicationDataLocation and these were used for different types of data. OD has system icons, the ones that populate the toolbars and user icons, the ones used when drawing. OD also has a layers directory for storing user layers information. There is a data directory which is used for storing OD's nav obj.xml and navob xml changes. Mixing up icons and changeable data is not very good, especially if the user is going to add/remove stuff from the directory it could cause all sorts of issues.

OD needs:
* A secure location to store non-user icons, these are read only
* A secure, writeable area to store runtime files that the user should not touch
* A location to place there own icons for use by OD
* A location where the user layers information is stored.

bdbcat, 2020-04-02:
Yes, we distinguish between static (bundled as part of the plugin installer) and dynamic data.

For the read-only data, use 
wxString GetPluginDataDir(const char* plugin_name);
to access the location as installed from the plugin package.

For the dynamic data (the last three in your list), you may use anything in GetpPrivateApplicationDataLocation() that you like.

~/.opencpn/plugins/ocpn_draw_pi/UserIcons seems fine for dynamic data belonging to your plugin. But this cannot, in general, be the same as a directory loaded by static data from the plugin package.

jongough, 2020-04-02:
The problem is that the Plugin Manager change appears to have changed the search path used. I have an ocpn_draw_pi deb which installs to /usr/local and puts all the files in the correct place but OCPN is now saying that the location is ~/.local/share/opencpn rather than /usr/local/share/opencpn. Here is an extract from the log:
```
18:51:59.392 MESSAGE pluginmanager.cpp:2079 PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libocpn_draw_pi.so
18:51:59.401 MESSAGE OCPNPlatform.cpp:1569 Using plugin data path: /home/jon/.local/share/opencpn/plugins;/usr/share/cinnamon/opencpn/plugins;/usr/share/gnome/opencpn/plugins;/home/jon/.local/share/flatpak/exports/share/opencpn/plugins;/var/lib/flatpak/exports/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/local/lib/opencpn
18:51:59.401 MESSAGE pluginmanager.cpp:486 PlugInManager: Using data dirs from: /home/jon/.local/share/opencpn/plugins;/usr/share/cinnamon/opencpn/plugins;/usr/share/gnome/opencpn/plugins;/home/jon/.local/share/flatpak/exports/share/opencpn/plugins;/var/lib/flatpak/exports/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/local/lib/opencpn
18:51:59.401 MESSAGE pluginmanager.cpp:499 PlugInManager: using data dir: /home/jon/.local/share/opencpn/plugins/ocpn_draw_pi
18:51:59.401 MESSAGE ODicons.cpp:63 ocpn_draw_pi data location: /home/jon/.local/share/opencpn/plugins/ocpn_draw_pi/data/
18:51:59.401   ERROR svgxml.cpp:445 File /home/jon/.local/share/opencpn/plugins/ocpn_draw_pi/data/ODManager_MUI.svg doesn't exist.
```
You can see that the correct location is in the list of paths but only the first in the list is being returned by the call. This would seem to invalidate plugins that have not been upgraded to the Plugin Manager.

bdbcat, 2020-04-02:
jon...
OK, I see.
The trouble is with the managed plugin un-installer.  It leaves some fluff in place which confuses a legacy plugins (of the same name) later.
Should be corrected by 367153929fd67078ed04df48d781a6b017900a6a
You will need to install and uninstall your plugin once to activate the new code.  After that, the legacy (.deb) version should work OK.

jongough, 2020-04-02:
Hi Dave,
   I uninstalled my deb file while OCPN was not running. Started OCPN and installed my plugin via the Plugin Manager and it worked OK. I then uninstalled the plugin vi the Plugin Manager, stopped OCPN, installed my deb file and restarted OCPN. The icons for the main toolbar are not found as OCPN is looking in the ~/.local/.... directory whereas the files are actually in the /usr/local/share/opencpn/plugins/ocpn_draw_pi/data directory.

[log of OCPN using deb installed plugin with icons not found.txt](https://github.com/OpenCPN/OpenCPN/files/4424033/log.of.OCPN.using.deb.installed.plugin.with.icons.not.found.txt)
[ocpn_draw_pi-1.6.46-package-list.txt](https://github.com/OpenCPN/OpenCPN/files/4424036/ocpn_draw_pi-1.6.46-package-list.txt)

jongough, 2020-04-02:
Sorry, just updated git and re-run the test. Looks like it is working. Will give it a couple more tests to see.

jongough, 2020-04-02:
3671539 appears to have fixed the issue. Thanks

closed 2020-04-02 by jongough