OpenCPN/OpenCPN #1816: Path to plugins folders in "Help"

Hakansv, 2020-04-02:
On Windows.
A "normal" user will have trouble finding the path to the new location like: "C:\Users\admin\AppData\Local\opencpn\plugins". That's rather "deep" in a folder structure seldom used. And no system shortcut is prepared for this like e.g. "Document", "Photos" 
I think we need a hint on the help message like the path to opencpn.log

[image]
[image]

rgleason, 2020-04-02:
Hakan,  Alec has suggested  in Windows using %LocalAppData%\opencpn\plugins\
 to shorten, but I agree with your suggestion, as it is still a long path.

leamas, 2020-04-02:
hm... but why would a "normal" user care?  There are use cases for the log and also occasionally the config file. But the plugin data?

My basic assumption is that if normal users need to access this path it is a bug which should be filed and fixed.

Hakansv, 2020-04-02:
Alec..
One example using WeatherRouting.
In %LocalAppData%\opencpn\plugins\weather_routing_pi\data\boats  and ...data\polars are the files located for different polar data. From installation are some examples located there but polar files for an own boat could be stored there as well. 
Would you suggest another path location for data like this that is possible to systematize for our different OS's? 
Without having checked further I suppose ocpn_draw and weather-fax has a similar case?

leamas, 2020-04-02:
Now, talk is cheap. So is also proposing that others should code.

That said, I *think* that the proper solution to what you describe is to add something like *Import Polar Data* or so to the plugin. That would mean a proper file selection dialog in the UI, and plugin storing the file in the proper place.  As I see it, if normal users are supposed to do it it would be the correct way.

Problem is just that it takes some cycles to design and some others to code.

leamas, 2020-04-02:
At  a second thought: storing the user-provided files in `%LOCALAPPDATA%\opencpn\plugins` is basically a bad idea. See https://github.com/OpenCPN/OpenCPN/issues/1815#issuecomment-607594794

rgleason, 2020-04-02:
The way the old packaged weather_routing has done this in the past, is to install the data to the normal plugin data directory, IE    C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\weather_routing_pi\data    and then "boats"  "polars"  and files like WeatherRoutingConfiguration.xml

Then I think when you start the plugin for the first time, the user writable files and directories get copied to C:\ProgramData\opencpn\plugins\weather_routing   (not a data directory) with "polars"  "contours"  and "boats"  and WeatherRoutingConfiguration.xml.    After that initial startup, this is the directory used by the plugin because it has to be writeable.

So for this new directory  %localappdata%/opencpn/plugins/weather_routing  it is writeable is it not?   BTW I was asking about  this stuff a long time ago, and realized that the plugin was not really setup for this new system, but I was able to redirect paths so that the plugins worked.  That is where I left it at the time.

rgleason, 2020-04-02:
> For the read-only data, use
> wxString GetPluginDataDir(const char* plugin_name);
> to access the location as installed from the plugin package.
> 
> For the dynamic data (the last three in your list), you may use anything in GetpPrivateApplicationDataLocation() that you like.

So why not just use the same folder that is writable by the user

C:\ProgramData\opencpn\plugins\weather_routing

leamas, 2020-04-02:
> So for this new directory %localappdata%/opencpn/plugins/weather_routing it is writeable is it not

It is writable, but handled solely by the installer.

> So why not just use the same folder that is writable by the user

Because these are fundamentally different. The %APPDATA% locations are managed by the installer whereas the GetpPrivateApplicationDataLocation() are managed by the plugin. To mix them in a single directory is doomed to cause all sorts of troubles when for example updating plugins.

Still, the basic scheme you describe should work just fine also in the new context. As I understand it, the plugin copies templates in %APPDATA% to the working area in C:\ProgramData\opencpn\, which is just fine.

And, if we want users to add boats and what not to  C:\ProgramData\opencpn  I see no problem with that. But if it is supposed to be done by normal users,  some kind of "import" GUI is IMHO the natural  solution rather than documenting procedures to handle  various obscure, platform-dependent paths.

rgleason, 2020-04-02:
`Still, the basic scheme you describe should work just fine also in the new context. As I understand it, the plugin copies templates in %APPDATA% to the working area in C:\ProgramData\opencpn, which is just fine.
`
Yes, that is what I propose.

leamas, 2020-04-02:
> may require some programming conditionals for a older packaged plugin vs a managed plugin.

Basically no. All existing plugins treats the installed data as read-only just because it is.  The difference is how to find the data directory, but besides that there should be no differences at all. There is no need for conditionals besides plugin API >= 1.16.

leamas, 2020-04-02:
From my side, this discussion boils down to that this bug should be closed. Displaying the path to the plugin installation directory doesn't not solve anything since users never should cope with this area.

For plugins which supports adding files to various directories in the plugin data dir (not in %LOCALAPPDATA%)  the paths  must be displayed in the plugin preferences page or, probably better,  being handled by adding an import GUI.

rgleason, 2020-04-02:
Ok, so there should be no changes required, but I recall and wrote about having some path difficulties when I tried PI Manager with weather_routing,  that now seems like a long time ago.

leamas, 2020-04-02:
Even if so, this is not about this bug.

rgleason, 2020-04-02:
Thanks for completing my understanding about how the new data directory works. Hakan has identified an issue related in wxrouting, and seeing the path might help.

Hakansv, 2020-04-03:
This issue is not about any OCPN bug. It's more a user orientated philosophic discussion.
If using WeatherRouting as a example we need a files location where the automated installation process can save user files that the user can easily reach for own edit and/or completion. The completion can be files created by an external program or by the user itself. These files are most convenient to having saved in the same folder as the ones installed by the plugin installation.
Previous plugin installation process used the same folder location as for the config and log files. For the user these are relatively easy to find by the links in the  OCPN info message (?) 
But now is %ProgramData% not used for plugin data. There is no automated way to user documents so the one left to use is %LOCALAPPDATA%. Of course could every plugin include a link to that folder when needed.
Still I though think a general link to explore plugin data, like %LOCALAPPDATA%\opencpn\plugins\ beside log/config -links would be handy and understandable for users copying own data. It wouldn't be very hard to code such a link suitable for all OS I suppose?

rgleason, 2020-04-03:
No it wouldn't be hard to code that.  Alec has it all mapped out here in his Opencpn wiki
https://github.com/leamas/OpenCPN/wiki/Installation-paths

leamas, 2020-04-04:
> Still I though think a general link to explore plugin data, like %LOCALAPPDATA%\opencpn\plugins\ beside log/config -links would be handy and understandable for users copying own data. I

No, this would not be hard to code. It's just IMHO a bad idea for several reasons. Please bear with me, this might become too long.

  - The locations under %APPDATA% corresponds to the C:\Program Files (x86)\Opencpn files on windows. It takes administrator privileges to install anything here on WIndows. On linux, using root privileges to install user files to the corresponding dirs under /usr/share is simply unacceptable. Ergo: Modifying files in these locations for unmanaged plugins is a bad idea.
  - The %APPDATA% locations are handled by the installer. This means that any file put here by a user potentially can be removed or modified by the installer when the plugin is updated or re-installed. Ergo: Although writable, these files and directories should not be modified by user.
  - Plugins which handles modified data (i. e., state) does so by creating directories under GetPrivateDataDir() which is ...\ProgramData\OpenCPN on windows. Ergo: If there should be a place where users adds or modifies data directtly in the file system it should lbe under  ProgramData
  -  If there is such a place it is specific to the plugin. Ergo: the correct place to display it to users is the plugin's preferences page.
  - Any workflow based on users coping directly with files in the file system will need excessive documentation for four platforms which is hard to maintain and by definition not that user-friendly. Ergo: Workflows for regular users should have GUI support like "Import...",  "Edit...", etc.
  - On Android, workflows based on direct file access is just not feasible (no file manager or explorer). Ergo: Create GUI for supported workflows, as above.

This just  is upside-down. What have here is a Linux oldboy trying to convince a Windows  expert about the importantance  of GUI support ;)

Hakansv, 2020-04-04:
Alec..

> Plugins which handles modified data (i. e., state) does so by creating directories under GetPrivateDataDir() which is ...\ProgramData\OpenCPN on windows. Ergo: If there should be a place where users adds or modifies data directtly in the file system it should lbe under ProgramData

I totally agree! That's the place all such files have been located earlier. Now with the new plugin management it has moved to %LOCALAPPDATA% for reasons extensively discussed the last months. All this may be a misunderstanding. So if WeatherRoutine's user data, like polars, are moved back to GetPrivateDataDir() I think all are happy? (Rick?)
@jongough your OCPN_Draw is another plugin having user data although not normally edited by the user. Any thoughts?

A GUI in the plugin to explore the user data folders is of course a good idea to the plugin developers.

The Android comment is relatively relevant. But since many OCPN users are dealing with to copy there own chart files it's normally not unknown how to find and use a file explorer also here.

leamas, 2020-04-04:
The setup is simple: 

  - The read-only data installed by the installer lives in GetPLuginDataDir(plugin); on windows this is  a directory below %LOCALAPPDATA%.
  -  Data which is modified (has state) lives in  GetPrivateDatDir() which means below ProgramData on windows. Nothing is installed here, it's the plugin's task to populate and maintain this area-

If a plugin puts file with state in %LOCALAPPDATA% it's a bug. The files here are read-only, possibly removed or modified by the installer.

Hakansv, 2020-04-04:
As plain as a pikestaff, very good. (In Swedish we talk about sausages instead :) )
I think some question marks have been straightened.
So Rick or Sean ?? WeatherRouting would be changed?
Thanks
Håkan

closed 2020-04-04 by Hakansv

rgleason, 2020-04-04:
The way weather_routing has worked in the past is to install the dll and data directory to (for example)
C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins  - dll
C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\weather_routing_pi\data
- boats
- polars
svg files
WeatherRoutingConfiguration.xml
WeatherRouting Information.html

--------
Then the files that the user has access to edit and change are copied to 
C:\ProgramData\opencpn\plugins\weather_routing
- boats
- contours
- polars
WeatherRoutingConfiguration.xml

There are comparable directories for Linux.

Now with the new  "Managed Plugins" %LocalAppData" installation directory for plugins. That is not a directory that users can edit, so as with the earlier arrangement, weather_routiing should copy those files over to the user "ediit" directory., C:\ProgramData\opencpn\plugins\weather_routing
or comparable in Linux.

I am not entirely clear what weather_routing actually does when newly installed as a "managed" plugin, but I do recall there was some issue with paths and I was able too successfully redirect the paths to the correct "editing" directory in C:\ProgramData.

I will try installing wxroute fresh and learn more about this, however Hakan what is your experience to date?

rgleason, 2020-04-04:
Have tried installing wx-routing version 1.13.5 for Ubuntu 18.04 and the path to "WeatherRoutingConfiguration.xml is not found automatically.  I found the file path point to 
rgleason>.opencpn  and had to redirect it to  Home > rgleason > .opencpn > plugins > weather_routing > "WeatherRoutingConfiguration.xml"  the main file for the plugin.

Also I found that rgleason > .opencpn > plugins > weather_routing > data > boats  did not have any files in it.  So they have to be copied over manually to that folder now. Same thing probably for Polars.   So using the files in  Home > .local > share > opencpn > plugins > weather_routing_pi > data, find  "boats" and "polars" directories and copy those 

However, I just found  Home > .opencpn > plugins > weather_routing  and copied the files into there respectively boats and polars.   Now I was able to do some routing, but I find that Grib is not reading some older files that it used to read (Not valid) and that weather_routing is no longer able to utilize the new grib, so routings can only be done using Climatology turned on so there is at least some data to work with.

So, in conclusion,
1. Grib functionality has to be checked and fixed.
2. Grib functionality in weather_routing is needed and should be checked.
3. Weather_routing installation has to be fixed so that files get copied to the right places and pathss get adjusted, such that wx-routing starts and works "out of the box".