OpenCPN/OpenCPN #1817: Plugin Manager - Ubunt18.04 upgrade install then uninstall of plugin segmentation fault

jongough, 2020-04-02:
Installed OD 1.6.46 into OCPN via deb file. Started OCPN and OD was working correctly. Opened Options/Plugins and selected OD and clicked the upgrade to managed button. All appeared to work OK and could create objects. Opened Options/Plugins selected OD and clicked uninstall button and OCPN seg faulted.
[image]
Here is the backtrace:
```
#0  0x00007ffff574a5df in wxAppConsoleBase::DeletePendingObjects() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#1  0x00007ffff574a661 in wxAppConsoleBase::ProcessIdle() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#2  0x00007ffff5e90d41 in wxAppBase::ProcessIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#3  0x00007ffff5db6bc9 in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#4  0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#5  0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#9  0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#10 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#11 0x00007ffff5e42cf3 in wxDialog::ShowModal() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#12 0x00005555557e1bea in TimedMessageBox::TimedMessageBox(wxWindow*, wxString const&, wxString const&, long, int, wxPoint const&) (this=0x7fffffffcd80, parent=0x55555644e520, message=..., caption=..., style=2052, timeout_sec=<optimised out>, pos=...) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:11568
#13 0x00005555557e1cc2 in OCPNMessageBox(wxWindow*, wxString const&, wxString const&, int, int, int, int) (parent=<optimised out>, message=..., caption=..., style=style@entry=2052, timeout_sec=timeout_sec@entry=-1, x=x@entry=-1, y=-1) at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:11606
#14 0x00005555559e8581 in pluginUtilHandler::OnPluginUtilAction(wxCommandEvent&) (this=<optimised out>, event=...) at /data/public/working area/opencpn/OpenCPN-5.0/src/pluginmanager.cpp:700
#15 0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#16 0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#17 0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#18 0x00007ffff58cda13 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#19 0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#20 0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#21 0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#22 0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#23 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#24 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#28 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#29 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#30 0x00007ffff5e42cf3 in wxDialog::ShowModal() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#31 0x00005555557f65f0 in MyFrame::DoOptionsDialog() (this=0x55555644e520)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:6131
#32 0x00005555557f6f29 in MyFrame::DoSettings() (this=0x55555644e520)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5032
#33 0x00005555557f7a21 in MyFrame::DoSettings() (this=0x55555644e520)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5029
#34 0x00005555557f7a21 in MyFrame::OnToolLeftClick(wxCommandEvent&) (this=0x55555644e520, event=...)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:4659
#35 0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#36 0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#37 0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#38 0x00007ffff58cd783 in wxEvtHandler::DoTryChain(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#39 0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#40 0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#41 0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#42 0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#43 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#44 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#45 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#46 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#47 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#48 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#49 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#50 0x00007ffff57510a6 in wxAppConsoleBase::MainLoop() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#51 0x00007ffff57dbae9 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#52 0x0000555555718db8 in main(int, char**) (argc=<optimised out>, argv=<optimised out>)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:1095
```
and the log at the time of the seg fault:
```
09:59:50.383 MESSAGE pluginmanager.cpp:1537 PlugInManager: Deactivating PlugIn: /home/jon/.local/lib/opencpn/libocpn_draw_pi.so
09:59:50.391 MESSAGE pluginmanager.cpp:696 Uninstalling OCPN Draw
09:59:50.391 MESSAGE pluginmanager.cpp:1537 PlugInManager: Deactivating PlugIn: /home/jon/.local/lib/opencpn/libocpn_draw_pi.so
```

bdbcat, 2020-04-02:
Is this repeatable?

jongough, 2020-04-02:
Yes, I got it once then repeated the process. I have just done a further test, if you upgrade and try the plugin it works OK, if you then go back and uninstall the plugin you get the seg fault. If you stop OCPN and restart it then go uninstall it works OK. Edge case I know, but.... I just got a seg fault when doing the upgrade I got offered after restarting OCPN.

I am continuing to investigate ways of recreating these issues. It seems there are inconsistencies in the management of plugins.

bdbcat, 2020-04-03:
Jon...
Think about this theory:
When uninstalling, at the point that the confirmation dialog appears, there is no OCPN Draw plugin active at all.  The uninstalled plugin has been removed, but the hidden legacy plugin has not yet bee activated.  And a general event loo[p is running still.
If anything happens to cause a screen refresh which involves OCPN Draw at all, it will crash, because the IS no OCPN Draw plugin running at this point.

Could this be happening?
If so, we need to readdress the confirmation dialog timing....

jongough, 2020-04-03:
I have just done a few tests on this process. 
* Install OD from deb file
* Draw OD object
* Upgrade to PI Man version
* Move created object
* Uninstall OD
I then get the Seg Fault. To try to narrow it down I have tried:
1. Unintall with OD enabled
1. Uninstall with OD disabled
1. Uninstall after removing OD objects and disabling OD via check box
1. Uninstall after removing OD objects and disabling OD via check box and clicking apply
1. Uninstall after removing OD objects and disabling OD via check box and restarting the dialog
All of which get the seg fault.

The question is if PI Manager can upgrade from a deb install to a PI Man install whilst the plugin is running why can it not uninstall the PI Man version without a crash? I was thinking that perhaps the uninstall process should 'disable' the plugin and do the 'apply' process before doing the uninstall, but I think there may be something else going on as well. Perhaps there should be a check to see if there is a deb install behind the PI Man install (could be marked as such during the 'upgrade') and then if there is apply the appropriate processing.

bdbcat, 2020-04-03:
Jon...
I could experiment.
Where is your current .deb file?

jongough, 2020-04-03:
All my files are now on cloudsmith, for the deb package files you will find them here:
[https://cloudsmith.io/~jon-gough/repos/ocpn_draw_pi-alpha/packages/?q=*.deb](https://cloudsmith.io/~jon-gough/repos/ocpn_draw_pi-alpha/packages/?q=*.deb)
If you want others replace the deb with exe or xml, or remove the 'search' criteria to see everything.

jongough, 2020-04-03:
If you use the ocpn-plugins.xml from your plugins git master branch that will work for you. You will get the option of OD 1.6.47

bdbcat, 2020-04-03:
Testing now...

bdbcat, 2020-04-03:
Jon...
Please try latest commit b25ed04823d10097fed9afc71f88badfa945efee
I reordered the uninstall confirmation dialog, to avoid re-entering the message loop while no plugin is activated.

jongough, 2020-04-03:
I just built that version and tried it and got the seg fault when I uninstalled the PI manager version. This is the backtrace:
```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x00007ffff58cd8b2 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
(gdb) bt
#0  0x00007ffff58cd8b2 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#1  0x00007ffff58cd95f in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#2  0x00007ffff58cda13 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#3  0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#4  0x00007ffff58cd7d7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#5  0x00007ffff5831661 in wxTimerImpl::SendEvent() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff5de9861 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#7  0x00007fffef268d03 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#9  0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#12 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#13 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#14 0x00007ffff5e42cf3 in wxDialog::ShowModal() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#15 0x00005555557e1bea in tFunctorMethodI14wxEventTypeTagI13wxSocketEventE12wxEvtHandler7wxEventS3_ED0Ev ()
#16 0x0000555557e0dc40 in  ()
#17 0x000055555644e3b0 in  ()
#18 0x00007fffef790a60 in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#19 0x00007fffffffcd80 in  ()
#20 0x00007fffffffd190 in  ()
#21 0x00007fffffffcee0 in  ()
#22 0x00005555566fce80 in  ()
#23 0x00007fffffffcf10 in  ()
#24 0x00007fffffffcf40 in  ()
#25 0x00005555557e1cc2 in N11PopUpDSlideD0Ev ()
#26 0x0000000000000068 in  ()
#27 0x0000000000000000 in  ()
```

bdbcat, 2020-04-03:
jon...
Well, not happening here, never did.
So the idea I had was probably void.
Still thinking.
You may try to remove the confirmation dialog completely, and see what happens.

pluginmanager.cpp:710
`            //OCPNMessageBox(gFrame, message, _("Un-Installation complete"), wxICON_INFORMATION | wxOK);
`

bdbcat, 2020-04-03:
Do you have any wxTimers running in the Draw plugin?

bdbcat, 2020-04-03:
Do you actually get the un-install successful confirmation dialog before the crash?

jongough, 2020-04-03:
Yes, I do have timers to handle rollover boxes.
Yes I do get the dialog, here is a screenshot:
[image]

jongough, 2020-04-03:
I have just done a similar test with testplugin_pi and it works OK. While it does have the includes to create a timer it does not actually instantiate one. I did the test with OD installed and both versions managed to find the installed OD and create objects.

bdbcat, 2020-04-03:
Jon...
This suggests that the trouble is in OCPN Draw, doesn't it?
Please try a test version with your ODTimer1 disabled.
`
    //ODTimer1.Start( TIMER_OD_1, wxTIMER_CONTINUOUS );
`
AFAICT, the plugin will still work, as long as "g_pODPointPropDialog" is not open.  Also no blinking.  This will try to discover if the errant timer is part of OCPN Draw, or coming from elsewhere.

I can see that the ODTimer1 is stopped in the DeInit() call, but lets be sure...

Thanks

bdbcat, 2020-04-03:
Of course, you will need to commit this test result to the master catalog, since we get no failures on the un-managed .deb build.

jongough, 2020-04-03:
If plugins having timers is an issue could OCPN have a timer call to the plugin if needed? This would allow OCPN to be in control of these events.

I have created OD 1.6.47.1 without the ODTimer1 being started. I installed the DEB file from cloudsmith, updated the ocpn-plugins.xml to point to the OD 1.6.47.1 tarball, then ran OCPN. OD worked as expected, no blinking of selected OD objects. Opened Options/Plugins, pressed the 'Upgrade' button, the download occured and the install said it was sucessful. Then immediately got a seg fault.
Here is the backtrace
```
#0  0x00007ffff58cd8b2 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#1  0x00007ffff58cd95f in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#2  0x00007ffff58cda13 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#3  0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#4  0x00007ffff58cd7d7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#5  0x00007ffff5831661 in wxTimerImpl::SendEvent() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff5de9861 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#7  0x00007fffef268d03 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#9  0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#12 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#13 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#14 0x00007ffff5e42cf3 in wxDialog::ShowModal() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#15 0x00005555557f65f0 in MyFrame::DoOptionsDialog() (this=0x555556453a70)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:6131
#16 0x00005555557f6f29 in MyFrame::DoSettings() (this=0x555556453a70)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5032
#17 0x00005555557f7a21 in MyFrame::DoSettings() (this=0x555556453a70)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5029
#18 0x00005555557f7a21 in MyFrame::OnToolLeftClick(wxCommandEvent&) (this=0x555556453a70, event=...)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:4659
#19 0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#20 0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#21 0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#22 0x00007ffff58cd783 in wxEvtHandler::DoTryChain(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#23 0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#24 0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#25 0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#27 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#28 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#32 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#33 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#34 0x00007ffff57510a6 in wxAppConsoleBase::MainLoop() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#35 0x00007ffff57dbae9 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#36 0x0000555555718db8 in main(int, char**) (argc=<optimised out>, argv=<optimised out>)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:1095
```

bdbcat, 2020-04-03:
Jon...
Timers can work fine in plugins, as long as they are carefully managed.

But for this bug, whose timer is is it that causes the crash? Difficult to debug....
And why only on this plugin, and only on your system, so far?
Are you able to install and uninstall all other plugins in the master catalog?
I need to think of another way to debug this....

bdbcat, 2020-04-03:
Jon...
Please try with all other plugins disabled.  Let's isolate OCPN Draw, and see what happens.

jongough, 2020-04-04:
I have tried on Ubuntu 16.04 and got a seg fault, here is the backtrace:
[ubuntu 16.04 backtrace.txt](https://github.com/OpenCPN/OpenCPN/files/4430475/ubuntu.16.04.backtrace.txt)
Here is the Log:
[log of ubuntu 16.04 seg fault.txt](https://github.com/OpenCPN/OpenCPN/files/4430485/log.of.ubuntu.16.04.seg.fault.txt)

I have all the plugins disabled execpt OD and none other than the standard set installed.

I built the OCPN (master b25ed04823) using 'cmake --build . -- -j6' and created the installable package using 'make package' both on the Ubuntu 16.04 machine. I downloaded the OD deb file from cloudsmith.

bdbcat, 2020-04-04:
Testing now, over and over...
I got the crash once, while disabling Draw.
I'll keep looking for a pattern.
FMI, what performance level is your system?  hi/med/lo?

jongough, 2020-04-04:
My testing is using qemu/kvm with Ubuntu 16.04 and Ubuntu 18.04 standard desktop installs as guests. I add gcc, cmake and other bits to let me build on these environments. I give the machines 6CPU's the 18.04 has 8GB memory and the 16.04 4GB memory. The virtual machine images are kept on a samba server with a 1Gb/s ethernet connection. This all runs on my desktop machine which is Linux Mint 19.3 with an AMD Ryzen 5 1600 six core processor with 2 threads per core and the system has 32GB memory. So I suppose 'med' would cover it.

The only thing that I cannot standardise is the deb file used to install OCPN as I have to build that myself. But I build it on the test machine so that it should match what is there.

For me the crash happens all the time. On Ubuntu18.04 it shows up as a locked dialog, on Ubuntu 16.04 the dialog closes, I can close the Options dialog, but the crash happens 'sometime' between clicking OK on the installation dialog box and getting back to the main screen.

jongough, 2020-04-04:
I have tried deleting all OD objects, then disabling the plugin, coming out of and going back into the options page. When I uninstall the Plugin Manager version I get the crash. So, I am doing:
1. Install OCPN via deb using 'sudo dpkg -i xxx.deb' after doing 'sudo apt purge opencpn'
1. Install OD via deb using 'sudo dpkg -i xxx.db'
1. Start OD via 'gdb opencpn', then 'r' to run it
1. Create an OD 3 point boundary, right click a boundary point to get the properties and make sure the boundary point doesn't blink (timer turned off). close properties
1. Open options/plugins and select OD and click 'Upgrade....' button
1. Click 'OK' on Installation complete dialog and 'OK' on Options dialog
1. Create a boundary to make sure all is working, then delete it
1. Open options/plugins select OD and select uninstall
1. Click on OK button of uninstall dialog - may crash at this time
1. Click on OK button of options dialog - does crash at this time

bdbcat, 2020-04-04:
Jon...
I may be chasing phantoms, but I may have identified the bogus timer callback...
Please make this patch to chcanv.cpp:10,494

    //pMovementTimer->Start( 2, wxTIMER_ONE_SHOT ); 

Also, be sure options->Display->General->Smooth Pan is OFF.

Thanks
Dave

bdbcat, 2020-04-04:
Jon...
Can't produce the fault with any plugin other than OCPN Draw.
May be still something funny going on with timers on OCPN Draw.  Either that, or some wxWindow not closing properly.
You may want to try shifting your wxTimer code from static event tables to dynamic binding, using Connect() or Bind().  Then you can directly UnBind() the timers during the DeInit() call.  This will ensure that there are no dangling timers which might be triggered later.

I'm still looking at OCPN code for a root cause..

rgleason, 2020-04-04:
I can test on a slow virtualbox ubuntu 18.04, with O 5.0.522 from this morning if it will help. I tried ocpn_draw and boundaries can be drawn but the lines don't show (win and ubuntu) but may not be related.

bdbcat, 2020-04-04:
Rick...
Sure, all testing appreciated.
We are most concerned with possible crash on un-install of managed OCPN Draw plugin.

rgleason, 2020-04-04:
I just tried uninstalling Ocpn_draw in very recent compile of  O 5.0.522 and OpenCPN crashed and disappeared.  Attached is log file, don't know as it shows anything that interesting, but perhaps it could show some more information that could be used to debug the problem?
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/4432276/opencpn.log)

rgleason, 2020-04-04:
I've tried Ubuntu 18.04 very recent build of Opencpn. and it crashed .
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/4432299/opencpn.log)

rgleason, 2020-04-04:
In ubuntu if I open up a tool like boundary, then go to options > plguins  the tool willl close. Then if I disable the plugin and uninstall it, there is no crash at all.

jongough, 2020-04-05:
With OD 1.6.51 I have moved the timers to Bind() rather than Connect(). This seems to have reduced the number of seg faults I am getting.  They are far more unpredictable now. I got one when doing the uninstall of the Plugin Manger version with the deb version still being there. But I have run the tests a few of times and only had the one seg fault. It appeared to be in the same place (chart1.cpp:1095), but I did not do a backtrace as I wanted to see if I could reproduce it.

I have put a few debugging messages in the constructors and destructors to see what is going on. Interestingly the ODEventHandler constructors and destructors give messages, but the ocpn_draw_pi (the main module) gives constructor, init, shutdown and deinit messages but never shows the destructor message. I am using wxLogMessage to provide these. Is there a reason why the main program destructor message is not coming out, or is the destructor never being called?

Going through the debugger I can see that the `extern "C" DECL_EXP opencpn_plugin* create_pi(void *ppimgr)` gets called but the `extern "C" DECL_EXP void destroy_pi(opencpn_plugin* p)` doesn't. I would have thought both the destroy_pi and the ~ocpn_draw_pi should have been called when terminating OCPN as well as when unloading and loading the plugin.

rgleason, 2020-04-05:
How are you guys installing these new versions?  

**Use the Custom URL Catalog** 
At first it did not seem to work for me.  I entered the cloudsmith URL to the tarball and clicked update.  There was a successful message, but I am not sure anything installed. There was no indicator that anything installed and there was no choice involved.  
You must enter the cloudsmith  *.xml metadata  URL to into the Custom URL.
On Cloudsmith, navigate to the most recent1.6.51.1 MSVC.xml  metadata file, right click on the green download icon and pick "Copy link location.  Then paste that into the Custom URL field in Plugin Manager.  Then "Update Custom".  A successful update message will pop up.
Now scroll to Ocpn_draw and select it and you can now Install version 1.6.51.1
This isn't that simple, but it does work. I picked upgrade to managed version 1.6.51.1 and it downloaded and upgraded.  Now when I go back to the master, beta or alpha catalogs, ocpn_draw disappears, even though the dll and data directories are properly installed. When I go back to the Custom URL and update the same url appears which is good, and opcn_draw reappears installed. But  IMHO this only allows one Custom URL when there should be at least several.
Would it be too confusing to pick up the dll's that are installed by "custom" and do work and show them as "Custom" even when on other catalogs?  What happens if you are not online with these catalogs?
[image]
  
**Download of the tarball and direct manual install.**
 I tried to expand the tarball into the proper %Localappdata% directory but that does not work because finding the new plugin is dependent on the xml catalogs, yet the dll exists in the correct directory!  Shouldn't OpenCPN find that dll?

**Use the manual installation exe file provided in Cloudsmith**
Then I decided to try the exe file that Jon provides and I believe it is loaded no  1.6.51 shows in the About tab.

**Adding the cloudsmith xml metadata to a meta-url**   
For more long term beta/alpha testing, if the developer has a meta-url in the alpha channel, this would be better.  Jon has not had time to do this yet.

rgleason, 2020-04-05:
Windows  1.6.51 I was able to select uninstall and it successfully uninstalled with a message, leaving 4 ocpn_draw green puzzle icons in the toolbar, that still worked. (perhaps an earlier version doubled?).

Then still on the Custom catalog, I could reinstall 1.6.51 with a download time bar and two of green puzzle icons went away and were replaced by the correct icons. Two green puzzle remained. They don't go away when I hide the tool bar and then show it.

When I go to the Alpha catalog, there are 4 ocpn_draw green icons and they all work.
It is Ocpn_Draw 1.6"   Under "Preferences"  About I find it is 1.6.51 and it must be the packaged "EXE version that I installed, that has the same version number.

bdbcat, 2020-04-05:
Jon...
I see the trouble with the plugin destroy call, or lack thereof.
Fixed i n github now, and I see your log messages on uninstall.

```
10:32:07.364 MESSAGE pluginmanager.cpp:1544 PlugInManager: Deactivating PlugIn: /home/dsr/.local/lib/opencpn/libocpn_draw_pi.so
10:32:07.364 MESSAGE ODEventHandler.cpp:150 Finished destructor
10:32:07.369 MESSAGE ocpn_draw_pi.cpp:889 Finished shutdown
10:32:07.369 MESSAGE ocpn_draw_pi.cpp:882 Finished deinit
10:32:07.369 MESSAGE pluginmanager.cpp:702 Uninstalling OCPN Draw
10:33:10.095 MESSAGE pluginmanager.cpp:1544 PlugInManager: Deactivating PlugIn: /home/dsr/.local/lib/opencpn/libocpn_draw_pi.so
10:33:12.453 MESSAGE ocpn_draw_pi.cpp:439 Finished destructor

This could have been the reason for some dangling timer pointers...
```

jongough, 2020-04-05:
I did wonder if it was something like that. The move to Bind rather than Connect aleviated the issue somewhat as there could be no timers left, but the code was still there. This would have been difficult to pick up without the 'legacy' timer mode and would have been a nightmare to find what the problem was. Pheeeew!

bdbcat, 2020-04-05:
So we call this closed?

jongough, 2020-04-05:
Yep

closed 2020-04-05 by jongough