OpenCPN/OpenCPN #1818: Plugin Manager - Ubuntu 18.04 uninstall leaves plugins and data in user directories

jongough, 2020-04-03:
Installed OCPN master deb build (36715392), installed OD 1.6.46 via Plugin Manager. Then used apt to uninstall OCPN using the 'purge' option. This cleaned out the /usr/local/* directories of OCPN data and programs, but the ~/.local/lib/opencpn, ~/.local/share/opencpn and ~/.opencpn directories were left. These should have been cleaned up. The apt purge command will not do so, but an installation script that could be run during the process would.

leamas, 2021-04-20:
Duplicate of #1539

jongough, 2021-04-26:
Different platform and hasn't been addressed on that one either. I would suspect that the issue will also be on Ubuntu 20.04 & 20.10.

leamas, 2021-04-26:
There are some open bugs filed about downloaded plugins being left in place after uninstalling the opencpn package. Extensive discussions has taken place in some of them.  The basic conclusion is that I and Jon disagree about these issues: Jon's position is that this is a  bug, my that things work as they should.

I don't see that continued discussion about this will be productive. These bugs are an example on case 2) in [Dave's opinion](https://github.com/OpenCPN/OpenCPN/discussions/2224#discussion-3334592) i. e., bugs "beaten to death due to honest differences of engineering opinion, but without resolution". 

According to Dave (same link) such bugs should be closed, basically as NOT-A-BUG (since status quo is the resolution). I concur.

leamas, 2021-04-26:
@rgleason:

> This one is more current.

Indeed. FYI, the traditional approach to duplicated bugs is to keep the oldest, basically telling reporters to not file the same bug again. The reasons for this policy seems obvious.

jongough, 2021-04-26:
Nothing happens with old bugs, they just get closed. It is sloppy to leave rubbish all over a system and expect users to clean up. I know you think it is the users responsibility but I disagree. If you create it then you clean it up. The installation processes Plugin Manager is replacing allows complete removal so why are you so reticent to clean up after yourself.

I now understand why ICT in the world is such a mess and nothing really worthwhile happens whilst billions get spent. I cannot be bothered to argue any further. I have already stopped submitting patches as you ignore them or overwrite them so why waste my time?

leamas, 2021-04-26:
> . I have already stopped submitting patches as you ignore them or overwrite them so why waste my time?

I have no role or permissions to do that -- only Dave and Pavel handles PRs and chenges in the OpenCPN code.  I from time to time make proposals, but I certainly don't merge.

jongough, 2021-04-26:
True, you don't do the merges, but I don't expect Dave or Pavel to inspect each and every change to see what it is doing. They are relying on people to play nice. I have had too many changes trampled over, the latest being the install process, so why bother putting anything up?

bdbcat, 2021-04-27:
Closing at status quo.

closed 2021-04-27 by bdbcat

jongough, 2021-04-27:
This is a change from the old way when doing an apt purge on linux would remove all the directories and data. I think this is sloppy coding, but.....

rgleason, 2021-04-27:
Alec wrote:
"Indeed. FYI, the traditional approach to duplicated bugs is to keep the oldest, basically telling reporters to not file the same bug again. The reasons for this policy seems obvious."

Alec as I have said before, we think differently  -- "Policy"?

leamas, 2021-04-27:
perhaps the wrong place, but since tht bug is closed anyway:

We can of course also in this case do whatever we want. But to me, active in multiple projects, it seems strange to have policy that differs to all other projects I have seen.  So I prefer if we could keep the oldest also here.

The argument against that is that no-one is looking at old bugs. The proper fix for this is to trim down the list of bugs, a process which is started but IMHO not finished until we can manage the bugs we have.