OpenCPN/OpenCPN #1819: Plugin Manager - Ubuntu 18.04 cancellation of tarball install locks up

jongough, 2020-04-03:
OCPN master (b25ed048) with OD 1.6.47.1 deb installed open Options/Plugins select OCPN Draw and click button 'Upgrade to managed Version 1.6.46', the download of the tarball starts. Cancel download and wait ~1min for response saying download aborted. The OK button on this dialog cannot be pressed as OCPN appears to have hung.
[image]
The OCPN log contains:
```
17:03:51.054 MESSAGE pluginmanager.cpp:642 Installing managed plugin: OCPN Draw
17:03:53.872 MESSAGE Downloader.cpp:132 filesize https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.46-ubuntu-18.04-bionic-tarball/versions/1.6.46.0+131.371ea5d/ocpn_draw_pi-1.6.46.0_ubuntu-18.04-bionic.tar.gz: 8604396 bytes

17:03:54.576 MESSAGE gshhs.cpp:1377 Loading World Chart Q=4 in 0 ms.
```
Here is the backtrace when this is happening
```
#0  0x00007ffff2c62775 in __GI___xstat (vers=<optimised out>, name=0x55555838c0b0 "", buf=0x7fffffffc710)
    at ../sysdeps/unix/sysv/linux/wordsize-64/xstat.c:35
#1  0x00007ffff57bf4cc in  () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#2  0x00007ffff57c09a7 in  () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#3  0x00007ffff57c40fa in wxFileName::DirExists() const () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#4  0x00005555559b8ef8 in PluginHandler::cleanup(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (filelist="/home/jon/.opencpn/plugins/install_data/ocpn draw.files", plugname="OCPN Draw")
    at /data/public/working area/opencpn/OpenCPN-5.0/src/PluginHandler.cpp:715
#5  0x00005555559e8fdf in pluginUtilHandler::OnPluginUtilAction(wxCommandEvent&) (this=<optimised out>, event=...) at /data/public/working area/opencpn/OpenCPN-5.0/src/pluginmanager.cpp:665
#6  0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00007ffff58cda13 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#10 0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#11 0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#12 0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#13 0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#14 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#15 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#19 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#20 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#21 0x00007ffff5e42cf3 in wxDialog::ShowModal() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#22 0x00005555557f65f0 in MyFrame::DoOptionsDialog() (this=0x55555644de60)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:6131
#23 0x00005555557f6f29 in MyFrame::DoSettings() (this=0x55555644de60)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5032
#24 0x00005555557f7a21 in MyFrame::DoSettings() (this=0x55555644de60)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:5029
#25 0x00005555557f7a21 in MyFrame::OnToolLeftClick(wxCommandEvent&) (this=0x55555644de60, event=...)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:4659
#26 0x00007ffff58cd4be in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#27 0x00007ffff58cd5c3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#28 0x00007ffff58cd98b in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#29 0x00007ffff58cd783 in wxEvtHandler::DoTryChain(wxEvent&) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#30 0x00007ffff58cda75 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#31 0x00007ffff58cebe2 in wxEvtHandler::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#32 0x00007ffff574f9e7 in wxAppConsoleBase::ProcessPendingEvents() ()
    at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#33 0x00007ffff5db6bbd in wxApp::DoIdle() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#34 0x00007ffff5db6cc3 in  () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#35 0x00007fffef268285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#36 0x00007fffef268650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#37 0x00007fffef268962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#38 0x00007fffefb3ba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#39 0x00007ffff5dd6b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#40 0x00007ffff5789a93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#41 0x00007ffff57510a6 in wxAppConsoleBase::MainLoop() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#42 0x00007ffff57dbae9 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#43 0x0000555555718db8 in main(int, char**) (argc=<optimised out>, argv=<optimised out>)
    at /data/public/working area/opencpn/OpenCPN-5.0/src/chart1.cpp:1095
```

bdbcat, 2020-04-07:
Fixed.
Confirm/close?

jongough, 2020-04-07:
Fixed

closed 2020-04-07 by jongough