OpenCPN/OpenCPN #1820: O 5.0.522 Auto follow

Hakansv, 2020-04-04:
We have Auto follow mode and Look ahead mode.
Auto follow shall show the boat in center or at a position moved to by a mouse click or mouse move.
Look ahead mode shall keep the position and if moved being forced back to correct position, earlier on "mouse up", e.g. when the mouse button is released.

Now on Win10 double canvas. 
Auto follow on the left and Look ahead at the right canvas.
If I click on the left canvas the boat is moved to a strange screen position. And if I move/drag the chart the boat symbol jumps to a opposite position. While moving or after.
Look ahead mode is moving the boat back while I drag/move the screen by the mouse and does not wait until mouse button is released.

Hard to describe correct but make an own try.
Håkan

bdbcat, 2020-04-05:
Corrected in github now.

As a second point, and I think we have talked about this before...
You could say that dragging a chart with "Look-Ahead" engaged is a lost cause.  The chart will always return to near the previous position.  Maybe a drag action on a "Look-Ahead" view should automatically cancel "Look-Ahead" mode.
What do you think?

Hakansv, 2020-04-05:
Dave..
First point.
Yes, partly corrected. The mouse drag is now working so I can move the boat to a desired position. But now I see there's another strange occurrence. Locate the boat symbol to for example upper right  quadrant. Zoom out or in using +/- sign. Then is the chart area redrawn with a huge flickering and on nearly every second zoom event the boat symbol is moved to the bottom right quadrant. The boat is occasionally altering between the upper and bottom quadrant while repeating the zoom keys. Also while using the mouse roll the same flickering and boat location moving occurs.

Second point.
Yes, it was discussed this a lot before version 5.0 release. We ended up in the present functionality. Personally I'm used to it and have trained many others to use it. Also I haven't seen any direct querulous opinions about this on the forum. So for me it's good as it is.

bdbcat, 2020-04-05:
Hakan...
Corrected in github now.

Hakansv, 2020-04-05:
Yes, now "back to business" again. Works as usual.
Thanks
Håkan

closed 2020-04-05 by Hakansv