OpenCPN/OpenCPN #1821: O 5.0.522. The boat position is changed when a route is completed.

Hakansv, 2020-04-04:
I'm out sailing using our alpha code. A strange and surprisingly occurrence happened.

I was following an activated route and come to the end of it. Then suddenly the boat position was changed! Well, as soon new GPS positions arrived the correct position was recovered. I saw this happened twice. (The AP was not engaged at the time if you wonder about the bad course stability)
See the shot with the trace. The route navigation was automatically ended at the arrow and the route deactivated. Then the position changed. It took some seconds until the correct position was established but I've 3 seconds course and speed filter and 10 seconds chart rotation averaging time. 
Could there be a single quote instead of a double somewhere?

[image]

bdbcat, 2020-04-05:
Hakan...
Could not replicate.
I tried to replicate using your standard VDR recording.  No luck.
Maybe you could try that to get a repeatable case?
Dave

Hakansv, 2020-04-05:
Dave.
I can't reproduce on this Win PC either.
But at the boat PC, also Win10, it happened three times today. Not only when the route is deactivated automatically but also when I, by hand, deactivated a route I was following. But for next time I'll try to record a VDR to try to catch what's happening, I'm single handed and today it was narrow waters so no time for cockpit work..
I'll be back.

Hakansv, 2020-04-05:
One of today's occurrences.
As seen here the jump position is often going back to where the route was initiated.
[image]

Hakansv, 2020-04-08:
Dave..
Today I made a VDR while navigating a route. When at the last WP and the route was automatically deactivated the position changed several times. 
Investigating the VDR file you can see the position is changed due to old RMC. From somewhere are RMC where the time stamp and position are from the past coming into the actual data stream.

Heres a copy from the NMEA debug window when the route was ended. The first time and pos is correct. Next message is old with a position earlier on the route. Then there are messages with altering time stamps for while. Then back to normal.
Where are old position messages saved? How can an old message appear in the multiplexer when the route is ended? (It's not VDR. This is the first time the VDR plugin was activated.)
`
$GPRMC,105243.00,A,5751.26103,N,01138.79647,E,5.276,9.95,080420,,,A*62<0x0D><0x0A>
$GPRMC,100320,A,5746.962,N,01138.642,E,004.4,351.7,080420,000.0,E*7E<0x0D><0x0A>
$GPRMC,100322,A,5746.965,N,01138.641,E,004.3,351.7,080420,000.0,E*7F<0x0D><0x0A>
$GPRMC,105245.00,A,5751.26396,N,01138.79717,E,5.281,8.12,080420,,,A*68<0x0D><0x0A>
$GPRMC,105246.00,A,5751.26542,N,01138.79748,E,5.205,10.36,080420,,,A*5D<0x0D><0x0A>
$GPRMC,105247.00,A,5751.26686,N,01138.79794,E,5.297,9.50,080420,,,A*65<0x0D><0x0A>
$GPRMC,100324,A,5746.969,N,01138.639,E,004.2,354.2,080420,000.0,E*7B<0x0D><0x0A>
$GPRMC,105248.00,A,5751.26829,N,01138.79846,E,5.297,8.66,080420,,,A*65<0x0D><0x0A>
$GPRMC,105249.00,A,5751.26981,N,01138.79886,E,5.373,7.70,080420,,,A*68<0x0D><0x0A>
$GPRMC,100326,A,5746.972,N,01138.637,E,004.3,355.1,080420,000.0,E*7E<0x0D><0x0A>
$GPRMC,105250.00,A,5751.27128,N,01138.79932,E,5.334,9.58,080420,,,A*63<0x0D><0x0A>
$GPRMC,105251.00,A,5751.27276,N,01138.79974,E,5.350,8.80,080420,,,A*6E<0x0D><0x0A>
$GPRMC,105252.00,A,5751.27424,N,01138.80005,E,5.260,6.13,080420,,,A*63<0x0D><0x0A>
$GPRMC,100328,A,5746.974,N,01138.638,E,004.3,353.7,080420,000.0,E*79<0x0D><0x0A>
$GPRMC,105253.00,A,5751.27573,N,01138.80032,E,5.277,5.80,080420,,,A*6A<0x0D><0x0A>
$GPRMC,105254.00,A,5751.27721,N,01138.80066,E,5.281,8.12,080420,,,A*66<0x0D><0x0A>
$GPRMC,100330,A,5746.975,N,01138.639,E,004.3,354.2,080420,000.0,E*72<0x0D><0x0A>
`

Two files attached: 
test_route_end1.txt :>  The interesting part of the whole saved VDR.
test_route_atfail.txt :> Part of the previous where the position failed started.

[test_route_end1.txt](https://github.com/OpenCPN/OpenCPN/files/4452108/test_route_end1.txt)
[test_route_atfail.txt](https://github.com/OpenCPN/OpenCPN/files/4452109/test_route_atfail.txt)

Today's shot:
[image]

transmitterdan, 2020-04-08:
Can you also supply the route as a GPX file?  With that I can simulate the route and perhaps discover what is happening.

bdbcat, 2020-04-09:
Hakan....
In absence of VDR, GPRMC messages are not saved anywhere.  The are sent to, and consumed by, the multiplexer.  More or less instantly, depending on system load.
Do you have you GPS connection configured for transmit from OCPN?  Is there any possibility that the GPS receiver is sending these errant GPRMC messages as a response to autopilot control messages (e.g.ECAPB) messages sent by OCPN?

Hakansv, 2020-04-09:
Dave..
> Do you have you GPS connection configured for transmit from OCPN?

No, the GPS has only its TX connected. 
 
> Is there any possibility that the GPS receiver is sending these errant GPRMC messages as a response to autopilot control messages (e.g.ECAPB) messages sent by OCPN?  

No, the AP has its own single wire.
And, this change of the boat position has occurred also when the AP was not engaged but I was steering after the route by hand.

My next step would be to catch the NMEA debug window to see the source when this false position happens. I'll try today. 

@Dan..
Two gpx attached where this false position happened. I've tried to recreate this issue on another PC but didn't succeed. 

[halso.gpx.txt](https://github.com/OpenCPN/OpenCPN/files/4454316/halso.gpx.txt)
[test.gpx.txt](https://github.com/OpenCPN/OpenCPN/files/4454317/test.gpx.txt)

Hakansv, 2020-04-09:
Dave.. Dan..
This has probably boiled down to a multiplexer filter malfunction?
A RMC from a navigation system I use to filter out is still coming through. Stay tuned, I'll be back when moored. 
Håkan

transmitterdan, 2020-04-09:
@Hakansv,

Not only GPRMC, but GPGLL as well.

Hakansv, 2020-04-09:
Dan..
Correct! See below
All..
So the problem was the filter function or more connection settings in the  config file.
I've a old Simrad IS15 that does something strange with a received RMC, and others, and return them in a strange manner. Some of them is the cause for this problem. How this gadget can return outdated position data is a mystery I wont evaluate now. I've always filtered out all such data from this device.
Now was the filter function not working. For example the RMC was set to be filtered out but still passed through. There where also strange filter behavior for other streams. What I did was to delete all connections and restarted to get a clean config file. Then again set up all connections in the same manner with filter and prioritization and now it's working as it should. And no position failure at route ends.

As a possible conclusion but no blame of any kind:
I've seen a corrupt config file connections string before on my home PC. In the early phase of O v5.0.522 when I started to play with SignalK the connection parameters was corrupt at some occasions so I'd to delete it and reset. It could be there was some residues of that on the boat PC as well. I once had it home and made some testing??
Anyhow, now all is fine and hopefully some lessons learned.
Thanks for your support and patience.
Håkan

closed 2020-04-09 by Hakansv

rgleason, 2020-04-09:
So when starting out using SignalK would it be best to assume you should reconfigure everything?

Hakansv, 2020-04-09:
Rick..
No I don't think that's at all necessary. Much is changed after my experiences above.