OpenCPN/OpenCPN #1822: Ubuntu 20.04:  Unreadable dialogues in Dark Themes

free-x, 2020-04-05:
I built O 5.0.522 (master) on Ubuntu 20.04 I notice that some dialogs in Settings are not readable in Dark Theme
Here are some screenshots
[image]
[image]

free-x, 2020-04-18:
after https://github.com/OpenCPN/OpenCPN/commit/f26b28f3c8aaf588cb315a08e131747a775e97a4
pluginmanager looks good

[image]

free-x, 2020-05-26:
after https://github.com/OpenCPN/OpenCPN/commit/70292f9d078fa9d7df58ebf90d40377827621a5e
connection manager looks good
templates dialog still remains
[image]

rgleason, 2021-04-23:
Can this be closed now?  Original Poster should close.

closed 2021-04-24 by free-x