OpenCPN/OpenCPN #1823: PI Manager: When xml is not in a Catalog

rgleason, 2020-04-05:
How are you guys installing these new versions that aren't in a Catalog?  

**Use the Custom URL Catalog** 

At first it did not seem to work for me.  I entered the cloudsmith URL to the tarball and clicked update.  There was a successful message, but I am not sure anything installed. There was no indicator that anything installed and there was no choice involved.  

You must enter the cloudsmith  *.xml metadata  URL to into the Custom URL.

1. On Cloudsmith, navigate to the most recent1.6.51.1 MSVC.xml  metadata file, 
2. Right click on the green download icon and pick "Copy link location.  
3. Then paste that into the Custom URL field in Plugin Manager.  
4. Then "Update Custom".  A successful update message will pop up.
5. Now scroll to Ocpn_draw and select it and you can now Install version 1.6.51.1
6. This isn't that simple, but it does work. 
7. I picked upgrade to managed version 1.6.51.1 and it downloaded and upgraded.  
8. Now when I go back to the master, beta or alpha catalogs, ocpn_draw disappears, even though the dll and data directories are properly installed. 
9. When I go back to the Custom URL and update the same url appears which is good, and opcn_draw reappears installed. 
10. But  IMHO this only allows one Custom URL when there should be at least several.

Would it be too confusing to pick up the dll's that are installed by "custom" and do work and show them as "Custom" even when on other catalogs?  What happens if you are not online with these catalogs?

[image]
  
**Download of the tarball and direct manual install.**

I tried to expand the tarball into the proper %Localappdata% directory but that does not work because finding the new plugin is dependent on the xml catalogs, yet the dll exists in the correct directory!  

Shouldn't OpenCPN find that dll?

**Use the manual installation exe file provided in Cloudsmith**

Then I decided to try the exe file that Jon provides and I believe it is loaded no  1.6.51 shows in the About tab.

**Adding the cloudsmith xml metadata to a meta-url**   

For more long term beta/alpha testing, if the developer has a meta-url in the alpha channel, this would be better.  Jon has not had time to do this yet.

leamas, 2020-04-05:
Custom URL only supports using a catalog. That is, the way to use it is to clone the plugins project, uupdate metadata and tarball(s) for modified plugin(s), rebuild the catalog and make this catalog available on an URL.

The alternative is to feed metadata + tarballl io ocpn-install which installs it in place.

rgleason, 2020-04-05:
I had forgotten about the python ocpn-install option. Have never used it.

leamas, 2020-04-05:
>   This seems like a step backward to me.

It's like it always has been, here is no change whatsoever.

rgleason, 2020-04-05:
Yes, it is the way it started, but coming from the previous version after years, this is pretty strange. You totally eliminated the search for plugins in the plugin folder and substituted xml files entirely. It was supposed to be backwards compatible as per bdbcat!  This isn't backwards compatible in my opinion.

Additionally, all those extra steps to publish the metadata, when we just want to test a plugin is really a pain.

rgleason, 2020-04-05:
Faster Alternative is to just clone the plugin and compile locally and install.

bdbcat, 2020-04-05:
Rick...
"So the whole routine in Opencpn, to find plugin executables (dll) is replaced by online xml's??

If I have installed a tarball properly and the dll exists in the right location, the plugin still will not work. It appears that for that plugin to work, you must be online and have access to the xml catalogs. This seems like a step backward to me."

Please elaborate.  I got lost in your explanation of steps taken.

leamas, 2020-04-06:
> You totally eliminated the search for plugins in the plugin folder and substituted xml files entirely.

No.

rgleason, 2020-04-06:
This is a python script which uses the metadata xml to install the tarball.
"The alternative is to feed metadata + tarballl io ocpn-install which installs it in place."

Ok, So I ask this. If I expand the tarball in the proper place, IE
%LocalAppData% \opencpn\plugins  OR  C:\Users\[user]\AppData\Local\opencpn\plugins

Will it be available for use?   NO

Or do I have to try to install the tarball into
C:\Program Files (x86)\OpenCPN [version]\plugins
in order to run it?

Using Plugin Manager, are online XML files required in order to run the program?
If NO, then how does it work?

rgleason, 2020-04-06:
After disconnecting wifi.  Wxroute, oesenc etc showed installation "from cache"
[image]

Where is the "cache" located?

However ocpn-draw showed this.
[image]

Previously I had installed this ocpn-draw  1.6.51  and then uninstalled it. Now for some reason v1.6.46 is not available.
[image]

leamas, 2020-04-06:
> Ok, So I ask this. If I expand the tarball in the proper place, [...]
Will it be available for use? NO

Yes, it will. However, it will probably miss supports files like data and/or helper libraries causing more or lessf trouble. The benefits of ocpn-install is that it installs all required files into proper locations and also adds metadata so the plugin can be uninstalled.

rgleason, 2020-04-06:
[image]
When wifi is turned off, you cannot shift to another catalog.

rgleason, 2020-04-06:
Where is the cache located?  Is it accessible?  If so how?

[image]

rgleason, 2020-04-06:
If a user wants to have certain plugins (master, beta, alpha, personal)  available for fresh installation while voyaging, what is the best procedure?

If we want to be able to quickly download from cloudsmith and install a plugin which does not have an xml record in one of the catalogs, what is the quickest and easiest procedure? 

What catalog should we use to access that plugin?

leamas, 2020-04-06:
> If a user wants to have certain plugins (master, beta, alpha, personal) available for fresh installation while voyaging, what is the best procedure?

I simply don't know. To be frank. this use case is not really supported...(@bdbcat: please correct me if I'm wrong...)

> If we want to be able to quickly download from cloudsmith and install a plugin which does not have an xml record in one of the catalogs, what is the quickest and easiest procedure?

There is no such support, and there should IMHO not not be any.  Any user which has the ability to download the tarball can also download the metadata. 

Note that ocpn-install doesn't require a catalog, it uses the xml file produced when building the plugin. 

Which boils down  to downloading all plugins planned for testing + their metadata files. Then use ocpn-install to install them (and the regular installer to uninstall). This should more or less accomplish what our yet unknown user seems to require ;)

rgleason, 2020-04-06:
I am going to try this, but do the plugins land in the same directory?

leamas, 2020-04-06:
It depends on you, you need to download them to some place (together with their xml metadata files). As I said earliesr, there is no support as such for this.

rgleason, 2020-04-06:
ocpn-install.sh  install requires python, sed and bash. Most Windows users don't even know about these programs.

Also there is no use explanation or Readmen  for the format and options.
It appears to download all 4 environments listed.

leamas, 2020-04-06:
> ocpn-install.sh install requires python, sed and bash. Most Windows users don't even know about these programs.

It's not meant for regular Windows user. It's meant for developers. In this context these dependencies shouldn't be intimidating. 

Regular Windows users are supposed to use supported plugins in the catalog.

bdbcat, 2020-04-06:
Rick...
I will try to address some of the questions.
First major points:
1.  Most of the time, Beta  and Alpha catalogs will be empty.  EMPTY.  There will be nothing that a "normal" user will want or need that is not in "master".
2.  "Normal" users will not see "Alpha" and "Custom" catalog choices unless they manually set a value (TBD) in the config file. 
3.  Presumably, if there is something new in Beta for specific testing, the user will be informed by CF announcement.  If the user can see the announcement, she is online, and so then Beta catalog works.
4.  Developers are "special"  It is assumed that devs are able to run CLI commands, bridge over the gaps in documentation, figure out what is needed, and make progress.  Normal users are not expected to do these things.
5.  To re-inforce:  Manual installation of tarballs is possible, but not supported for end users.  Devs can and will do what is necessary to test.

Now, to your questions:
1._"If a user wants to have certain plugins (master, beta, alpha, personal) available for fresh installation while voyaging, what is the best procedure?"_
Procedure is TBD, there are several choices.  When this is settled, it will be communicated.
I understand the visceral need for this functionality...

2. _Where is the cache located? Is it accessible? If so how?_
Cache is in {config dir}/plugins/cache.  Easily accessible.  No plans to make these tarballs usable directly by end users.  No promise that the cache will be maintained, or the current and coherent.  It is mainly for OCPN internal use.  Devs will know what to do with these tarballs, if they want.

3. _"If we want to be able to quickly download from cloudsmith and install a plugin which does not have an xml record in one of the catalogs, what is the quickest and easiest procedure?"_
This is probably for Devs only.  I see no use case for an end user to do this.  I assume that this is a transient problem with OCPN Draw, since the metadata for v51has not been merged to any catalog yet.  So a real end-user will not see this scenario.
Simple solution is to download the metadata XML file, and manually update the ocpn-plugins.xml file found in the OCPN config directly. This edited ocpn-plugins.xml will persist until another catalog is downloaded. 

Any More?

rgleason, 2020-04-07:
Thanks,

If a plugin is uploaded to cloudsmith, it is reasonably easy to download and install the tarball, and I guess I should make a batch file for that sometime.

I have been wondering if the catalogs are saved somewhere.  (C:\ProgramData\opencpn\ocpn-plugins.xml ).  You advise that there is a catalog file stored temporarily, so I guess that ocpn_plugins.xml is replaced each time a new catalog is selected (master, Beta, Alpha) too?

I don't believe I am a dev.

bdbcat, 2020-04-07:
_"so I guess that ocpn_plugins.xml is replaced each time a new catalog is selected (master, Beta, Alpha) too?"_
Absolutely correct.
But please note:
The currently installed plugins are not disturbed during this session, because their metadata is preserved by the plugin manager FOR THIS SESSION ONLY.
When you next start OCPN, any plugin this is installed, but not found in the currently loaded catalog, will be described as "unmanaged".

So I see the use-case for Alpha and Custom as follows:
1.  For use by the dev of the plugin itself, for fast turn around of build iterations, without having to wait for me to accept a PR.  In this case, the dev does not likely care about any other plugins, so safe to just stay on Alpha (or Custom) catalog.
.or.
2.  For the plugin dev to share with a trusted tester, for a one-off quick test of a bug fix or feature.

In both cases, I would expect that the "lifetime" of an Alpha catalog plugin would be only for a few OCPN trials.  Very soon, the plugin would migrate to Beta.

We are definitely looking at transient issues here.
Let's get master catalog correct, and focus testing on what a normal user would see.

rgleason, 2020-04-07:
Thanks, Closing.

closed 2020-04-07 by rgleason