OpenCPN/OpenCPN #1825: Plugin Manager - Ubuntu 18.04 Plugin Manager screen not resized when picking custom catalog

jongough, 2020-04-05:
Using OCPN master (b25ed04823) open Options/Plugins
[image]
select custom catalog
[image]
The box to put the url in is not displayed as the dialog size is incorrect.

rgleason, 2020-04-05:
This is also occurring in Windows, but we see just 1/2 or 1/4 of the URL box.  The slider does not control this, we need to grab the corner and make the window larger. Is this a problem on the raspberry PI screens?

bdbcat, 2020-04-06:
Corrected by b45153f1f9c1f35fb0cc756813cc05b5e4a6e743

jongough, 2020-04-06:
Almost, if you open the Options dialog with the Plugins tab already selected and 'Choose Remote Catalog' already set to 'Custom...' the URL box does not show up.
b45153f1f9c1f35fb0cc756813cc05b5e4a6e743
This was done with master (b45153f1)

bdbcat, 2020-04-07:
jon...
Sorry, cannot reproduce this.
Exact steps, please?

bdbcat, 2020-04-07:
Jon...
OK, I see it.  Requires restart of OCPN after selecting a Custom... catalog.

Fixed by eda25b76638cfacb0bf489a36e705009d8d721ee
Please confirm/close.

jongough, 2020-04-07:
Fixed

closed 2020-04-07 by jongough