OpenCPN/OpenCPN #1826: Plugin Manger - Ubuntu 18.04 Plugin Manager seems to start multiple copies of plugin

jongough, 2020-04-05:
Using OCPN master (b25ed0482) start program
[image]
open Options/Plugins and select a different catalog and update it (this step seems to variable in the number of times needed to get the effect) then click OK and the OCPN toolbar changes to have multiple copies of the same plugin although the icons are missing
[image]
[log of ocpn showing multiple icon locations for same plugin.txt](https://github.com/OpenCPN/OpenCPN/files/4434931/log.of.ocpn.showing.multiple.icon.locations.for.same.plugin.txt)

rgleason, 2020-04-05:
Checking plugin candidate: /usr/local/lib/opencpn/libocpn_draw_pi.so
Isn't this the only location?

jongough, 2020-04-05:
Just had it again using master (ffcb285d). Looking in the log OD does not appear to have been reloaded as there are no messages about constructors (these are messages inserted to debug another issue) or destructors. With the mouse over the jigsaw icon this is what can be seen.
[image]

bdbcat, 2020-04-06:
Jon...
Cannot repro.
Are you repeatedly loading the same catalog, or switching each time?  Which catalogs?

jongough, 2020-04-06:
No I am loading different catalogs, rather randomly at the moment, so I have not worked out a repeatable process, but I think the issue occurs when the OK button is clicked on the main Options/Plugins dialog. I have just caused it, but with, this time, four extraneous icons:
[image]
and after further clicking on the 'extra' icons they start to change and sometimes they work and sometimes they don't. It looks like the toolbar manager has become 'confused'
[image]

bdbcat, 2020-04-06:
OK, I can repro this.
Debugging....

bdbcat, 2020-04-07:
Corrected by f4755cd0ffb716576acb0b7b1827871b809076ba
Please confirm/close.

jongough, 2020-04-07:
Fixed

closed 2020-04-07 by jongough